project-chip/connectedhomeip #17859: Paring issue noticed  when trying to pair with an on-network device

phonnakasturi-apple, 2022-04-28:
Observed in SHA cfc35951be66a664a6efdadea56d1b8ea6e63e96

Frequency of occurrence: Noticed thrice out of 10 times

**Steps to reproduce:**

1. cd ~/connectedhomeip
2. source scripts/activate.sh
3. scripts/examples/gn_build_example.sh examples/all-clusters-app/linux out/debug chip_config_network_layer_ble=false
4. ./out/debug/chip-all-clusters-app --discriminator 3841 --secured-device-port 5560 --KVS /tmp/chip_kvs_test

Attaching the complete log file:

[on-network_pairing_issue_log.txt](https://github.com/project-chip/connectedhomeip/files/8586822/on-network_pairing_issue_log.txt)

tcarmelveilleux, 2022-04-28:
This is missing the steps done on the commissioner side. Please provide those.

Re: the failire:

```
1651181773522] [5869:6087525] CHIP: [SPT] VerifyOrDie failure at ../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/lib/support/Pool.h:217: Allocated() == 0
zsh: abort      ./out/debug/chip-all-clusters-app --discriminator 3841 --secured-device-port 
```

Please ensure your bug reports have:

- Summary of scenario
- Full TH and DUT steps
- Logs on both sides
- Expected behavior
- Observed error/bad behavior snippet.

phonnakasturi-apple, 2022-04-28:
Updated the issues with the requested info

kghost, 2022-04-29:
@phonnakasturi-apple

Can you provide full log ? both device side and commissioner side.
Can you provide the steps done on the commissioner side ?

TorkelIKEA, 2022-04-29:
I've seen similar issues when running on-network-commission in a LAN multiple Matter devices (around 20).
In my case, I could fix it by extending the Pools.

``` diff --git a/src/platform/Linux/InetPlatformConfig.h b/src/platform/Linux/InetPlatformConfig.h
index 3aab9a7b9..02e664edd 100644
--- a/src/platform/Linux/InetPlatformConfig.h
+++ b/src/platform/Linux/InetPlatformConfig.h
@@ -33,7 +33,7 @@
 // ========== Platform-specific Configuration Overrides =========
 
 #ifndef INET_CONFIG_NUM_TCP_ENDPOINTS
-#define INET_CONFIG_NUM_TCP_ENDPOINTS 32
+#define INET_CONFIG_NUM_TCP_ENDPOINTS 128
 #endif // INET_CONFIG_NUM_TCP_ENDPOINTS
 
 #ifndef IPV6_MULTICAST_IMPLEMENTED
@@ -41,7 +41,7 @@
 #endif
 
 #ifndef INET_CONFIG_NUM_UDP_ENDPOINTS
-#define INET_CONFIG_NUM_UDP_ENDPOINTS 32
+#define INET_CONFIG_NUM_UDP_ENDPOINTS 128
 #endif // INET_CONFIG_NUM_UDP_ENDPOINTS
 
 // On linux platform, we have sys/socket.h, so HAVE_SO_BINDTODEVICE should be set to 1

TorkelIKEA, 2022-04-29:
My fix was based on the assumption that the problem was triggered by exhausted pools and the "Allocated()" counter got out of sync

phonnakasturi-apple, 2022-05-03:
Adding logs from commissioner + commissionee device:

[on-network-maclog.txt](https://github.com/project-chip/connectedhomeip/files/8612182/on-network-maclog.txt)
[Uploading on-network-iphonelog.txt…]()

closed 2022-11-12 by stale[bot]