project-chip/connectedhomeip #17868: Does iOS Home app support commissioning for already-on-network case?

ti-tpan, 2022-04-28:
#### Problem

Hi all,

I see that in https://github.com/project-chip/connectedhomeip/issues/13296, the Home app can enable Matter support with a profile configuration from Apple.

Do we know if already-on-network commissioning is possible?

Specifically, we are interested in a sequence like the following:

1. Using proprietary provisioning, provision Wi-Fi MCU onto local Wi-Fi network. This Wi-Fi MCU is now on a Wi-Fi network.

2. Use Home app on iPhone (connected to same local Wi-Fi network as 1.) to scan QR code for that MCU. 
- This QR code should indicate that this MCU is “already-on-network”, i.e. MCU is on same Wi-Fi network as iPhone

3. Home app can discover and pair with MCU. 
- This should allow Home app to now interact with MCU (e.g. toggle an LED).

However, after the iPhone scans the device's QR code, it seems to not work. We expect the iPhone to send MDNS queries to find our Matter MCU on the Wi-Fi network, but this is not the case (we don't see any MDNS query from the phone).

 #### Proposed Solution

Look forward to hearing your thoughts!

ti-tpan, 2022-05-03:
Tagging some folks as in this post: https://github.com/project-chip/connectedhomeip/issues/13296
(tagging some of the Apple team for visibility @carol-apple , @vivien-apple @pan-apple @bzbarsky-apple) 

Is this the right channel to ask about issues like this?

Thanks

bzbarsky-apple, 2022-05-03:
Asking on Slack is probably better if asking questions, but this sounds like a bug report based on the "seems to not work" bit above?.

That said, Matter spec requires that on-network discovery always be performed and the Matter SDK always performs it.  So I would expect this to work.  @sagar-apple might know reasons why it would not work and how you could gather relevant logs.

ti-tpan, 2022-05-04:
Thank you.
Yes, it looks like a bug we are seeing, but wanted to see if anyone had tried it before.

In the meantime, we will continue debugging.

ti-tpan, 2022-05-06:
One update is that the iOS Home app indicates something like "failure to join this network" (after scanning the QR code).

Are the logging instructions/profiles provided here also relevant when debugging Matter-related Homekit issues (on both iOS and tvOS)?
https://developer.apple.com/bug-reporting/profiles-and-logs/

bzbarsky-apple, 2022-05-06:
They should be, yes.

closed 2022-11-12 by stale[bot]