project-chip/connectedhomeip #17873: M1 Mac (Apple Silicon) Issues

rafaliya-amazon, 2022-04-28:
### Problem
 * expected behavior
     * `source scripts/activate.sh` should run properly. And we should be able to run **darwin** in Xcode.
 
* actual behavior
    * Getting error on M1 Mac (Apple Silicon)
    *Please find the error below*
   
 * steps to reproduce
    * Clone the code by `git clone --recurse-submodules git@github.com:project-chip/connectedhomeip.git`
    * Try running `source scripts/activate.sh`
    OR
    * Open `Darwin.xcworkspace` inside 
 * system configuration: M1 Mac Silicon

* Xcode version: 13.2.1 & 13.3.1 (does not work on either)

 ### Logs:

WELCOME TO...

                ░▓░
                ▓█▓
                ▓█▓                                                     ▒█     ▒█
           ▒██▒▒▓██▒███▒                ░▒▓▒░  ░▒▓▒░       ░░▓█▒░ ░█  █████████████░    ░▒█▒░       ░░▒░
            ░▓█████▓██░               ▒█▒░░▒▓██▓▒░░▒█▒   ░█▓▒░░▒████    █▓░    █▓░    ▒█▒░░░▒█▓░   ██▒░░
         ▒█▒░         ░██▒           ░█░     █▓     ░█░ ░▓▒      ░▓█    █▓     █▓    ▒█░░    ░██  ▒▓
         ░▓██▓░     ░██▓█░           ░█      ▓█      █░ ░█░       ██    █▓     █▓    ▓██████████  ▒█
          ░▓███▒   ▒███▒░            ░█      ▓█      █░  ██░    ░███    █▓     █▓    ░█▒░         ▒█
      ░▒████████░ ░███▓▓█▓▓▒         ░█      ▒▒      █░   ░█▓██▓█░▒█    ░▓▓█░  ░▓▓█░   ▒▓▓██▓█░   ▒▓ 
      ░██▒░  ▒██▒ ▒██░  ░▒█▓
              ░▓░ ░▓░

  **BOOTSTRAP! Bootstrap may take a few minutes; please be patient.**

**Downloading and installing packages into local source directory:**

  Setting up CIPD package manager...done (3.6s)
  Setting up Python environment.....Traceback (most recent call last):
  File "/Users/rafaliya/Matter/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 679, in <module>
    sys.exit(main())
  File "/Users/rafaliya/Matter/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 671, in main
    return EnvSetup(**vars(parse())).setup()
  File "/Users/rafaliya/Matter/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 414, in setup
    result = step(spin)
  File "/Users/rafaliya/Matter/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 533, in virtualenv
    if not virtualenv_setup.install(
  File "/Users/rafaliya/Matter/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 148, in install
    version = subprocess.check_output(
  File "/Users/rafaliya/Downloads/Xcode 13.2.1.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/subprocess.py", line 415, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/Users/rafaliya/Downloads/Xcode 13.2.1.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/subprocess.py", line 493, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/Users/rafaliya/Downloads/Xcode 13.2.1.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/subprocess.py", line 858, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/Users/rafaliya/Downloads/Xcode 13.2.1.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/subprocess.py", line 1704, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/usr/local/bin/python3'
pw_cleanup:unset:32: no such hash table element: _pw_cleanup
pw_cleanup:unset:33: no such hash table element: _pw_alias_check
scripts/activate.sh:unset:102: no such hash table element: pw_cleanup
scripts/activate.sh:unset:103: no such hash table element: _pw_hello

mrcao2011, 2022-04-29:
your python version is wrong，what is your python3 version?my python3 --version Python 3.9.6, it work ok

rafaliya-amazon, 2022-04-29:
Previously, my python version was 3.8.9 & then I even updated the version to 3.9.6. 
However, the script is still failing.

rafaliya-amazon, 2022-04-29:
Thanks a lot for the help. I was able to run the script and even ran all the tests. 

I followed this: https://stackoverflow.com/questions/50236117/scraping-ssl-certificate-verify-failed-error-for-http-en-wikipedia-org

Please do the following:  

     Application << Python 3.9.6 << Run "Install Certificates.command"

closed 2022-04-29 by rafaliya-amazon