project-chip/connectedhomeip #17904: [TV-APP]While verifying the "wild card" read for all the cluster for all the attributes, chiptool responds as "General error: 0x01 (FAILURE)" in TV-APP

Rajashreekalmane, 2022-04-29:
#### Problem
In tv-app While verifying the wildcard read using the below command, towards the end, Chiptool throws an error with a code 
 "General error: 0x01 (FAILURE)" 
 
* expected behavior:-Success(0x00) 
 * actual behavior:-FAILURE(0x01)

 * steps to reproduce:- sudo ./chip-tool any  read-by-id 0xFFFFFFFF 0xFFFFFFFF 1 0xFFFF

system configuration
Chip-Tool: RPI
DUT: RPI
Platform : IP and Wifi
Commit-ID: https://github.com/project-chip/connectedhomeip/commit/2982c0ca0623773327def4cc7180d07f6b959c42
Dut: tv-app
 
PFA screenshot:-
[image]

Spec reference:-
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/data_model/Interaction-Model.adoc#valid-read-attribute-paths

Rajashreekalmane, 2022-04-29:
Problem:
While verifying the "wild card" read for all the cluster for all the attributes for specific endpoint(0), chiptool responds as Invalid Cluster-Specific error code 0x11.

steps to reproduce:-Execution Command:-./chip-tool any read-by-id 0xFFFFFFFF 0xFFFFFFFF 1 0

PFA screenshot below:-
[image]

chrisdecenzo, 2022-05-13:
this problem is no longer happening for me. It might have been a bug fix in the chip tool

closed 2022-05-13 by chrisdecenzo