project-chip/connectedhomeip #17937: Python chip-device-ctl crashing

agners, 2022-04-30:
#### Problem
<what's wrong or missing, please include any applicable:
 * expected behavior
 * actual behavior
 * steps to reproduce
 * system configuration
 * screenshots, images, logs, etc.
 >
Followed the guide at `docs/guides/python_chip_controller_building.md` (using Debian 11 instead of Debian 10, since at least Python 3.8 seems to be required nowadays) to build the Python CHIP Controller.

I am building in a Container. I tried running the container with and without `--privileged`, same result. I also have similar issues on my native host (Arch Linux).

However, Python CHIP Controller seems to crash on startup:

```
[1651306836.729084][40566:40570] CHIP:DL: Platform main loop started.                                                                                                                                                                                                                                                                                                                                                         
2022-04-30 08:20:36 ccf35f435144 root[40566] ERROR [Errno 2] No such file or directory: '/tmp/chip-device-ctrl-storage.json'                                                                                                                                                                                                                                                                                                  
2022-04-30 08:20:36 ccf35f435144 root[40566] WARNING Could not load configuration from /tmp/chip-device-ctrl-storage.json - resetting configuration...                                                                                                                                                                                                                                                                        
2022-04-30 08:20:36 ccf35f435144 root[40566] INFO SetSdkKey: g/gcc = b'\xe8\x03\x00\x00'                                                                                                                                                                                                                                                                                                                                      
2022-04-30 08:20:36 ccf35f435144 root[40566] INFO SetSdkKey: g/gdc = b'\xe8\x03\x00\x00'                                                                                                                                                                                                                                                                                                                                      
2022-04-30 08:20:36 ccf35f435144 chip.ZCL[40566] INFO Using ZAP configuration...                                                                                                                                                                                                                                                                                                                                              
CHIP Error 0x000000AD: Open file failed                                                                                                                                                                                                                                                                                                                                                                                       
Failed to bringup CHIPDeviceController CLI                                                                                                                                                                                                                                                                                                                                                                                    
Exception ignored in: <function PersistentStorage.__del__ at 0x7f15d5428160>                                                                                                                                                                                                                                                                                                                                                  
Traceback (most recent call last):                                                                                                                                                                                                                                                                                                                                                                                            
  File "/connectedhomeip/out/python_env/lib/python3.9/site-packages/chip/storage/__init__.py", line 156, in __del__                                                                                                                                                                                                                                                                                                           
    builtins.chipStack.Call(                                                                                                                                                                                                                                                                                                                                                                                                  
AttributeError: module 'builtins' has no attribute 'chipStack'                                                                                                                                                                                                                                                                                                                                                                
Aborted (core dumped)
```

Using gdb to get a stack trace leads to the following:

```
Starting program: /connectedhomeip/out/python_env/bin/python3 ./out/python_env/bin/chip-device-ctrl
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[1651307861.863674][150:150] CHIP:CTL: Setting attestation nonce to random value
[1651307861.863739][150:150] CHIP:CTL: Setting CSR nonce to random value
[New Thread 0x7ffff36b3700 (LWP 154)]
[New Thread 0x7ffff2eb2700 (LWP 155)]
[New Thread 0x7ffff26b1700 (LWP 156)]
[1651307861.864724][150:150] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-9cMBkx)
[1651307861.864832][150:150] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1651307861.864839][150:150] CHIP:DL: NVS set: chip-counters/reboot-count = 5 (0x5)
[1651307861.864945][150:150] CHIP:DL: Got Ethernet interface: eth0
[1651307861.864975][150:150] CHIP:DL: Found the primary Ethernet interface:eth0
[1651307861.865011][150:150] CHIP:DL: Failed to get WiFi interface
[1651307861.865014][150:150] CHIP:DL: Failed to reset WiFi statistic counts
[New Thread 0x7ffff1eb0700 (LWP 157)]
[1651307861.865093][150:157] CHIP:DL: Platform main loop started.
2022-04-30 08:37:41 0ce5d83bb951 root[150] CRITICAL Loading configuration from /tmp/chip-device-ctrl-storage.json...
2022-04-30 08:37:41 0ce5d83bb951 root[150] INFO SetSdkKey: g/gcc = b'\xa0\x0f\x00\x00'
2022-04-30 08:37:41 0ce5d83bb951 root[150] INFO SetSdkKey: g/gdc = b'\xa0\x0f\x00\x00'
2022-04-30 08:37:41 0ce5d83bb951 chip.ZCL[150] INFO Using ZAP configuration...
CHIP Error 0x000000AD: Open file failed
Failed to bringup CHIPDeviceController CLI
Exception ignored in: <function PersistentStorage.__del__ at 0x7ffff5e831f0>
Traceback (most recent call last):
  File "/connectedhomeip/out/python_env/lib/python3.9/site-packages/chip/storage/__init__.py", line 156, in __del__
    builtins.chipStack.Call(
AttributeError: module 'builtins' has no attribute 'chipStack'

Thread 1 "python3" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff7c65537 in __GI_abort () at abort.c:79
#2  0x00007ffff44de61c in chipAbort() () at ../../src/lib/support/CodeUtils.h:489
[... cut ...]
#12 0x000000000060d4e3 in Py_RunMain ()
#13 0x00000000005ea6e9 in Py_BytesMain ()
#14 0x00007ffff7c66d0a in __libc_start_main (main=0x5ea6b0, argc=2, argv=0x7fffffffec48, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffec38) at ../csu/libc-start.c:308
#15 0x00000000005ea5ea in _start ()
```

I tried a older commit from about 2 weeks ago, with a similar result. Is there a known working version of the Python CHIP Controller? How can the Python CHIP controller be used?

 #### Proposed Solution

agners, 2022-04-30:
Crash using ArchLinux with Python 3.9

```
2022-04-30 10:44:54 allenwind chip.ZCL[256359] INFO Using ZAP configuration...
2022-04-30 10:44:54 allenwind chip.DL[256359] INFO Avahi re-register required
2022-04-30 10:44:54 allenwind chip.IN[256359] INFO CASE Server enabling CASE session setups
2022-04-30 10:44:54 allenwind chip.DIS[256359] INFO Updating services using commissioning mode 0
2022-04-30 10:44:54 allenwind chip.DIS[256359] ERROR Failed to initialize advertiser: ../../src/platform/Linux/DnssdImpl.cpp:333: CHIP Error 0x00000003: Incorrect state
2022-04-30 10:44:54 allenwind chip.DIS[256359] INFO Advertise commission parameter vendorID=65521 productID=32769 discriminator=3840/15
2022-04-30 10:44:54 allenwind chip.DIS[256359] ERROR Failed to advertise commissioner: ../../src/lib/dnssd/Discovery_ImplPlatform.cpp:479: CHIP Error 0x00000003: Incorrect state
2022-04-30 10:44:54 allenwind chip.DIS[256359] INFO Delaying proxy of operational discovery: missing delegate
New FabricAdmin: FabricId: 1(1)
2022-04-30 10:44:54 allenwind root[256359] INFO SetReplKey: fabricAdmins = {'1': {'fabricId': 1}}
name 'path' is not defined
Failed to bringup CHIPDeviceController CLI
Exception ignored in: <function FabricAdmin.__del__ at 0x7ff94681cee0>
Traceback (most recent call last):
  File "/home/sag/projects/project-chip/connectedhomeip/out/python_env/lib/python3.9/site-packages/chip/FabricAdmin.py", line 209, in __del__
    self.Shutdown(False)
  File "/home/sag/projects/project-chip/connectedhomeip/out/python_env/lib/python3.9/site-packages/chip/FabricAdmin.py", line 188, in Shutdown
    if (self._isActive):
AttributeError: 'FabricAdmin' object has no attribute '_isActive'
[1]    256358 segmentation fault  sudo out/python_env/bin/chip-device-ctrl
```

agners, 2022-05-03:
It seems that the minimal mDNS implementation works much better, e.g. using

```
scripts/build_python.sh -m minimal
```

instead of

```
scripts/build_python.sh -m platform
```

Is the platform mDNS implementation (Avahi in Linux' case) not supported anymore?

closed 2022-11-12 by stale[bot]

Firesuiry, 2022-11-24:
same problem