project-chip/connectedhomeip #17941: OTA-R and Chip-tool Linux applications crashes when AnnounceOTAProvider command is run after event subscription

isiu-apple, 2022-04-30:
#### Problem

1. Launch and Commission OTA-Provider and OTA-Requestor reference Linux apps
2. Subscribe events on OTA-R Linux app using chip-tool
3. Initiate query using AnnounceOTAProvider command from chip-tool

OTA-R crashes with:

```
1651290483036] [18041:322992] CHIP: [DMG] Fetched 0 events
[1651290483036] [18041:322992] CHIP: [DMG] <RE> Sending report (payload has 15 bytes)...
[1651290483036] [18041:322992] CHIP: [-] ../../../examples/ota-requestor-app/linux/third_party/connectedhomeip/src/app/ReadHandler.cpp:241: CHIP Error 0x00000003: Incorrect state at ../../../examples/ota-requestor-app/linux/third_party/connectedhomeip/src
[1651290483036] [18041:322992] CHIP: [DMG] <RE> Error sending out report data with ../../../examples/ota-requestor-app/linux/third_party/connectedhomeip/src/app/ReadHandler.cpp:241: CHIP Error 0x00000003: Incorrect state!
```

Chip-tool crashes with:

```
[1651290475776] [18050:323522] CHIP: [DMG] Received Command Response Status for Endpoint=0 Cluster=0x0000_002A Command=0x0000_0000 Status=0x7e
[1651290475776] [18050:323522] CHIP: [TOO] Error: IM Error 0x0000057E: General error: 0x7e (UNSUPPORTED_ACCESS)
[1651290475776] [18050:323522] CHIP: [DMG] ICR moving to [AwaitingDe]
[1651290475776] [18050:323522] CHIP: [EM] Sending Standalone Ack for MessageCounter:14490838 on exchange 32362i
[1651290475776] [18050:323522] CHIP: [IN] Prepared secure message 0x700000acf4e0 to 0x0000001234567890 (1)  of type 0x10 and protocolId (0, 0) on exchange 32362i with MessageCounter:1347447.
[1651290475776] [18050:323522] CHIP: [IN] Sending encrypted msg 0x700000acf4e0 with MessageCounter:1347447 to 0x0000001234567890 (1) at monotonic time: 00000000026C5804 msec
[1651290475776] [18050:323522] CHIP: [EM] Flushed pending ack for MessageCounter:14490838 on exchange 32362i
[1651290475776] [18050:323516] CHIP: [CTL] Shutting down the commissioner
[1651290475776] [18050:323516] CHIP: [CTL] Shutting down the controller
[1651290475776] [18050:323516] CHIP: [CTL] Shutting down the commissioner
[1651290475776] [18050:323516] CHIP: [CTL] Shutting down the controller
[1651290475776] [18050:323516] CHIP: [IN] Expiring all connections for fabric 1!!
[1651290475776] [18050:323516] CHIP: [CTL] Shutting down the commissioner
[1651290475776] [18050:323516] CHIP: [CTL] Shutting down the controller
[1651290475776] [18050:323516] CHIP: [IN] Expiring all connections for fabric 2!!
[1651290475776] [18050:323516] CHIP: [CTL] Shutting down the commissioner
[1651290475776] [18050:323516] CHIP: [CTL] Shutting down the controller
[1651290475776] [18050:323516] CHIP: [IN] Expiring all connections for fabric 3!!
[1651290475776] [18050:323516] CHIP: [CTL] Shutting down the System State, this will teardown the CHIP Stack
[1651290475776] [18050:323516] CHIP: [DMG] IM WH moving to [Uninitialized]
[1651290475776] [18050:323516] CHIP: [DMG] IM WH moving to [Uninitialized]
[1651290475776] [18050:323516] CHIP: [DMG] IM WH moving to [Uninitialized]
[1651290475776] [18050:323516] CHIP: [DMG] IM WH moving to [Uninitialized]
[1651290475777] [18050:323516] CHIP: [DMG] All ReadHandler-s are clean, clear GlobalDirtySet
[1651290475777] [18050:323516] CHIP: [BLE] CancelConnection
[1651290475777] [18050:323516] CHIP: [DL] Inet Layer shutdown
[1651290475777] [18050:323516] CHIP: [DL] BLE shutdown
[1651290475777] [18050:323516] CHIP: [DL] System Layer shutdown
[1651290475777] [18050:323516] CHIP: [TOO] Run command failure: IM Error 0x0000057E: General error: 0x7e (UNSUPPORTED_ACCESS)
```

 #### Proposed Solution
<suggested fix, suggested enhancement>

vivien-apple, 2022-05-03:
It works for me on ToT with the following steps:
```
$ chip-ota-provider-app -f ~/Downloads/test2.ota --passcode 20202021 --discriminator 3840 --KVS /tmp/chip_kvs_provider
$ chip-ota-requestor-app --discriminator 30 --secured-device-port 5560 --KVS /tmp/chip_kvs_requestor
$ chip-tool interactive start
```

Once in interactive mode:
```
$ pairing onnetwork 0xABCD 20202021
$ pairing onnetwork-long 0x1234567890 20202021 30
# Workaround ACL permissions
$ accesscontrol write acl [{"fabricIndex":1,"privilege":5,"authMode":2,"subjects":[112233],"targets":null},{"fabricIndex":1,"privilege":3,"authMode":2,"subjects":null,"targets":null}] 0xABCD 0
$ subscribe-event state-transition 5 10 0x1234567890 0
$ otasoftwareupdaterequestor announce-ota-provider 0x00000000ABCD 0 0 0 0x0000001234567890 0
```

I wonder if the `ReadClient` failure that is seen when not everything is launched from the interactive mode is related to the fact that `chip-tool` is running in interactive mode on one side, and then is running in standalone mode when the command `otasoftwareupdaterequestor announce-ota-provider 0x00000000ABCD 0 0 0 0x0000001234567890 0` is issued.

isiu-apple, 2022-05-03:
I discussed this issue with @bzbarsky-apple, who pointed out that this issue may have been introduced by this PR: https://github.com/project-chip/connectedhomeip/pull/17779

The `ReadHandler.cpp:241: CHIP Error 0x00000003: Incorrect state!` error from OTA-R is due to `mSessionHandle` becoming NULL in `ReadHandler::SendReportData() `when a sanity check is done before use.  When the AnnounceOTAProvider command is run from the second instance of chip-tool in standalone mode, it causes the previous session from the same node to be killed, and thus NULLed by the destructor.

bzbarsky-apple, 2022-07-01:
@isiu-apple #17568 has been fixed.  Could you retest this, please?

closed 2023-05-08 by stale[bot]