project-chip/connectedhomeip #17947: Fail Installation on Ubuntu 22.04 LTS

ashegheparyshun, 2022-05-02:
#### Problem
I've tried Matter SDK on Ubuntu 20.04 LTS and Docker with no problem. Now I've switched to Ubuntu 22.04 LTS. I applied the same installation process to build Matter SDK but the installation failed. The first mismatch result to Ubuntu 20.04 LTS is this line:
`sudo apt-get install git gcc g++ python pkg-config libssl-dev libdbus-1-dev \
     libglib2.0-dev libavahi-client-dev ninja-build python3-venv python3-dev \
     python3-pip unzip libgirepository1.0-dev libcairo2-dev`
Which it successes on Ubuntu 20 but fails on Ubuntu 22 the result is 
 
[image]
Then I tried 
`sudo apt-get install git gcc g++ python3 pkg-config libssl-dev libdbus-1-dev \
     libglib2.0-dev libavahi-client-dev ninja-build python3-venv python3-dev \
     python3-pip unzip libgirepository1.0-dev libcairo2-dev`
Note that, I changed python to python3 at the first line.
After this change the installation finished successfully. Then I've followed the installation according to [https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/BUILDING.md](url) until
`source scripts/activate.sh`
But I got this error
[image]

I tried python3-is-python
`sudo apt install python-is-python3`
But the same result.
 System information:
OS: Ubuntu 22.04 LTS
CPU: Intel(R) Core(TM) i7-4702MQ CPU @ 2.20GHz
RAM: 8G

rosahay-silabs, 2022-05-03:
Duplicate of https://github.com/project-chip/connectedhomeip/issues/16664

mikaelhm, 2022-07-29:
This can be closed, verified no longer an issue.

franck-apple, 2022-07-29:
Closing per above comment.

closed 2022-07-29 by franck-apple