project-chip/connectedhomeip #17948: [TC-DM-2.2], [TC-DM-2.4] - Need a signal to trigger the mandatory attribute SHUTDOWN event

sowmyassp, 2022-05-02:
**Issue** : Results  doesn't match with test-plan that is the when I pass shut-down event which is mandatory and should have  priority set as CRITICAL for this to happen we need a signal to trigger the shut-down event.

**Expected behavior** : ShutDown event has priority set as CRITICAL

**Actual behavior** : Didn't receive any priority

**Steps to reproduce** : ./chip-tool basic read-event shut-down 1 0

**Screenshot** : 
[image]

**Commit used** : 5df932897db4658504309446322ad09cbf7aaf2c
[TC-DM-2.2_cntrl.txt](https://github.com/project-chip/connectedhomeip/files/8602340/TC-DM-2.2_cntrl.txt)

**System configuration** : DUT - RPI
                                          Chip-tool - RPI

**PFA Logs** :

vivien-apple, 2022-05-18:
Looking at the log, the command used is `basic read-event shut-down 1 0` but this can not work.

In order to ge the shutdown event, your only chance is to subscribe to it. So you need to:

Launch `chip-tool` into interactive mode with the command `chip-tool interactive start`

Then once in interactive mode:
```
basic subscribe-event shut-down 1 3 1 0
```

If you now turn off the DUT you should see something like
```
[1652908731239] [73677:5468727] CHIP: [EM] Handling via exchange: 35551r, Delegate: 0x10aa4c608
[1652908731239] [73677:5468727] CHIP: [DMG] ReportDataMessage =
[1652908731239] [73677:5468727] CHIP: [DMG] {
[1652908731239] [73677:5468727] CHIP: [DMG] 	SubscriptionId = 0xd24b6235,
[1652908731239] [73677:5468727] CHIP: [DMG] 	EventReportIBs =
[1652908731239] [73677:5468727] CHIP: [DMG] 	[
[1652908731239] [73677:5468727] CHIP: [DMG] 		EventReportIB =
[1652908731239] [73677:5468727] CHIP: [DMG] 		{
[1652908731239] [73677:5468727] CHIP: [DMG] 			EventDataIB =
[1652908731239] [73677:5468727] CHIP: [DMG] 			{
[1652908731239] [73677:5468727] CHIP: [DMG] 				EventPath =
[1652908731239] [73677:5468727] CHIP: [DMG] 				{
[1652908731239] [73677:5468727] CHIP: [DMG] 					Endpoint = 0x0,
[1652908731240] [73677:5468727] CHIP: [DMG] 					Cluster = 0x28,
[1652908731240] [73677:5468727] CHIP: [DMG] 					Event = 0x1,
[1652908731240] [73677:5468727] CHIP: [DMG] 				},
[1652908731240] [73677:5468727] CHIP: [DMG] 				
[1652908731240] [73677:5468727] CHIP: [DMG] 				EventNumber = 0x40006,
[1652908731240] [73677:5468727] CHIP: [DMG] 				PriorityLevel = 0x2,
[1652908731240] [73677:5468727] CHIP: [DMG] 				SystemTimestamp = 0x2a720af7,
[1652908731240] [73677:5468727] CHIP: [DMG] 				EventData = 
[1652908731240] [73677:5468727] CHIP: [DMG] 				{
[1652908731240] [73677:5468727] CHIP: [DMG] 				},
[1652908731240] [73677:5468727] CHIP: [DMG] 			},
[1652908731240] [73677:5468727] CHIP: [DMG] 			
[1652908731240] [73677:5468727] CHIP: [DMG] 		},
[1652908731240] [73677:5468727] CHIP: [DMG] 		
[1652908731240] [73677:5468727] CHIP: [DMG] 	],
[1652908731240] [73677:5468727] CHIP: [DMG] 	
[1652908731240] [73677:5468727] CHIP: [DMG] 	InteractionModelRevision = 1
[1652908731240] [73677:5468727] CHIP: [DMG] }
[1652908731241] [73677:5468727] CHIP: [TOO] Endpoint: 0 Cluster: 0x0000_0028 Event 0x0000_0001
[1652908731241] [73677:5468727] CHIP: [TOO] 	 Event number: 262150
[1652908731241] [73677:5468727] CHIP: [TOO] 	 Priority: Critical
[1652908731241] [73677:5468727] CHIP: [TOO] 	 Timestamp: 712116983
[1652908731242] [73677:5468727] CHIP: [TOO]   ShutDown: {
[1652908731242] [73677:5468727] CHIP: [TOO]    }

``

sowmyassp, 2022-05-20:
Thank you @vivien-apple, its working 
Similarly for leave event also we have to follow the same procedure?

woody-apple, 2022-05-23:
Cert Blocker Review: Given this is can now be resolved in an updated test plan, closing.