project-chip/connectedhomeip #17950: [TC-IDM-1.2]DUT fails to responds has UNSUPPORTED_ACCESS, When TH sends invoke message to DUT that it should not have the privilege for the cluster.

Rajashreekalmane, 2022-05-02:
#### Problem
As per Test-Plan, test step 4 says "Setup the TH,  to send invoke request message from TH to DUT such that it should not have the privilege for the cluster" DUT should responds with UNSUPPORTED_ACCESS to TH ,but  DUT fails to responds has UNSUPPORTED_ACCESS.

 * **expected behavior:**-UNSUPPORTED_ACCESS(0x7e)
 * **actual behavior:-**
 On TH---Run command failure: ../../src/app/ReadClient.cpp:595: CHIP Error 0x00000032: Timeout
On DUT---Secure transport received message, but failed to decode/authenticate it, discarding
 
* steps to reproduce:-./chip-tool basic read vendor-name  1 0 --commissioner-nodeid 332211

**system configuration**
Chip-Tool: RPI
DUT: RPI
Platform : IP and Wifi
Commit-ID: 5df932897db4658504309446322ad09cbf7aaf2c
Dut:All-cluster-app

**PFA log:-**
[IDM-1.2-step4.txt](https://github.com/project-chip/connectedhomeip/files/8602853/IDM-1.2-step4.txt)

**Test-Plan Reference:-**
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/interactiondatamodel.adoc#tc-idm-1-2-invoke-response-action-from-dut-to-th

**Spec-Reference:-**
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/data_model/Interaction-Model.adoc#status-codes

Rajashreekalmane, 2022-05-02:
Facing the same issue in TC-IDM-2.2(test-step 21) , TC-IDM 3.2(Test-step 18), TC-IDM-6.2(test-step-14).

mlepage-google, 2022-05-02:
Need details on what command is trying to run. This generally works, even in chip-tool.

mlepage-google, 2022-05-02:
This seems like a test setup / environment issue.

Rajashreekalmane, 2022-05-03:
@mlepage-google 
The command I tried ./chip-tool basic read vendor-name 1 0 --commissioner-nodeid 332211

mlepage-google, 2022-05-03:
Right, but need more details on what the DUT app is, and how it was commissioned. Do you have a sequence of commands which demonstrates the failure?

Rajashreekalmane, 2022-05-04:
@mlepage-google we are using Dut as All-cluster-app and I tried in both WIFI & IP commission.
here I attached sequence of commands before this step.
[IDM 1.2.txt](https://github.com/project-chip/connectedhomeip/files/8621018/IDM.1.2.txt)

mlepage-google, 2022-05-04:
OK I tried the same thing here having built master, on Linux x64, and I get the same result. This seems like a problem with that `--commissioner-nodeid` argument to chip-tool. The message is not properly getting through.

mlepage-google, 2022-05-04:
This is due to an issue with chip-tool and the credentials it uses when the nodeid is changed like this, compared to the sessions etc. it has stored. It needs to be improved to handle that better.

[Issue #17221](https://github.com/project-chip/connectedhomeip/issues/17221)

vivien-apple, 2022-05-18:
See comment https://github.com/project-chip/connectedhomeip/issues/18413#issuecomment-1129998162

I suggest manipulating the ACL since the `commissioner-nodeid` behaviour seems to have changed Let me know if that helps....

Rajashreekalmane, 2022-05-19:
Thank you  @vivien-apple getting expected behavior.

closed 2022-05-19 by Rajashreekalmane