project-chip/connectedhomeip #17970: [TC-PSCFG-1.1] Attribute list showing value mismatch error

manjunath-grl, 2022-05-02:
Summary:
As per testplan of TC-PSCFG-1.1 attributelist exepected value value: [0, 65528, 65529, 65530, 65531, 65532, 65533] but got 
value: [0, 65528, 65529, 65531, 65532, 65533] mandatory attribute 65530 is missing
 
Testplan screenshot:
[image]

Execution logs:
[PSCFG_1_1_AL.txt](https://github.com/project-chip/connectedhomeip/files/8605584/PSCFG_1_1_AL.txt)
[image]

bzbarsky-apple, 2022-05-21:
65530 == 0xfffa is EventList.

bzbarsky-apple, 2022-05-21:
Duplicate of #15011

closed 2022-05-21 by bzbarsky-apple