project-chip/connectedhomeip #17981: Commissioning using Python CHIP Controller fails at AttestationVerification

agners, 2022-05-02:
#### Problem
<what's wrong or missing, please include any applicable:
 * expected behavior
 * actual behavior
 * steps to reproduce
 * system configuration
 * screenshots, images, logs, etc.
 >
When trying to commission a device using the Python CHIP Controller (as of today master branch), with a nRF52840 running `lighting-app` example (built from master branch as well), the Python CHIP Controller fails to commission the device completely:
```
...
2022-05-03 00:21:31 local-chip-controller chip.DMG[209] INFO Received Command Response Data, Endpoint=0 Cluster=0x0000_003E Command=0x0000_0001
2022-05-03 00:21:31 local-chip-controller chip.CTL[209] INFO Received Attestation Information from the device
2022-05-03 00:21:31 local-chip-controller chip.CTL[209] INFO Successfully finished commissioning step 'SendAttestationRequest'
2022-05-03 00:21:31 local-chip-controller chip.CTL[209] INFO Commissioning stage next step: 'SendAttestationRequest' -> 'AttestationVerification'
2022-05-03 00:21:31 local-chip-controller chip.CTL[209] INFO Performing next commissioning step 'AttestationVerification'
2022-05-03 00:21:31 local-chip-controller chip.CTL[209] INFO Verifying attestation
2022-05-03 00:21:31 local-chip-controller chip.CTL[209] ERROR Failed in verifying 'Attestation Information' command received from the device: err 101. Look at AttestationVerificationResult enum to understand the errors
2022-05-03 00:21:31 local-chip-controller chip.CTL[209] INFO Error on commissioning step 'AttestationVerification': '../../src/controller/CHIPDeviceController.cpp:945: CHIP Error 0x000000AC: Internal error'
2022-05-03 00:21:31 local-chip-controller chip.CTL[209] ERROR Failed to perform commissioning step 8
2022-05-03 00:21:31 local-chip-controller chip.CTL[209] INFO Going from commissioning step 'AttestationVerification' with lastErr = '../../src/controller/CHIPDeviceController.cpp:945: CHIP Error 0x000000AC: Internal error' -> 'Cleanup'
2022-05-03 00:21:31 local-chip-controller chip.CTL[209] INFO Performing next commissioning step 'Cleanup' with completion status = '../../src/controller/CHIPDeviceController.cpp:945: CHIP Error 0x000000AC: Internal error'
2022-05-03 00:21:31 local-chip-controller chip.CTL[209] INFO Expiring failsafe on proxy 0x7f035c00c360
2022-05-03 00:21:31 local-chip-controller chip.IN[209] INFO Prepared secure message 0x7f036f7f83b8 to 0xFFFFFFFB00000000 (0)  of type 0x8 and protocolId (0, 1) on exchange 13482i with MessageCounter:6677141.
2022-05-03 00:21:31 local-chip-controller chip.IN[209] INFO Sending encrypted msg 0x7f036f7f83b8 with MessageCounter:6677141 to 0xFFFFFFFB00000000 (0) at monotonic time: 00000000002A6D51 msec
2022-05-03 00:21:31 local-chip-controller chip.EM[209] INFO Received message of type 0x9 with protocolId (0, 1) and MessageCounter:2424698 on exchange 13482i
2022-05-03 00:21:31 local-chip-controller chip.DMG[209] INFO Received Command Response Data, Endpoint=0 Cluster=0x0000_0030 Command=0x0000_0001
2022-05-03 00:21:31 local-chip-controller chip.CTL[209] INFO Failsafe disarmed
2022-05-03 00:21:31 local-chip-controller chip.CTL[209] INFO Successfully finished commissioning step 'Cleanup'
```

Python CHIP Controller is running on Debian bullseye.

Any hint what this could be?

 #### Proposed Solution
<suggested fix, suggested enhancement>

agners, 2022-05-03:
It turns out this was due to a missing trust store (`credentials` directory). The error really doesn't have a good indication of that fact. As a hint I added #17996 which reports the path used at startup.

zhouF96, 2022-06-01:
Hi @agners  We still have the same issue while using Python CHIP Controller and the device is lighting-app esp32. Both TE9 and Master failed. Any idea?

closed 2022-12-11 by stale[bot]