project-chip/connectedhomeip #17994: [TC-IDM-2.1] and [TC-IDM-2.2]-DUT "fails" to send a report data action with the attribute value from all the clusters and endpoints to the TH.

Rajashreekalmane, 2022-05-03:
#### Problem
When TH sends the Read Request Message to the DUT to read a specific attribute from all endpoints and all clusters.
AttributePath = [[ Attribute = specific attribute]] On receipt of this message, DUT should send a report data action with the attribute value to the TH. But Currently DUT "fails" send a report data action with the attribute value from all the clusters and endpoints to the TH.

 * actual behavior:- 0x80 (INVALID_ACTION)
 * expected behavior:-On the TH, DUT should send report data message has all the right attribute values.

 * steps to reproduce:-sudo ./chip-tool any read-by-id 0xFFFFFFFF 0x0 1 0xFFFF

system configuration:-
Chip-Tool: RPI
DUT: RPI
Platform : IP and Wifi
Commit-ID:4d50b1084284018c4a628f919b0d2e01ea2a39f8
Dut: all-cluster-app

PFA log:-
[[TC-IDM-2.1].txt](https://github.com/project-chip/connectedhomeip/files/8609810/TC-IDM-2.1.txt)

erjiaqing, 2022-05-11:
This is a expected behavior, 0xFFFFFFFF is not a valid cluster id, and 0xFFFF is not a valid endpoint id.

Rajashreekalmane, 2022-05-12:
@erjiaqing which is valid cluster ID and endpoint to read the attribute value from all the clusters and endpoints  @cjandhyala

erjiaqing, 2022-05-12:
We don't have a valid cluster id and endpoint id for all the clusters and endpoints.

If you mean wildcard paths, @vivien-apple seems we need the ability to send read requests to wildcard paths?

cjandhyala, 2022-05-12:
@Rajashreekalmane per our conversation pls close the ticket

closed 2022-05-13 by Rajashreekalmane