project-chip/connectedhomeip #17995: [TC-IDM-2.1] and [TC-IDM-2.2]--TH "fails" to send a report data action with the attribute value to the DUT with the attribute value from all the clusters for specific endpoint.

Rajashreekalmane, 2022-05-03:
#### Problem
When TH sends the Read Request Message to the DUT to read a specific attribute and specific endpoints from all clusters.
On receipt of this message, DUT should send a report data action with the attribute value to the TH. But Currently DUT "fails" send a report data action with the attribute value and endpoints from all the clusters to the TH.

    actual behavior:- 0x80 (INVALID_ACTION)

    expected behavior:-On the TH, DUT should send report data message has all the right attribute values.

    steps to reproduce:-
sudo ./chip-tool any read-by-id 0xFFFFFFFF 0x0 1 0
sudo ./chip-tool any read-by-id 0xFFFFFFFF 0x0 1 1

system configuration:-
Chip-Tool: RPI
DUT: RPI
Platform : IP and Wifi
Commit-ID:4d50b1084284018c4a628f919b0d2e01ea2a39f8
Dut: all-cluster-app

PFA log:-
[[TC-IDM-2.2].txt](https://github.com/project-chip/connectedhomeip/files/8609902/TC-IDM-2.2.txt)

Test-Plan reference
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/interactiondatamodel.adoc#tc-idm-2-1-read-request-action-from-dut-to-th-dut-controller

erjiaqing, 2022-05-11:
This is a expected behavior, 0xFFFFFFFF is not a valid cluster id

franck-apple, 2022-05-13:
relates to #17994, closing.

closed 2022-05-13 by franck-apple