project-chip/connectedhomeip #18013: Memory leaks in when DeviceControllerFactory::InitSystemState fails

msandstedt, 2022-05-03:
#### Problem

DeviceControllerFactory::InitSystemState allocates memory here:

https://github.com/project-chip/connectedhomeip/blob/1dc522db9d2670c967fcc90261d26b72d9f15e11/src/controller/CHIPDeviceControllerFactory.cpp#L150

These objects are freed in DeviceControllerSystemState::Shutdown, here:

https://github.com/project-chip/connectedhomeip/blob/1dc522db9d2670c967fcc90261d26b72d9f15e11/src/controller/CHIPDeviceControllerFactory.cpp#L311

However, this can't happen unless DeviceControllerFactory::InitSystemState succeeds and passes ownership of the allocated objects to DeviceControllerSystemState, here:

https://github.com/project-chip/connectedhomeip/blob/1dc522db9d2670c967fcc90261d26b72d9f15e11/src/controller/CHIPDeviceControllerFactory.cpp#L236

This means that all of these early error returns are leaking memory:

https://github.com/project-chip/connectedhomeip/blob/1dc522db9d2670c967fcc90261d26b72d9f15e11/src/controller/CHIPDeviceControllerFactory.cpp#L158
https://github.com/project-chip/connectedhomeip/blob/1dc522db9d2670c967fcc90261d26b72d9f15e11/src/controller/CHIPDeviceControllerFactory.cpp#L159
... (many more in between)
https://github.com/project-chip/connectedhomeip/blob/1dc522db9d2670c967fcc90261d26b72d9f15e11/src/controller/CHIPDeviceControllerFactory.cpp#L233

 #### Proposed Solution

The manual memory management here is concerning.  It would probably be simpler for DeviceControllerSystemState to just have all of the allocated objects contained within.  If this is not feasible, we might consider giving `DeviceControllerSystemStateParams` some ability to clean itself up, or perhaps the use of Shared pointers when passing between `DeviceControllerSystemStateParams` and `DeviceControllerSystemState`.

msandstedt, 2022-05-03:
CC @kghost , @bzbarsky-apple , @wbschiller

bzbarsky-apple, 2022-05-04:
Duplicate of https://github.com/project-chip/connectedhomeip/issues/16231?

msandstedt, 2022-05-04:
@bzbarsky-apple ,

Yes it is a duplicate of #16231.  Thanks for letting me know.

closed 2022-05-04 by msandstedt