project-chip/connectedhomeip #18028: [TC-GC-1.1]General commissioning cluster---Value mismatch for FeatureMap attribute

Rajashreekalmane, 2022-05-04:
#### Problem
As per test plan [TC-UL-1.1] FeatureMap attribute showing value mismatch error when executed, DUT should
responds with feature map value 0 but currently DUT responding feature map value has 6.

- expected behavior:-FeatureMap attribute value should be bit 0.

- actual behavior:- DUT responds has bit 6

-  steps to reproduce:-./chip-tool generalcommissioning read feature-map 1 0

- system configuration:-
Chip-Tool: RPI
DUT: RPI
Platform : IP and Wifi
Commit-ID:4d50b1084284018c4a628f919b0d2e01ea2a39f8
Dut: all-cluster-app

PFA log
[TC-GC-1.1.txt](https://github.com/project-chip/connectedhomeip/files/8618110/TC-GC-1.1.txt)

TestPlan reference:-
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/General_Commissioning.adoc#tc-gc-1-1-global-attributes-dut-commissionee

bzbarsky-apple, 2022-05-09:
@Rajashreekalmane This is using all-clusters-app as the DUT?

bzbarsky-apple, 2022-05-09:
It looks like this changed in #17840.  @fessehaeve could you please take a look?  Is this just a bad merge on the ZAP file, so the feature map default for the wrong cluster ended up getting changed?

Rajashreekalmane, 2022-05-09:
@bzbarsky-apple Yes I am using DUT as all-clusters-app.

fessehaeve, 2022-05-10:
> It looks like this changed in #17840. @fessehaeve could you please take a look? Is this just a bad merge on the ZAP file, so the feature map default for the wrong cluster ended up getting changed?

@bzbarsky-apple  It must be a bad merge since I did nothing related to General commissioning cluster. The feature-map value was already 6 in the previous state https://github.com/project-chip/connectedhomeip/commit/a9656d5e6fe29e1e900088610299caf250bc83cd from @yufengwch .

bzbarsky-apple, 2022-05-10:
Good catch.  @yufengwangca Do you recall why it's 6 there?

bzbarsky-apple, 2022-05-17:
The _General Commissioning_ cluster has no features defined at all in the spec, so its FeatureMap should always be 0 and the test is correct....

cecille, 2022-05-17:
Ah, you're correct. I did until https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/4956. So that's where the 6 came from. It was a spec change without a corresponding SDK change.

bzbarsky-apple, 2022-05-17:
Ah, great catch.  I'll just put up a PR, then.

closed 2022-05-19 by bzbarsky-apple