project-chip/connectedhomeip #18034: [SQA] THREAD-In (TC-PS-2.1) SDK needs implementation on optional attributes.

Survensa, 2022-05-04:
**Problem:**
While Executing the  (TC-PS-2.1) non-global attributes of the power source cluster, few optional attributes throws error code as 0x86(unsupported attribute). 

**Attributes:** 

```
./chip-tool powersource read wired-assessed-input-voltage 1 1
./chip-tool powersource read wired-assessed-input-frequency 1 1
./chip-tool powersource read wired-assessed-current 1 1
./chip-tool powersource read wired-nominal-voltage 1 1
./chip-tool powersource read wired-maximum-current 1 1
./chip-tool powersource read wired-present 1 1
./chip-tool powersource read active-wired-faults 1 1
./chip-tool powersource read battery-present 1 1
./chip-tool powersource read battery-common-designation 1 1
./chip-tool powersource read battery-ansidesignation 1 1
./chip-tool powersource read battery-iecdesignation 1 1
./chip-tool powersource read battery-approved-chemistry 1 1
./chip-tool powersource read battery-capacity 1 1
./chip-tool powersource read battery-time-to-full-charge 1 1
./chip-tool powersource read battery-charging-current 1 1
./chip-tool powersource read active-battery-charge-faults 1 1
```

**Expected behavior:** status 0x0

**Actual behavior:**  status = 0x86 (UNSUPPORTED_ATTRIBUTE)

**Platform:** Thread

**CommitID** used : 20e1a4a705770d5222a8ebf04e12ab2ce0304df6

**Reference:**   [Spec](https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/service_device_management/PowerSourceCluster.adoc#attributes) , [Test-plan](https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/powersource.adoc#tc-ps-2-1-attributes-with-server-as-dut) .

**Logs:**

> [1650535089.588787][29787:29792] CHIP:EM: Removed CHIP MessageCounter:14107679 from RetransTable on exchange 5386i
[1650535089.588912][29787:29792] CHIP:DMG: ReportDataMessage =
[1650535089.588977][29787:29792] CHIP:DMG: {
[1650535089.589030][29787:29792] CHIP:DMG: 	AttributeReportIBs =
[1650535089.589119][29787:29792] CHIP:DMG: 	[
[1650535089.589179][29787:29792] CHIP:DMG: 		AttributeReportIB =
[1650535089.589270][29787:29792] CHIP:DMG: 		{
[1650535089.589346][29787:29792] CHIP:DMG: 			AttributeStatusIB =
[1650535089.589425][29787:29792] CHIP:DMG: 			{
[1650535089.589501][29787:29792] CHIP:DMG: 				AttributePathIB =
[1650535089.589590][29787:29792] CHIP:DMG: 				{
[1650535089.589672][29787:29792] CHIP:DMG: 					Endpoint = 0x1,
[1650535089.589760][29787:29792] CHIP:DMG: 					Cluster = 0x2f,
[1650535089.589847][29787:29792] CHIP:DMG: 					Attribute = 0x0000_0006,
[1650535089.589929][29787:29792] CHIP:DMG: 				}
[1650535089.590027][29787:29792] CHIP:DMG: 					
[1650535089.590114][29787:29792] CHIP:DMG: 				StatusIB =
[1650535089.590195][29787:29792] CHIP:DMG: 				{
[1650535089.590278][29787:29792] CHIP:DMG: 					status = 0x86 (UNSUPPORTED_ATTRIBUTE),
[1650535089.590365][29787:29792] CHIP:DMG: 				},
[1650535089.590447][29787:29792] CHIP:DMG: 				
[1650535089.590521][29787:29792] CHIP:DMG: 			},
[1650535089.590605][29787:29792] CHIP:DMG: 			
[1650535089.590672][29787:29792] CHIP:DMG: 		},
[1650535089.590757][29787:29792] CHIP:DMG: 		
[1650535089.590820][29787:29792] CHIP:DMG: 	],
[1650535089.590897][29787:29792] CHIP:DMG: 	
[1650535089.590957][29787:29792] CHIP:DMG: 	SuppressResponse = true, 
[1650535089.591025][29787:29792] CHIP:DMG: 	InteractionModelRevision = 1
[1650535089.591088][29787:29792] CHIP:DMG: }
[1650535089.591326][29787:29792] CHIP:TOO: Response Failure: IM Error 0x00000586: General error: 0x86 (UNSUPPORTED_ATTRIBUTE)
[1650535089.591480][29787:29792] CHIP:EM: Sending Standalone Ack for MessageCounter:9731713 on exchange 5386i

bauerschwan, 2022-05-07:
@Survensa can you pleae re-check this? Looks like this was fixed with PR https://github.com/project-chip/connectedhomeip/pull/18120

bauerschwan, 2022-05-07:
@fessehaeve @Survensa can you pleae re-check this? Looks like this was fixed with PR https://github.com/project-chip/connectedhomeip/pull/18120
@Survensa Please close if it it is fixed now.

fessehaeve, 2022-05-09:
@Survensa  Is this issue addressing a specific application like all-clusters-app?

Survensa, 2022-05-25:
@fessehaeve  yes, I am facing this in all-cluster-app. 
commit used: 516799116dd36a74adef3657e84bdd411ff8dbff

closed 2022-07-20 by Survensa