project-chip/connectedhomeip #18035: Power source cluster testcase [TC-PS-2.1] SDK needs implementation on non-global optional attributes

somu1710, 2022-05-04:
**Problem:**

While Executing the behavior of the non-global attributes of the power source cluster, optional attributes throws error code as 0x86(unsupported attribute). [TC-PS-2.1], [TC-PS-2.2], [TC-PS-3.1].

 **Expected behavior** - pass , status 0x0
 **Actual behavior** - failure , status - 0x86 , General error [Unsupported attribute]
failing commands -

OPTIONAL:

./chip-tool powersource read wired-assessed-input-voltage 1 1
./chip-tool powersource read wired-assessed-input-frequency 1 1
./chip-tool powersource read wired-assessed-current 1 1
./chip-tool powersource read wired-nominal-voltage 1 1
./chip-tool powersource read wired-maximum-current 1 1
./chip-tool powersource read wired-present 1 1
./chip-tool powersource read active-wired-faults 1 1
./chip-tool powersource read battery-present 1 1
./chip-tool powersource read battery-common-designation 1 1
./chip-tool powersource read battery-ansidesignation 1 1
./chip-tool powersource read battery-iecdesignation 1 1
./chip-tool powersource read battery-approved-chemistry 1 1
./chip-tool powersource read battery-capacity 1 1
./chip-tool powersource read battery-time-to-full-charge 1 1
./chip-tool powersource read battery-charging-current 1 1
./chip-tool powersource read active-battery-charge-faults 1 1

Here i attached the logs for your reference
https://github.com/project-chip/connectedhomeip/files/8206433/PRS2.1.FAIL.txt

bauerschwan, 2022-05-04:
What DUT is used here? Since these attributes are all optional the DUT is not forced to implement any of them.

somu1710, 2022-05-05:
We are using Raspberrypi device ,The attributes are optional but it has a test plan how could we verify this optional things?There is no need to implement optional thing then why test plan is written,I have some doubts thats why i am asking.If i am wrong pls crt me

bauerschwan, 2022-05-05:
@somu1710 yes, you are right if the DUT is the all-cluster-app. This should implement even the optional attributes. But keep in mind that this is not done in many clusters. And if all optional attributes are implemented then you cannot test the case where an optional attribute is *not* implemented.

somu1710, 2022-05-05:
Yes, I agree with your words not every clusters had implemted optional attributes but few of them implemented and few of them not implemented,in this case we just left the optional attribute test case and run with the mandatory attributes?

bauerschwan, 2022-05-05:
Yes, you can obviously only test attributes which are  implemented by the DUT you are using. So you can easily skip the ones which are optional and not implemented.

adabreuti, 2022-05-13:
I'm seeing that the following mandatory attributes are not available via the all-clusters applications(TC-PS-2.1 reference log attached):
PICS_A_BATREPLACEMENTNEEDED
PICS_A_WIREDCURRENTTYPE
PICS_A_BATREPLACEABILITY
PICS_A_BATREPLACEMENTDESCRIPTION
PICS_A_BATFUNCTIONALWHILECHARGING

Ref: https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/powersource.adoc
[TC_PS_2_1.log](https://github.com/project-chip/connectedhomeip/files/8691012/TC_PS_2_1.log)

fessehaeve, 2022-05-27:
This is kind of similar to https://github.com/project-chip/connectedhomeip/issues/17592. Optional attributes can only be tested if they exist and there is no optional attribute which is not tied to a feature in Power Source cluster. Therefore I don't see any relevance in adding an attributes that is optional for a feature. I also don't see the test document insisting on verifying this.

somu1710, 2022-05-27:
Optional attributes are 'not verifiable' for V1.0 as per discussion. So I am closing the ticket.

closed 2022-05-27 by somu1710