project-chip/connectedhomeip #18048: Chip-tool build fails due to InteractiveCommands.cpp including non-existent headers

FrankZRS, 2022-05-04:
#### Problem
Hi all, first time user here. I've been following the [BUILDING.md](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/BUILDING.md) guide. Here's the commands I've run: 
```
git clone --recurse-submodules https://github.com/project-chip/connectedhomeip.git
sudo apt-get install git gcc g++ python pkg-config libssl-dev libdbus-1-dev      libglib2.0-dev libavahi-client-dev ninja-build python3-venv python3-dev      python3-pip unzip libgirepository1.0-dev libcairo2-dev
cd connectedhomeip
source scripts/activate.sh
gn gen out/host
ninja -C out/host
```
But the build fails due to InteractiveCommands.cpp including readline/history.h and readline/readline.h which couldn't be found. Here's the error message: 
```
[454/1613] c++ obj/examples/chip-tool/commands/interactive/chip-tool-utils.InteractiveCommands.cpp.o
FAILED: obj/examples/chip-tool/commands/interactive/chip-tool-utils.InteractiveCommands.cpp.o 
g++ -MMD -MF obj/examples/chip-tool/commands/interactive/chip-tool-utils.InteractiveCommands.cpp.o.d -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wno-deprecated-declarations -Wno-unknown-warning-option -Wno-maybe-uninitialized -Wno-missing-field-initializers -Wno-unused-parameter -Wno-psabi -Wno-cast-function-type -fdiagnostics-color -fno-strict-aliasing -Wconversion -D_REENTRANT -I/usr/include/gio-unix-2.0 -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -Wno-implicit-fallthrough -std=gnu++14 -fno-rtti -Wnon-virtual-dtor -DCONFIG_USE_SEPARATE_EVENTLOOP=true -DCONFIG_USE_INTERACTIVE_MODE=true -DCONFIG_ENABLE_YAML_TESTS=true -DCHIP_HAVE_CONFIG_H=1 -DCHIP_ADDRESS_RESOLVE_IMPL_INCLUDE_HEADER=\<lib/address_resolve/AddressResolve_DefaultImpl.h\> -DCHIP_MINMDNS_USE_EPHEMERAL_UNICAST_PORT=1 -I../../examples/chip-tool -I../../zzz_generated/chip-tool -I../../src/lib -I../../src/include -I../../src -Igen/include -I../../zzz_generated/app-common -I../../config/standalone -I../../third_party/nlassert/repo/include -I../../third_party/nlio/repo/include -I../../third_party/nlfaultinjection/repo/include -I../../third_party/inipp/repo/inipp -I../../zzz_generated -I../../zzz_generated/controller-clusters -I../../third_party/jsoncpp/repo/include -I../../examples/common/tracing -c ../../examples/chip-tool/commands/interactive/InteractiveCommands.cpp -o obj/examples/chip-tool/commands/interactive/chip-tool-utils.InteractiveCommands.cpp.o
../../examples/chip-tool/commands/interactive/InteractiveCommands.cpp:21:10: 
fatal error: readline/history.h: No such file or directory
   21 | #include <readline/history.h>
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
```
 #### Proposed Solution
I can see that all other scripts that include these headers have them surrounded by ifdef, would it be possible to add one for this script as well?

YuriChen1028, 2022-05-05:
I have the same problem as you. Because the new dependency was introduced libreadline-dev, so you must 
**sudo apt install libreadline-dev**, then the chip-tool will be built successfully.

FrankZRS, 2022-05-05:
> I have the same problem as you. Because the new dependency was introduced libreadline-dev, so you must **sudo apt install libreadline-dev**, then the chip-tool will be built successfully.

Thanks, 感谢！I've updated the docs and submitted a PR.

closed 2022-05-05 by FrankZRS