project-chip/connectedhomeip #18084: [MF-1.3] Second pairing always fails

adamb-q, 2022-05-05:
When trying to conduct any MF test, (using --commissioner-name option) there is the same error when commissioning second device:

`OnMessageReceived failed, err = ../../examples/chip-tool/third_party/connectedhomeip/src/messaging/ExchangeMgr.cpp:262: CHIP Error 0x00000070: Unsolicited msg with originator bit clear`
`Error on commissioning step 'FindOperational': '../../examples/chip-tool/third_party/connectedhomeip/src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:174: CHIP Error 0x00000032: Timeout`

Tried to run it both on RPI and x64 Ubuntu 21.10 machine and the error is the same. Attaching full logs.

[mf-log.txt](https://github.com/project-chip/connectedhomeip/files/8631081/mf-log.txt)

Damian-Nordic, 2022-05-06:
@adamb-q Is the log coming from the right DUT for sure? I'm asking because on the controller side I can see the following fabric IDs:
```
[1651744275.937547][643:643] CHIP:DIS: Assigned compressed fabric ID: 0x23AF42BE0E98AB1B, node ID: 0x000000000001B669
[1651744275.940699][643:643] CHIP:DIS: Assigned compressed fabric ID: 0x38B9A1AAD07E3D92, node ID: 0x000000000001B669
[1651744275.944410][643:643] CHIP:DIS: Assigned compressed fabric ID: 0xFA5F3CF302515A2B, node ID: 0x000000000001B669
```

while DUT receives the following ones:
```
Assigned compressed fabric ID: 0xC7714E8415D9BF48, node ID: 0x00[P][DIS] Fabric (1) persisted to storage. Calling OnFabricPersistedToStorage
Assigned compressed fabric ID: 0x04A21B08AA4FABBC, node ID: 0x00000000[P][DIS] Fabric (2) persisted to storage. Calling OnFabricPersistedToStorage
```
Anyway, from the controller logs I can see that avahi-browse is able to correctly find the service, so I would suggest to re-test with the https://github.com/project-chip/connectedhomeip/pull/18114 PR included.

adamb-q, 2022-05-06:
@Damian-Nordic yes, it's from the correct device, but you are right - fabrics don't match because I collected DUT not in the same test session as from chip-tool - my bad. I will test it with the PR mentioned and post results here.

adamb-q, 2022-05-06:
After cherrypicking PR#18114 still fails, but this time more logs: 

[TC_MF_1_3.log](https://github.com/project-chip/connectedhomeip/files/8639446/TC_MF_1_3.log)
[MF_1_3_DUT.txt](https://github.com/project-chip/connectedhomeip/files/8639447/MF_1_3_DUT.txt)

yinyihu-silabs, 2022-05-06:
[Thread network 2nd controller log.txt](https://github.com/project-chip/connectedhomeip/files/8640143/Thread.network.2nd.controller.log.txt)
Adding logs mentioned in slack.

andy31415, 2022-05-10:
I believe this is: https://github.com/project-chip/connectedhomeip/issues/18256

TLDR is: thread border router answers mDNS in a way that minMDNS is not prepared to handle.

andy31415, 2022-05-18:
I believe this should be fixed in ToT. @adamb-q would you be able to verify or provide me the steps so I check myself.

I am not yet familiar with how to fetch steps from the code of tests like `MF-1.3`.

adamb-q, 2022-05-19:
@andy31415 compiled qpg-light on ToT (70e259c834c69b4a54bb78ce3fa9693274543887) but error still occurs. Attaching the logs (pretty much the same as uploaded before) In the logs there are also steps if you want to reproduce it yourself:

[DUT.txt](https://github.com/project-chip/connectedhomeip/files/8727303/DUT.txt)
[TC_MF_1_3.1652945057.log](https://github.com/project-chip/connectedhomeip/files/8727304/TC_MF_1_3.1652945057.log)

andy31415, 2022-05-19:
I will try to reproduce - I do not have QPG hardware, so will have to attempt on some different thread board (NRF)

andy31415, 2022-05-19:
Worked for me using nrf-nrf52840dk-light and connecting from my workstation.

had to fix #18615 before that, however after this the test passed.

Here is the test script I used:

```sh
#!/usr/bin/env bash

set -ex

export NODEID1=$RANDOM
export NODEID2=$RANDOM

export THREAD_CRED=hex:0e080000000000010000000300001235060004001fffe00208c6757e4f25e551ef0708fd150126fbadd43e051083469630d05b75596a69887d5bf90c79030f4f70656e5468726561642d626231650102bb1e0410621ac8d0e53905dd1ca9b0ff94d3233d0c0402a0fff8
export CHIP_TOOL="./out/linux-x64-chip-tool-ipv6only/chip-tool"

log_file="run_log_$(date +'%Y-%m-%d_%H-%M-%S').txt"

echo "NODE ID USED: $NODEID1" | tee $log_file

echo "=============================================================" >>$log_file
echo $CHIP_TOOL pairing ble-thread $NODEID1 $THREAD_CRED 20202021 3840 --commissioner-name alpha | tee -a $log_file
$CHIP_TOOL pairing ble-thread $NODEID1 $THREAD_CRED 20202021 3840 --commissioner-name alpha  >>$log_file

echo "=============================================================" >>$log_file
echo $CHIP_TOOL pairing open-commissioning-window $NODEID1 1 200 1000 3840 | tee -a $log_file
$CHIP_TOOL pairing open-commissioning-window $NODEID1 1 200 1000 3840 | tee -a $log_file | grep "Manual pairing code"

# Avahi seems for debug purposes only
echo "=============================================================" >>$log_file
echo avahi-browse -rt _matter._tcp | tee -a $log_file
avahi-browse -rt _matter._tcp 2>&1 | tee -a $log_file | grep -B1 -A3 hostname | tail -n 9

echo "=============================================================" >>$log_file
echo $CHIP_TOOL basic write node-label te9 $NODEID1 0 | tee -a $log_file
$CHIP_TOOL basic write node-label te9 $NODEID1 0 >>$log_file

echo "=============================================================" >>$log_file
echo "Need setup payload (numeric like 34999401957)"
read setup_payload

echo "=============================================================" >>$log_file
echo $CHIP_TOOL payload parse-setup-payload $setup_payload >>$log_file
$CHIP_TOOL payload parse-setup-payload $setup_payload >> $log_file

pin_code=$($CHIP_TOOL payload parse-setup-payload $setup_payload | grep SetUpPINCode | sed 's/.*Code: //')

echo "=============================================================" >>$log_file
echo "Setup Pin code is $pin_code" | tee -a $log_file
echo "Need ip address" | tee -a $log_file
read ip_address

echo "USING IP $ip_address" | tee -a $log_file

echo "=============================================================" >>$log_file
echo $CHIP_TOOL pairing ethernet $NODEID2 $pin_code 3840 "$ip_address" 5540 --commissioner-name beta | tee -a $log_file
$CHIP_TOOL pairing ethernet $NODEID2 $pin_code 3840 "$ip_address" 5540 --commissioner-name beta >>$log_file

# Avahi seems for debug purposes only
echo avahi-browse -rt _matter._tcp | tee -a $log_file
avahi-browse -rt _matter._tcp 2>&1 | tee -a $log_file
```

andy31415, 2022-05-19:
@adamb-q what hardware do you have available for testing? I would like to figure out under what circumstances this happens: if this is qpg only or if this is some environment thing (e.g. I rebooted my OTBR every time and I use random nodeids instead of 1/2 .. I also run on linux-x64 instead of rpi).

andy31415, 2022-05-19:
Also note: the fixes were in chip-tool, so please confirm that you also tried using chip-tool using ToT (not only the DUT).

woody-apple, 2022-05-19:
Cert Blocker Review: Assigning to @cjandhyala to verify if this is resolved.

adamb-q, 2022-05-20:
@andy31415 it's working now for me on ToT, probably had to recompile both chip-tool and qpg-light from the same commit :) thanks for the support, issue can be closed now.

closed 2022-05-20 by adamb-q