project-chip/connectedhomeip #18087: [TC-GC-2.2] General Commissioning Cluster-DUT fails to send  CommissionComplete command to TH With field ErrorCode as 'OK'(0)

Rajashreekalmane, 2022-05-05:
#### Problem
As per the test-plan, test-step says  TH Sends CommissionComplete command to the DUT,  DUT should responds 
With field ErrorCode as 'OK'(0). currently DUT  responds  CommissionComplete command to TH With field ErrorCode as 'OK'(3) 
 
* expected behavior:- DUT sends CommissioningCompleteResponse Command to TH1 With field ErrorCode as 'OK'(0)
 * actual behavior:-DUT  responds  CommissionComplete command to TH With field ErrorCode as 'OK'(3) 

 * steps to reproduce:-./chip-tool generalcommissioning commissioning-complete 1 0
 
system configuration:-
Chip-Tool: RPI
DUT: RPI
Platform : IP and Wifi
Commit-ID:4d50b1084284018c4a628f919b0d2e01ea2a39f8
Dut: all-cluster-app

Test-Plan Reference:-
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/General_Commissioning.adoc#tc-gc-2-2-armfailsafe-command-verification-dut-commissionee

PFA log:-
[TC-GC-2.2.txt](https://github.com/project-chip/connectedhomeip/files/8631424/TC-GC-2.2.txt)

Spec Reference:-
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/service_device_management/GeneralCommissioningCluster.adoc#ref_CommissioningError

cecille, 2022-05-16:
Error 3 is "no failsafe". Was the failsafe set before attempting to use this command? That should be step 1 of the test procedure. What was the commissioning timeout that was used? 

Device logs would be helpful here, for the entire set of commands.

Rajashreekalmane, 2022-05-17:
Test-plan has been changed, now its working so closing this issue here.

closed 2022-05-17 by Rajashreekalmane