project-chip/connectedhomeip #18104: [TC-CC-2.1] ColorTemperatureMireds, ColorMode , EnhancedColorMode default value is mimatching

kowsisoundhar12, 2022-05-05:
As per spec and Testplan 
1. ColorTemperatureMireds default value is "250" but instead of that getting "0" as default value 
2.ColorMode  default value is "1" but instead of that getting "2" as default value 
3.EnhancedColorMode default value is "1"but instead of that getting "2" as default value 

Spec Screenshot:
[image]

Testplan Screenshot:
[image]

 Error Screenshot:
[image]
[image]
[image]

[enhanced-color-mode-value-mismatch.zip](https://github.com/project-chip/connectedhomeip/files/8632933/enhanced-color-mode-value-mismatch.zip)
[color-mode-value-mismatch.zip](https://github.com/project-chip/connectedhomeip/files/8632934/color-mode-value-mismatch.zip)
[color-temprature-value-mismatch.zip](https://github.com/project-chip/connectedhomeip/files/8632935/color-temprature-value-mismatch.zip)

krypton36, 2022-05-12:
@kowsisoundhar12 Can you please attach the YAML file?

kowsisoundhar12, 2022-05-30:
@krypton36 pls find attached yaml file
[Test_TC_CC_2_1.zip](https://github.com/project-chip/connectedhomeip/files/8797594/Test_TC_CC_2_1.zip)

bzbarsky-apple, 2022-10-06:
@kowsisoundhar12 The "default value" as defined in the spec is not the "factory reset" value on a device.