project-chip/connectedhomeip #18121: GeneratedCommandList empty for DoorLock cluster implementations

jrhees-cae, 2022-05-05:
#### Problem
A device implementing the DoorLock cluster has an empty GeneratedCommandList, even though the appropriate responses are in the .zap

 #### Proposed Solution
Discovered that changing the 'optional' attribute in the xml fixes the issue. However, it is unclear whether this is the correct fix or if there is a bug in zap-tool

For example, making the following change in src/app/zap-templates/zcl/data-model/chip/door-lock-cluster.xml:
-        <command source="server" code="28" name="GetUserResponse" disableDefaultResponse="true" optional="true">
+        <command source="server" code="28" name="GetUserResponse" disableDefaultResponse="true" optional="false">

...will cause "GetUserResponse" to be included in the GeneratedCommandList (assuming, of course, it and its corresponding command are also present in the .zap)

bzbarsky-apple, 2022-05-19:
@jrhees-cae Are the commands actually enabled in the ZAP config for the device?  It doesn't sound like they are.

jrhees-cae, 2022-05-19:
Yes they are. However, having the commands in the .zap is not enough. As stated, the generation tool will not include the commands in the generated-command-list unless the .xml template lists the commandResponse as optional=false.

bzbarsky-apple, 2022-05-19:
@jrhees-cae That should be fixed since #17833 merged: as long as the client-to-server command is enabled, the corresponding response will show up in the generated-command-list, even if marked optional.

The responses will also show up in accepted-command-list in some cases, but https://github.com/project-chip/zap/pull/515 will fix that once it merges.

Please let me know if there is still an issue on tip, especially with that ZAP change.

andy31415, 2022-05-20:
Marked Boris as assignee since #18674 will resolve this.

closed 2022-05-20 by andy31415