project-chip/connectedhomeip #18140: [TC-SC-5.1] Write ACL by setting AuthMode as Group fails

pankore, 2022-05-06:
#### Problem
- TC-SC-5.1 step 1
- Write ACL by setting AuthMode as Group returns error code 0x01 (FAILURE)
- Subsequent steps will fail with 0x7e (UNSUPPORTED ACCESS)

mlepage-google, 2022-05-06:
Hi pankore, this generally works so there may be something wrong with the test setup or environment.

pankore, 2022-05-07:
Hi @mlepage-google , may I know what SHA are you using? I am using the TE9 tag and I also see the same FAILURE code written in RED in the test case document of TC-SC-5.1 step 1a.

mlepage-google, 2022-05-07:
I am on latest master, but this has been working for quite a while.

Are you trying to write a list containing a single ACL entry that has group auth mode? That won't work.

When you write the ACL attribute, you are writing a list. A complete list. If you are writing one entry that is group auth mode, you are replacing the existing ACL list (which contains the "admin" ACL that is giving you access) with your new list. As soon as the pre-existing "admin" ACL is gone, you will lose all access.

If you are trying to make a group auth mode ACL entry, it will need to be an *additional* ACL entry; you need to write a list containing an "admin" ACL and any other ACL entry you wish.

bzbarsky-apple, 2022-05-21:
@pankore What is the _exact_ command being executed?  The TE9 discussions I saw had people executing all sorts of incorrect ACL write commands, because some of the test plans were wrong.

cjandhyala, 2022-05-25:
@pankore Can you re-test on latest ToT and update the results.

mrjerryjohns, 2022-05-25:
Please provide the specific chip-tool commands used.

pankore, 2022-05-27:
The command I used was copied from the V1 verification steps document. This command returns FAILURE.
` ./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode":3 , "subjects": [112233], "targets": [{"cluster": null, "endpoint": 0, "deviceType": null}]}]' 1 0`

On the V1.2 verification steps document, there is an uipdate to the command. This command returns UNSUPPORTED_ACCESS.
` ./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode":2 , "subjects": [112233], "targets": [{"cluster": null, "endpoint": 0, "deviceType": null}]}, {"fabricIndex": 1, "privilege": 4, "authMode":3 , "subjects": [112233], "targets": null}]' 1 0`

franck-apple, 2022-05-27:
assigning to @mrjerryjohns , @bzbarsky-apple , @mlepage-google for feedback on the command used.

bzbarsky-apple, 2022-05-27:
>  ./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode":3 , "subjects": [112233], "targets": [{"cluster": null, "endpoint": 0, "deviceType": null}]}]' 1 0

OK, so this uses AuthMode == Group, with a subject value of 112233.  

Per spec, if the auth mode is group then the subjects must be group ids.  In particular, they must be 64 bit values with the group id in the lowest 16 bits and the high 48 bits set to 0.  112233 == 0x1b669 and does not have the 48 high bits set to 0, so this is not a valid subject value for the Group authmode, and the command should fail.  And yes, this matches the bogus things I saw happening during TE9...

>  ./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode":2 , "subjects": [112233], "targets": [{"cluster": null, "endpoint": 0, "deviceType": null}]}, {"fabricIndex": 1, "privilege": 4, "authMode":3 , "subjects": [112233], "targets": null}]' 1 0

This has the same exact problem: group AuthMode, subject that is not a valid group id.

bzbarsky-apple, 2022-05-27:
Oh, and after that first (failed) command the ACLs have been wiped out, which is why you get UNSUPPORTED_ACCESS after that.

@cjandhyala Can we please get the verification steps fixed?

closed 2022-05-27 by bzbarsky-apple