project-chip/connectedhomeip #18143: [TC-MF-1.9], [TC-MF-1.10], [TC-MF-1.11], [TC-MF-1.12] - Couldn't verify this test case because of timeout issue

sowmyassp, 2022-05-06:
**Issue** : 
We have to verify that the test case mentioned above does  verifies the DUT exits its commissioning window after the 20 failed commissioning attempts but couldn't do so because when we pass a invalid passcode in TH2 after opening a commissioning window in TH1 of timeout 900 sec which is 15 min (maximum value) at it takes 5 to 10 min to give the log so within the maximum timeout I couldn't verify 20 attempts of invalid passcode.

**Commit used** : 4d50b1084284018c4a628f919b0d2e01ea2a39f8

**Expected behavior** : Only 5 attempt we could verify the invalid passcode because one attempt takes 5 to 10 min to give log 

**Actual behavior** : We should verify 20 attempts of invalid passcode

**Steps to reproduce** : 
- DUT side :

1. sudo rm -rf /tmp/chip_*
2. sudo ./chip-all-clusters-app --wifi

- TH side :

1. ./chip-tool pairing ble-wifi 1 zigbeehome matter123 20202021 3840 (on TH1)
2. ./chip-tool pairing open-commissioning-window 1 1 900 1000 3840(on TH1)
3. ./chip-tool pairing manualcode 2 36366524220(on TH2)(with 20 attempts of invalid passcode)

**Test platform** :
Chip-tool - RPI/MAC
DUT - RPI

**PFA Logs** : 
[TC-MF-1.9_cntrl.txt](https://github.com/project-chip/connectedhomeip/files/8639142/TC-MF-1.9_cntrl.txt)
[TC-MF-1.9_cntrl02.txt](https://github.com/project-chip/connectedhomeip/files/8639143/TC-MF-1.9_cntrl02.txt)

sowmyassp, 2022-05-18:
With latest commitID (25e241ebcbf11b1f63dbe25546b1f10219866ad0) I am facing same issue

sowmyassp, 2022-05-18:
PFA Logs : 
[MF_1_9_cntrl2.txt](https://github.com/project-chip/connectedhomeip/files/8714094/MF_1_9_cntrl2.txt)
[MF_1_9_cntrl.txt](https://github.com/project-chip/connectedhomeip/files/8714096/MF_1_9_cntrl.txt)
[MF_1_9_dut.txt](https://github.com/project-chip/connectedhomeip/files/8714097/MF_1_9_dut.txt)

vivien-apple, 2022-05-18:
Looking at the logs the issue is that it takes "2" minutes before failure. It happens before this is the default configured timeout for the "pairing" module.
I did https://github.com/project-chip/connectedhomeip/pull/18555 to makes it configurable. Something like `--timeout 25` should be fine looking at the logs...

closed 2022-05-18 by vivien-apple