project-chip/connectedhomeip #18150: [TC-DM-2.1]--While verifying few optional non global attribute to write chip-tool throws an error 0x86((UNSUPPORTED_ATTRIBUTE)) instead 0x88(UNSUPPORTED_WRITE(0x88)

Rajashreekalmane, 2022-05-06:
**#### Problem**
[[TC-DM-2.1]--While verifying few optional non global attribute to write chip-tool throws an error 0x86((UNSUPPORTED_ATTRIBUTE)) instead 0x88(UNSUPPORTED_WRITE(0x88) those attributes are mentioned below.
1. CapabilityMinima attribute
2.  Reachable attribute
3.  UniqueID  attribute
4. SerialNumber_test attribute
5.  ProductLabel  attribute
6. SoftwareVersionString attribute

**expected behavior**:-UNSUPPORTED_WRITE(0x88)
 **actual behavior:**- CONSTRAINT_ERROR(0x87)

 *** steps to reproduce**
 sudo ./chip-tool basic write-by-id 0x14 partlabel 1 0
sudo ./chip-tool basic write-by-id 0x15 newserialnumber 1 0
sudo ./chip-tool basic write-by-id 0x17 0 1 0
sudo ./chip-tool basic write-by-id 0x18 newid 1 0
sudo ./chip-tool basic write-by-id 0x19 4 1 0

**system configuration**
Chip-Tool: RPI
DUT: RPI
Platform : IP and Wifi
Commit-ID:4d50b1084284018c4a628f919b0d2e01ea2a39f8
Dut: All-Clusters-app

**PFA logs:-**
[DM-2.1.txt](https://github.com/project-chip/connectedhomeip/files/8639931/DM-2.1.txt)

**Test-Plan reference;-**
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/devicemanagement.adoc#tc-dm-2-1-basic-information-cluster-server-attributes

bzbarsky-apple, 2022-05-21:
Is the actual behavior CONSTRAINT_ERROR or UNSUPPORTED_ATTRIBUTE?  The above is not self-consistent....

UNSUPPORTED_ATTRIBUTE is somewhat expected if the attributes are actually not supported, no?

bzbarsky-apple, 2022-05-21:
@Rajashreekalmane

Rajashreekalmane, 2022-05-23:
@bzbarsky-apple As per the test-plan we should get 0x88(UNSUPPORTED_WRITE(0x88) but currently I am getting  0x86((UNSUPPORTED_ATTRIBUTE)).
expected behavior:-UNSUPPORTED_WRITE(0x88)
actual behavior:- 0x86((UNSUPPORTED_ATTRIBUTE))

bzbarsky-apple, 2022-05-23:
> sudo ./chip-tool basic write-by-id 0x14 partlabel 1 0
sudo ./chip-tool basic write-by-id 0x15 newserialnumber 1 0
sudo ./chip-tool basic write-by-id 0x17 0 1 0
sudo ./chip-tool basic write-by-id 0x18 newid 1 0
sudo ./chip-tool basic write-by-id 0x19 4 1 0

Every single one of those attribute ids does not exist in the spec.  So UNSUPPORTED_ATTRIBUTE is correct.

I _assume_ you meant `14` instead of `0x14` and so forth (if the intent is to test ProductLabel and the like).  Those attribute ids work fine and return

closed 2022-05-23 by bzbarsky-apple