project-chip/connectedhomeip #18155: [TC-MF-1.4] Second commissioning fails.

yinyihu-silabs, 2022-05-06:
#### Problem
After running _administratorcommissioning open-basic-commissioning-window_ from first controller, second controller _pairing onnetwork_ command failed with timeout.

[ctrl1-chip-tool.txt](https://github.com/project-chip/connectedhomeip/files/8641153/ctrl1-chip-tool.txt)
[ctrl2-onnetwork.txt](https://github.com/project-chip/connectedhomeip/files/8641154/ctrl2-onnetwork.txt)
[device-rtt.txt](https://github.com/project-chip/connectedhomeip/files/8641155/device-rtt.txt)

andy31415, 2022-05-18:
@yinyihu-silabs - do you happen to have tests to validate this?

I have submitted a final fix for #18256 yesterday, would like to validate if this fixes this issue or not.

woody-apple, 2022-05-19:
Cert Blocker Review: Assigning to @cjandhyala to follow up.

cjandhyala, 2022-05-20:
pre @yinyihu-silabs this ticket can be closed

closed 2022-05-20 by cjandhyala