project-chip/connectedhomeip #18157: [SQA] Door-lock Clusters are not  working in Lock-App.

Survensa, 2022-05-06:
**Problem :** 
While Executing the TC-DL-2.17 , All 5 events are not Working in the Lock-App , But in All-clusters-App 2( lock-operation and lock-operation-error)  events are working . 

**Step to reproduce :**  `./chip-tool doorlock read-event  lock-operation 1 1`

**Commit** Used :  4d50b1084284018c4a628f919b0d2e01ea2a39f8

**Platform :** RPI- RPI.

**Log of All-cluster-app:** 

> [1651822378.426088][5624:5629] CHIP:EM: Received message of type 0x5 with protocolId (0, 1) and MessageCounter:1339971 on exchange 65074i
[1651822378.426130][5624:5629] CHIP:EM: Found matching exchange: 65074i, Delegate: 0xffff6c004350
[1651822378.426172][5624:5629] CHIP:EM: Rxd Ack; Removing MessageCounter:14632831 from Retrans Table on exchange 65074i
[1651822378.426203][5624:5629] CHIP:EM: Removed CHIP MessageCounter:14632831 from RetransTable on exchange 65074i
[1651822378.426300][5624:5629] CHIP:DMG: ReportDataMessage =
[1651822378.426335][5624:5629] CHIP:DMG: {
[1651822378.426364][5624:5629] CHIP:DMG:        EventReportIBs =
[1651822378.426408][5624:5629] CHIP:DMG:        [
[1651822378.426441][5624:5629] CHIP:DMG:                EventReportIB =
[1651822378.426498][5624:5629] CHIP:DMG:                {
[1651822378.426535][5624:5629] CHIP:DMG:                        EventDataIB =
[1651822378.426582][5624:5629] CHIP:DMG:                        {
[1651822378.426622][5624:5629] CHIP:DMG:                                EventPath =
[1651822378.426671][5624:5629] CHIP:DMG:                                {
[1651822378.426719][5624:5629] CHIP:DMG:                                        Endpoint = 0x1,
[1651822378.426771][5624:5629] CHIP:DMG:                                        Cluster = 0x101,
[1651822378.426821][5624:5629] CHIP:DMG:                                        Event = 0x2,
[1651822378.426868][5624:5629] CHIP:DMG:                                },
[1651822378.426919][5624:5629] CHIP:DMG:
[1651822378.426963][5624:5629] CHIP:DMG:                                EventNumber = 0x2,
[1651822378.427012][5624:5629] CHIP:DMG:                                PriorityLevel = 0x2,
[1651822378.427057][5624:5629] CHIP:DMG:                                SystemTimestamp = 0x617656,
[1651822378.427100][5624:5629] CHIP:DMG:                                EventData =
[1651822378.427148][5624:5629] CHIP:DMG:                                {
[1651822378.427194][5624:5629] CHIP:DMG:                                        0x0 = 0,
[1651822378.427244][5624:5629] CHIP:DMG:                                        0x1 = 7,
[1651822378.427295][5624:5629] CHIP:DMG:                                        0x2 = 0,
[1651822378.427346][5624:5629] CHIP:DMG:                                        0x3 = 1,
[1651822378.427423][5624:5629] CHIP:DMG:                                        0x4 = 112233,
[1651822378.427482][5624:5629] CHIP:DMG:                                        0x5 = NULL
[1651822378.427562][5624:5629] CHIP:DMG:                                },
[1651822378.427622][5624:5629] CHIP:DMG:                        },
[1651822378.427684][5624:5629] CHIP:DMG:
[1651822378.427728][5624:5629] CHIP:DMG:                },
[1651822378.427803][5624:5629] CHIP:DMG:
[1651822378.427843][5624:5629] CHIP:DMG:                EventReportIB =
[1651822378.427906][5624:5629] CHIP:DMG:                {
[1651822378.427951][5624:5629] CHIP:DMG:                        EventDataIB =
[1651822378.428008][5624:5629] CHIP:DMG:                        {
[1651822378.428056][5624:5629] CHIP:DMG:                                EventPath =
[1651822378.428110][5624:5629] CHIP:DMG:                                {
[1651822378.428171][5624:5629] CHIP:DMG:                                        Endpoint = 0x1,
[1651822378.428234][5624:5629] CHIP:DMG:                                        Cluster = 0x101,
[1651822378.428294][5624:5629] CHIP:DMG:                                        Event = 0x2,
[1651822378.428350][5624:5629] CHIP:DMG:                                },
[1651822378.428411][5624:5629] CHIP:DMG:
[1651822378.428467][5624:5629] CHIP:DMG:                                EventNumber = 0x5,
[1651822378.428526][5624:5629] CHIP:DMG:                                PriorityLevel = 0x2,
[1651822378.428580][5624:5629] CHIP:DMG:                                DeltaSystemTimestamp = 0x635a5,
[1651822378.428631][5624:5629] CHIP:DMG:                                EventData =
[1651822378.428689][5624:5629] CHIP:DMG:                                {
[1651822378.428749][5624:5629] CHIP:DMG:                                        0x0 = 0,
[1651822378.428812][5624:5629] CHIP:DMG:                                        0x1 = 7,
[1651822378.428874][5624:5629] CHIP:DMG:                                        0x2 = 0,
[1651822378.428937][5624:5629] CHIP:DMG:                                        0x3 = 1,
[1651822378.428996][5624:5629] CHIP:DMG:                                        0x4 = 112233,
[1651822378.429056][5624:5629] CHIP:DMG:                                        0x5 = [
[1651822378.429116][5624:5629] CHIP:DMG:
[1651822378.429181][5624:5629] CHIP:DMG:                                                {
[... 76 lines cut ...]
[1651822378.433873][5624:5629] CHIP:TOO:         Event number: 6
[1651822378.433909][5624:5629] CHIP:TOO:         Priority: Critical
[1651822378.433944][5624:5629] CHIP:TOO:         Timestamp: 7326552
[1651822378.433999][5624:5629] CHIP:TOO:   LockOperation: {
[1651822378.434038][5624:5629] CHIP:TOO:     LockOperationType: 0
[1651822378.434074][5624:5629] CHIP:TOO:     OperationSource: 7
[1651822378.434110][5624:5629] CHIP:TOO:     UserIndex: 0
[1651822378.434145][5624:5629] CHIP:TOO:     FabricIndex: 1
[1651822378.434180][5624:5629] CHIP:TOO:     SourceNode: 112233
[1651822378.434221][5624:5629] CHIP:TOO:     Credentials: 1 entries
[1651822378.434268][5624:5629] CHIP:TOO:       [1]: {
[1651822378.434306][5624:5629] CHIP:TOO:         CredentialType: 1
[1651822378.434341][5624:5629] CHIP:TOO:         CredentialIndex: 0
[1651822378.434376][5624:5629] CHIP:TOO:        }
[1651822378.434414][5624:5629] CHIP:TOO:    }
[1651822378.434597][5624:5629] CHIP:EM: Sending Standalone Ack for MessageCounter:1339971 on exchange 65074i
[1651822378.434696][5624:5629] CHIP:IN: Prepared secure message 0xffff80b70938 to 0x0000000000000001 (1)  of type 0x10 and protocolId (0, 0) on exchange 65074i with MessageCounter:14632832.

**log of Lock-App:** 

> [1651216572.325926][3144:3149] CHIP:DMG: ReportDataMessage =
[1651216572.325964][3144:3149] CHIP:DMG: {
[1651216572.325998][3144:3149] CHIP:DMG:        SuppressResponse = true,
[1651216572.326035][3144:3149] CHIP:DMG:        InteractionModelRevision = 1
[1651216572.326070][3144:3149] CHIP:DMG: }

Survensa, 2022-05-25:
Verified with commit `516799116dd36a74adef3657e84bdd411ff8dbff` but not able to fetch any event.

woody-apple, 2022-05-25:
Cert Blocker Review: Assigning to myself and @franck-apple to find an assignee. (Maybe @Morozov-5F)

franck-apple, 2022-05-26:
@Morozov-5F, I'm assigning it to you, please update assignee if not appropriate.

Morozov-5F, 2022-06-03:
I could not reproduce it with the latest lock-app. Here's my steps: 
1. Pair the lock-app with chip-tool
2. Create a user on a lock:
   ```
   ? chip-tool doorlock set-credential 0 '{ "credentialType" : 1 , "credentialIndex" : 1 }' 123456 null null null 1 1 --timedInteractionTimeoutMs 1000
   ```
3. Read the lock user change event and verify there are two events (one for user, one for credential added to user):
   ```
   ? chip-tool doorlock read-event lock-user-change 1 1
   ...
   [1654238395187] [74426:8157271] CHIP: [TOO] Endpoint: 1 Cluster: 0x0000_0101 Event 0x0000_0004
   [1654238395187] [74426:8157271] CHIP: [TOO] 	 Event number: 3
   [1654238395187] [74426:8157271] CHIP: [TOO] 	 Priority: Info
   [1654238395187] [74426:8157271] CHIP: [TOO] 	 Timestamp: 34530712
   [1654238395188] [74426:8157271] CHIP: [TOO]   LockUserChange: {
   [1654238395188] [74426:8157271] CHIP: [TOO]     LockDataType: 2
   [1654238395188] [74426:8157271] CHIP: [TOO]     DataOperationType: 0
   [1654238395188] [74426:8157271] CHIP: [TOO]     OperationSource: 7
   [1654238395188] [74426:8157271] CHIP: [TOO]     UserIndex: 1
   [1654238395188] [74426:8157271] CHIP: [TOO]     FabricIndex: 1
   [1654238395188] [74426:8157271] CHIP: [TOO]     SourceNode: 112233
   [1654238395188] [74426:8157271] CHIP: [TOO]     DataIndex: 1
   [1654238395188] [74426:8157271] CHIP: [TOO]    }
   [1654238395188] [74426:8157271] CHIP: [TOO] Endpoint: 1 Cluster: 0x0000_0101 Event 0x0000_0004
   [1654238395188] [74426:8157271] CHIP: [TOO] 	 Event number: 4
   [1654238395188] [74426:8157271] CHIP: [TOO] 	 Priority: Info
   [1654238395188] [74426:8157271] CHIP: [TOO] 	 Timestamp: 34530712
   [1654238395188] [74426:8157271] CHIP: [TOO]   LockUserChange: {
   [1654238395188] [74426:8157271] CHIP: [TOO]     LockDataType: 6
   [1654238395188] [74426:8157271] CHIP: [TOO]     DataOperationType: 0
   [1654238395188] [74426:8157271] CHIP: [TOO]     OperationSource: 7
   [1654238395188] [74426:8157271] CHIP: [TOO]     UserIndex: 1
   [1654238395188] [74426:8157271] CHIP: [TOO]     FabricIndex: 1
   [1654238395188] [74426:8157271] CHIP: [TOO]     SourceNode: 112233
   [1654238395188] [74426:8157271] CHIP: [TOO]     DataIndex: 1
   [1654238395188] [74426:8157271] CHIP: [TOO]    }
   ```
3. Unlock the door:
   ```
   ? chip-tool doorlock unlock-door 1 1 --PinCode 123456 --timedInteractionTimeoutMs 1000
   ```
4. Read the lock operation event and verify that it is present:
   ```
   ? chip-tool doorlock read-event lock-operation 1 1
   ...
   [1654238642435] [74447:8159485] CHIP: [TOO] Endpoint: 1 Cluster: 0x0000_0101 Event 0x0000_0002
   [1654238642435] [74447:8159485] CHIP: [TOO] 	 Event number: 5
   [1654238642435] [74447:8159485] CHIP: [TOO] 	 Priority: Critical
   [1654238642435] [74447:8159485] CHIP: [TOO] 	 Timestamp: 34778164
   [1654238642435] [74447:8159485] CHIP: [TOO]   LockOperation: {
   [1654238642435] [74447:8159485] CHIP: [TOO]     LockOperationType: 1
   [1654238642435] [74447:8159485] CHIP: [TOO]     OperationSource: 7
   [1654238642435] [74447:8159485] CHIP: [TOO]     UserIndex: 1
   [1654238642435] [74447:8159485] CHIP: [TOO]     FabricIndex: 1
   [1654238642435] [74447:8159485] CHIP: [TOO]     SourceNode: 112233
   [1654238642435] [74447:8159485] CHIP: [TOO]     Credentials: 1 entries
   [1654238642435] [74447:8159485] CHIP: [TOO]       [1]: {
   [1654238642435] [74447:8159485] CHIP: [TOO]         CredentialType: 1
   [1654238642435] [74447:8159485] CHIP: [TOO]         CredentialIndex: 1
   [1654238642435] [74447:8159485] CHIP: [TOO]        }
   [1654238642435] [74447:8159485] CHIP: [TOO]    }
   ...
   ```
I ran lock-app both on my laptop and on RPI 4 8GB. Is there any other steps I can take to reproduce the issue?