project-chip/connectedhomeip #18180: TE9 Groups Chip tool Darwin testing: Groupnames attribute not present to read

Garik098, 2022-05-07:
As per the testcase TC-GR-3.1 step 2, Groupnames attribute needs to be present to read, but the attribute is not present in chip tool Darwin. This attribute is not present in chip tool either. 
[image]
[image]

dwgaul, 2022-05-20:
Actually it looks like this is an issue with the test plan that is being resolved here: https://github.com/CHIP-Specifications/chip-test-plans/pull/1279

Looking at the spec, Group Name is not an attribute, but is a data field used in the AddGroup, AddGroupIfIdentifying, and ViewGroupResponse commands.

cjandhyala, 2022-05-20:
@Garik098 @dwgaul also I don't see the featureMap attribute in here ..

Garik098, 2022-05-20:
Raised a ticket on test plan - https://github.com/CHIP-Specifications/chip-test-plans/issues/1301

closed 2022-05-20 by Garik098