project-chip/connectedhomeip #18185: TE9 TC-DL-3.3 chip tool Darwin testing: read-event attributes support needed

Garik098, 2022-05-07:
To test TC-DL-3.3, read-event attributes support is needed and currently the attributes were not not present in chip tool Darwin. Below is the screenshot between chip tool and chip tool Darwin on the same,
[image]

krypton36, 2022-05-13:
This will not be supported in CHIP.framework. @bzbarsky-apple can you please confirm.

bzbarsky-apple, 2022-05-14:
For the moment that is correct.  If that ever changes, we'll add chip-tool-darwin support at that point.

woody-apple, 2022-08-02:
Moving this back into 1.0 for Darwin. Given we should support this.

Garik098, 2022-08-03:
@woody-apple

As per our discussion in the meeting, the read-event, subscribe-event, read-event-by-id, subscribe-event-by-id support in Darwin tool is needed for the below testcases

TC-SWTCH-3.2
TC-DRLK-3.3
TC-BINFO-2.4
TC-IDM-6.3
TC-IDM-6.4

Here is the screenshot from the latest commit on Darwin tool for Doorlock where the read-event is missing. The same with other clusters.

[image]

bzbarsky-apple, 2022-08-03:
@Garik098 Why can't those tests be done with the `subscribe-all-events` command?

woody-apple, 2022-08-03:
What does the test step require? We should make sure it’s clear what’s specifically needed for the test and the component.

woody-apple, 2022-08-03:
(And which commands to support those tests)

Garik098, 2022-08-03:
ATLs follow the commands in the verification steps document to certify. So providing the test case screenshot and the verification step for it. 

For TC-DRLK-3.3, we need read-event support. They are testing each event individually in this test case. 
[image]

[image]

For IDM 6.3, the verification steps document gives this command - 
sudo ./chip-tool any  read-event-by-id 0xFFFFFFFF 0xFFFFFFFF 1 0xFFFF

[image]

For IDM 6.4, the verification steps document gives this command - 
./chip-tool interactive start
basic subscribe-event start-up 10 100 1 0
basic subscribe-event-by-id 0x000 20 400 1 0

[image]

For Switch 3.2, I guess we can use Subscribe-all-events as the verification steps document gives a generic verification statement. 
[image]

[image]

For the BINFO-2.4, I couldn't find the test case. Looks like it was removed post SVE1.

bzbarsky-apple, 2022-08-03:
In terms of the test plans:

* TC-SWTCH-3.2 (https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/switch.adoc#tc-swtch-3-2-primary-functionality-with-client-as-dut) requires:
   1. "Commission DUT to TH.  Use client mechanism to set it up so switch state can be observed (e.g. UI shows state of switch)"
   2. "DUT sets up eventing so it will receive events when the switch is operated. TH responds accordingly."
   3. "DUT receives these events"
* TC-DRLK-3.3 (https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/door_lock_Cluster.adoc#tc-drlk-3-3-verification-for-the-door-lock-events-dut-client) requires (without ever saying anything about the DUT subscribing to events):
   1. "Verify that DUT receives the DoorLockAlarm event"
   2. "Verify that DUT receives DoorStateChange Event from TH"
   3. etc.
*  TC-BINFO-2.4 (https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/devicemanagement.adoc#424-tc-binfo-24-events-dut-client) requires:
    1. "DUT subscribes to the events it supports"
    2. "Verify that the DUT receives the ReachableChanged event"
    3. "Verify that the DUT receives the ShutDown event."
    4. etc.
*  TC-IDM-6.4 (https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/interactiondatamodel.adoc#tc-idm-6-4-events-subscribe-interaction-from-dut-to-th-dut_client) requires:
    1. DUT sends Subscribe Request Message to the TH for a supported event.

Those can all be handled with `subscribe-all-events`.

*  TC-IDM-6.3 (https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/interactiondatamodel.adoc#tc-idm-6-3-events-read-interaction-from-dut-to-th-dut_client) requires reading events via ReadRequest, conditioned only on `MCORE.IDM.C.ReadRequest`.  That seems like a problem: clients are not required to implement Read Request for events (and in fact, that's not how events should generally be consumed).  There should be a separate PICS for "supports doing a Read Request for an event", imo.

bzbarsky-apple, 2022-08-03:
> For TC-DRLK-3.3, we need read-event support. They are testing each event individually in this test case.

This is not a reasonable requirement for a client, and the test plan does not impose the constraint that the client use a Read interaction to get the events....  

More to the point, I don't see how verification steps written specifically for `chip-tool` can apply to other clients.  Other clients will need verification steps that make sense for them, which implement the actual things the test plan says to do.

Garik098, 2022-08-05:
Ya I agree with the statement that the steps written for chip tool can't apply to other clients. We can clarify that to the ATLs while testing. 

Working on few test plan PRs this week as we have a deadline. I will check all the above testcases early next week with subscribe-all-events and see if we can execute them completely without depending on other ones. If that is the case we need not make any changes and go ahead with subscribe-all-events.

Garik098, 2022-08-18:
I am on this commit in master 

commit 0691bbcbb26f7fa39d90fc54c35fb66fe12196c5 (HEAD -> master)

I used this command to do subscribe-all-events - ./darwin-framework-tool any subscribe-all-events 2 1000 1 0

I think that the subscribe all events command should wait at the end without exiting and show events when they triggered right. But I see that when I run the command it is probably showing the current state of various events and is exiting. 

[subscribe-all-events.txt](https://github.com/project-chip/connectedhomeip/files/9377793/subscribe-all-events.txt)

woody-apple, 2022-09-06:
Darwin Issue Review: This is not needed for 1.0, moving out.

xtyHY, 2022-12-22:
@woody-apple 

Our iOS app is testing for get the Matter-certified.  During the testing, the same problem occurred. 

The **TC-BINFO-2.4** require `subscribe-event start-up` and `read-event start-up`, but we can't read the event.

The Matter-certified PICs requirement must be implemented. If not chose, it return Error like this:
[image]

The `BINFO.C.E00` associated the use cases `read-event start-up`:
[image]

bzbarsky-apple, 2022-12-22:
Just as a note: a DUT can subscribe to all events and then process the StartUp when it receives that on the subscription, which is the normal mode of operation for events.

The testcase verification steps there are an example, and not a requirement to implement "read an event", as I understand it.  But @cjandhyala could confirm.

cjandhyala, 2022-12-22:
Correct @bzbarsky-apple, the client TCs verification steps are there as an example how these were verified using chip-tool and not a necessary requirement on client to implement the same way as chip-tool does. 

@xtyHY The purpose of TC is the client able to receive the events, on the Darwin platform you can use 'subscribe-all-events ' and filter for the StartUp and other events. On the example app (TH) you can trigger events using the commands mentioned at https://github.com/project-chip/connectedhomeip/blob/e56c2ce42f4bf5d7a5aab8898174b4f14e7888e6/examples/all-clusters-app/linux/README.md

xtyHY, 2022-12-23:
@cjandhyala Thanks for your explanation. 
We will communicate with the testing agency to synchronize relevant information.

bzbarsky-apple, 2023-01-27:
#24057 added read-event-by-id support to darwin-framework-tool.  We could add read-event using a similar backend, I suspect.

woody-apple, 2025-01-29:
Closing this, as we believe we should fix the test plans.

closed 2025-01-29 by woody-apple