project-chip/connectedhomeip #18194: Commission fail with chip-tool over existed WiFi network

wxiangcheng, 2022-05-09:
Appreciate if anyone can point out how to debug more and any comments are welcome.
 #### Problem
**Commission fail with chip-tool over existed WiFi network**
On Commissionee device, from log it indicates "Commissioning completed session establishment step" and "Device completed Rendezvous process". I can also see the next AccessControl info is started but finally **failed at unknown state**.

On commissioner controller which running chip-tool on Linux test PC, the error is at "Error parsing commissioning information" and "Failed to perform commissioning step 2". From chip-tool's log , the PASE is completed ("Remote device completed SPAKE2+ handshake"). But chip-tool prints "ReportDataMessage" as below:

_[1651910032.851598][412605:412610] CHIP:DMG: ReportDataMessage =
[1651910032.851647][412605:412610] CHIP:DMG: {
[1651910032.851687][412605:412610] CHIP:DMG: 	AttributeReportIBs =
[1651910032.851742][412605:412610] CHIP:DMG: 	[
[1651910032.851776][412605:412610] CHIP:DMG: 		AttributeReportIB =
[1651910032.851827][412605:412610] CHIP:DMG: 		{
[1651910032.851866][412605:412610] CHIP:DMG: 			AttributeStatusIB =
[1651910032.851909][412605:412610] CHIP:DMG: 			{
[1651910032.851943][412605:412610] CHIP:DMG: 				AttributePathIB =
[1651910032.851983][412605:412610] CHIP:DMG: 				{
[1651910032.852022][412605:412610] CHIP:DMG: 					Endpoint = 0x0,
[1651910032.852066][412605:412610] CHIP:DMG: 					Cluster = 0x28,
[1651910032.852116][412605:412610] CHIP:DMG: 					Attribute = 0x0000_0009,
[1651910032.852173][412605:412610] CHIP:DMG: 				}
[1651910032.852221][412605:412610] CHIP:DMG: 					
[1651910032.852265][412605:412610] CHIP:DMG: 				StatusIB =
[1651910032.852299][412605:412610] CHIP:DMG: 				{
[1651910032.852336][412605:412610] CHIP:DMG: 					status = 0x1,
[1651910032.852367][412605:412610] CHIP:DMG: 				},
[1651910032.852400][412605:412610] CHIP:DMG: 				
[1651910032.852423][412605:412610] CHIP:DMG: 			},
[1651910032.852458][412605:412610] CHIP:DMG: 			
[1651910032.852475][412605:412610] CHIP:DMG: 		},
[1651910032.852510][412605:412610] CHIP:DMG: 		
[1651910032.852530][412605:412610] CHIP:DMG: 		AttributeReportIB =
[1651910032.852560][412605:412610] CHIP:DMG: 		{
[1651910032.852579][412605:412610] CHIP:DMG: 			AttributeStatusIB =
[1651910032.852604][412605:412610] CHIP:DMG: 			{
[1651910032.852624][412605:412610] CHIP:DMG: 				AttributePathIB =
[1651910032.852649][412605:412610] CHIP:DMG: 				{
[1651910032.852675][412605:412610] CHIP:DMG: 					Endpoint = 0x0,
[1651910032.852704][412605:412610] CHIP:DMG: 					Cluster = 0x28,
[1651910032.852750][412605:412610] CHIP:DMG: 					Attribute = 0x0000_0004,
[1651910032.852774][412605:412610] CHIP:DMG: 				}
[1651910032.852807][412605:412610] CHIP:DMG: 					
[1651910032.852830][412605:412610] CHIP:DMG: 				StatusIB =
[1651910032.852856][412605:412610] CHIP:DMG: 				{
[1651910032.852879][412605:412610] CHIP:DMG: 					status = 0x1,
[1651910032.852905][412605:412610] CHIP:DMG: 				},
[1651910032.852930][412605:412610] CHIP:DMG: 				
[1651910032.852952][412605:412610] CHIP:DMG: 			},
[1651910032.852980][412605:412610] CHIP:DMG: 			
[1651910032.852998][412605:412610] CHIP:DMG: 		},
[1651910032.853035][412605:412610] CHIP:DMG: 		
[1651910032.853054][412605:412610] CHIP:DMG: 		AttributeReportIB =
[1651910032.853083][412605:412610] CHIP:DMG: 		{
[1651910032.853101][412605:412610] CHIP:DMG: 			AttributeStatusIB =
[1651910032.853126][412605:412610] CHIP:DMG: 			{
[1651910032.853146][412605:412610] CHIP:DMG: 				AttributePathIB =
[1651910032.853169][412605:412610] CHIP:DMG: 				{
[1651910032.853193][412605:412610] CHIP:DMG: 					Endpoint = 0x0,
[1651910032.853218][412605:412610] CHIP:DMG: 					Cluster = 0x28,
[1651910032.853244][412605:412610] CHIP:DMG: 					Attribute = 0x0000_0002,
[1651910032.853270][412605:412610] CHIP:DMG: 				}
[1651910032.853299][412605:412610] CHIP:DMG: 					
[1651910032.853322][412605:412610] CHIP:DMG: 				StatusIB =
[1651910032.853348][412605:412610] CHIP:DMG: 				{
[1651910032.853374][412605:412610] CHIP:DMG: 					status = 0x1,
[1651910032.853399][412605:412610] CHIP:DMG: 				},
[1651910032.853425][412605:412610] CHIP:DMG: 				
[1651910032.853444][412605:412610] CHIP:DMG: 			},
[1651910032.853471][412605:412610] CHIP:DMG: 			
[1651910032.853489][412605:412610] CHIP:DMG: 		},
[... 121 lines cut ...]
[1651910032.856536][412605:412610] CHIP:EM: Sending Standalone Ack for MessageCounter:6400497 on exchange 40475i
[1651910032.856629][412605:412610] CHIP:IN: Prepared secure message 0x7f89f494e060 to 0x0000000000BC5C01 (1)  of type 0x10 and protocolId (0, 0) on exchange 40475i with MessageCounter:7609558.
[1651910032.856653][412605:412610] CHIP:IN: Sending encrypted msg 0x7f89f494e060 with MessageCounter:7609558 to 0x0000000000BC5C01 (1) at monotonic time: 685942376 msec
[1651910032.856731][412605:412610] CHIP:EM: Flushed pending ack for MessageCounter:6400497 on exchange 40475i
[1651910032.856843][412605:412605] CHIP:CTL: Shutting down the commissioner
[1651910032.856915][412605:412605] CHIP:CTL: Shutting down the controller
[1651910032.856936][412605:412605] CHIP:CTL: Shutting down the commissioner
[1651910032.856971][412605:412605] CHIP:CTL: Shutting down the controller
[1651910032.856998][412605:412605] CHIP:CTL: Shutting down the commissioner
[1651910032.857030][412605:412605] CHIP:CTL: Shutting down the controller
[1651910032.857054][412605:412605] CHIP:CTL: Shutting down the commissioner
[1651910032.857087][412605:412605] CHIP:CTL: Shutting down the controller
[1651910032.857106][412605:412605] CHIP:CTL: Shutting down the System State, this will teardown the CHIP Stack
[1651910032.857209][412605:412605] CHIP:DMG: IM WH moving to [Uninitialized]
[1651910032.857223][412605:412605] CHIP:DMG: IM WH moving to [Uninitialized]
[1651910032.857237][412605:412605] CHIP:DMG: IM WH moving to [Uninitialized]
[1651910032.857260][412605:412605] CHIP:DMG: IM WH moving to [Uninitialized]
[1651910032.857465][412605:412605] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-Mo6Kys)
[1651910032.857867][412605:412605] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1651910032.857902][412605:412605] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1651910032.857933][412605:412605] CHIP:DL: Inet Layer shutdown
[1651910032.857942][412605:412605] CHIP:DL: System Layer shutdown
[1651910032.858014][412605:412605] CHIP:TOO: Run command failure: ../../examples/chip-tool/third_party/connectedhomeip/src/app/AttributeCache.cpp:135: CHIP Error 0x000000CA: Interaction Model Error
[1651910032.860769][412605:412605] CHIP:SPT: VerifyOrDie failure at ../../examples/chip-tool/third_party/connectedhomeip/src/lib/support/Pool.h:311: Allocated() == 0
Aborted_

wxiangcheng, 2022-05-19:
Thanks Cecille. I tried to make it passed PASE phase now(some timeout issue) and currently the commission still fails at CASE phase.
The current log indicates: chip[DIS]Verifying the received credentials, but then it failed with "CHIP_ERROR_INVALID_SIGNATURE" because of "MBEDTLS_ERR_ECP_VERIFY_FAILED".

I'm wondering how to get a standard debug method for the new comer on a new board (not in src/platform). Where can I get some reference process that is helpful to my developing?

andy31415, 2022-05-30:
@wxiangcheng - it sounds like a certificate validation error

My best guess for debugging would be to try to log the content of data passed in into mbedtls_ecdsa_verify and see if the values are as expected.

cecille, 2022-05-30:
Apologies for the delay.

Are you saying that this is failing to establish a CASE connection? The error you cited here looks like a failure at the AddNOC command, which happens over the PASE connection and before we attempt to establish CASE. If you could attach full logs, it would be helpful. 

What are you using for commissioning? Is this using chip-tool? 

Assuming this failure is at the AddNOC command, the device is saying that it is unable to verify the signature on the node operational certificate provided by the commissioner. Given that chip-tool is used fairly regularly and this is a new platform, I expect the signature is OK and the device doesn't have the crypto set up correctly. 

Give that error message, it looks like you're configured to use mbedTLS for the signature verification. You probably want to narrow down where the error is actually happening in the mbedTLS code, but also verify that your mbedTLS config is sensible for your platform.

closed 2022-12-05 by stale[bot]