project-chip/connectedhomeip #18195: [TC-DD-3.19] qrcode pairing fails with Invalid Argument

pankore, 2022-05-09:
#### Problem
* TC-DD-3.19 Step 2
* ./chip-tool pairing qrcode 1234 MT:-24J042C00KA0648G00
* Got Invalid argument error

[chiptool.txt](https://github.com/project-chip/connectedhomeip/files/8648983/chiptool.txt)

cecille, 2022-05-17:
Is this over BLE? Commissioning is failing because there are no network credentials given.
```
[1652074038.396546][3131:3136] CHIP:CTL: Successfully finished commissioning step 'SendNOC'
[1652074038.396569][3131:3136] CHIP:CTL: Required network information not provided in commissioning parameters
```
Suggest using qrcode-wifi or qrcode-thread variant as required.

pankore, 2022-05-18:
Thanks for clarifying and yes, it is over BLE. I have tried qrcode-wifi and it works.

closed 2022-05-18 by pankore