project-chip/connectedhomeip #18238: [TC-OO-2.4] Reboot command is failing for On/Off cluster

manjunath-grl, 2022-05-10:
Summary:
QA team tested [TC-OO-2.4] using Raspi against All Clusters App. we are using Reboot command for Power off and on the DUT but showing " ./src/app/tests/suites/commands/system/scripts/Reboot.py: not found" when executed
[Semi_Automated_TC_log.txt](https://github.com/project-chip/connectedhomeip/files/8657588/Semi_Automated_TC_log.txt)

Testplan screenshot:
[image]

Execution logs:
[image]

bzbarsky-apple, 2022-05-22:
@manjunath-grl Is the current working directory when running whatever is running here the root of the Matter tree?  The log does not show the actual commands being run, but I suspect the answer is "no".  And it needs to be for Reboot to work correctly as things are currently implemented.

@cjandhyala

cjandhyala, 2022-05-25:
@manjunath-grl Could you re-test this on latest ToT ?

manjunath-grl, 2022-06-16:
Tested with YAML script on ToT issue not observed