project-chip/connectedhomeip #18243: Pressure Measurement cluster optional attribute showing status = 0x86 (UNSUPPORTED_ATTRIBUTE)

manjunath-grl, 2022-05-10:
Summary:
[TC-PRS-1.1] as per testplan need to read optional attribute constraints checks when executed got status = 0x86 (UNSUPPORTED_ATTRIBUTE) response
Spec screenshot:
[image]
Testplan screenshot:
[image]
Execution logs:
[image]
[exe_log.txt](https://github.com/project-chip/connectedhomeip/files/8659694/exe_log.txt)

bzbarsky-apple, 2022-05-17:
@manjunath-grl That test plan says those tests are conditioned on some PICS checks (PICS_A_PRESSURE_SCALED_MAX, etc).  Are the tests in fact conditioned on those PICS checks?

manjunath-grl, 2022-05-17:
> @manjunath-grl That test plan says those tests are conditioned on some PICS checks (PICS_A_PRESSURE_SCALED_MAX, etc). Are the tests in fact conditioned on those PICS checks?

I even tried with these PICS condition getting same output: (UNSUPPORTED_ATTRIBUTE)
[image]

bzbarsky-apple, 2022-05-17:
@manjunath-grl `A_PRESSURE_SCALED` is not a defined PICS value for our YAML tests.

It sounds like we should define the PICS value, use it in the YAML, set it correctly for the test app if it does not implement this functionality yet.

closed 2022-06-16 by manjunath-grl