project-chip/connectedhomeip #18244: [TC-DIAGTH-NW-2.1] configured channel attribute value is failing in darwin

kowsisoundhar12, 2022-05-10:
Summary:
As per Testplan need to configure Channel  attribute value in config section ,but in Darwin configured discriminator value is Failing but its passing in linux environment

Testplan Screenshot:
[image]

Yaml Screenshot:
[image]

Build Error:
[image]
[image]
[TC-DIAG-TH-NW-logs.zip](https://github.com/project-chip/connectedhomeip/files/8659856/TC-DIAG-TH-NW-logs.zip)

manjunath-grl, 2022-05-11:
Its effecting [TC-MC-8.1]
[image]
[image]

manjunath-grl, 2022-05-11:
https://github.com/project-chip/connectedhomeip/pull/18267

bzbarsky-apple, 2022-05-11:
See https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/5195

The expected behavior here should be that the attributes are nullable and darwin returns null, since it's not on a Thread network....

kowsisoundhar12, 2022-05-30:
this issue has been fixed ,hence closing this issue