project-chip/connectedhomeip #18245: [TE9] [TC-IDM-4.2] Multiple subscriptions to the same attribute only generate a report for one subscription

sstruchtrup, 2022-05-10:
TC-IDM-4.2, step 17

**Expected behaviour** as per the TE9 excel testplan:

_TH sends a subscription request action for an attribute to the DUT with the KeepSubscriptions flag set to True. Activate the subscription between DUT and the TH. Initiate another subscription request action to the DUT for the same attribute with the KeepSubscriptions flag set to True. Change both the attribute values on the DUT._

**Actual behaviour:**
Only one subscription remains active.

This was observed during TE9 while testing our Matter bridge, but confirmed by testing against the example lighting app.

Testing against commit cfc35951be66a664a6efdadea56d1b8ea6e63e96 (tag: TE9)

**Steps to reproduce:**

```
# chip-tool interactive start
>>> onoff subscribe on-off 1 30 1 1 --keepSubscriptions 1
...
[1652172116.499411][1562:1563] CHIP:DMG: SubscribeResponseMessage =
[1652172116.499464][1562:1563] CHIP:DMG: {
[1652172116.499494][1562:1563] CHIP:DMG:        SubscriptionId = 0x7b033dff9bea8a25,
[1652172116.499534][1562:1563] CHIP:DMG:        MinIntervalFloorSeconds = 0x1,
[1652172116.499574][1562:1563] CHIP:DMG:        MaxIntervalCeilingSeconds = 0x1e,
[1652172116.499669][1562:1563] CHIP:DMG:        InteractionModelRevision = 1
[1652172116.499707][1562:1563] CHIP:DMG: }
...
>>> onoff subscribe on-off 1 30 1 1 --keepSubscriptions 1
[1652172138.997394][1562:1563] CHIP:DMG: SubscribeResponseMessage =
[1652172138.997425][1562:1563] CHIP:DMG: {
[1652172138.997451][1562:1563] CHIP:DMG:        SubscriptionId = 0x52f08c644642b534,
[1652172138.997495][1562:1563] CHIP:DMG:        MinIntervalFloorSeconds = 0x1,
[1652172138.997617][1562:1563] CHIP:DMG:        MaxIntervalCeilingSeconds = 0x1e,
[1652172138.997747][1562:1563] CHIP:DMG:        InteractionModelRevision = 1
[1652172138.997902][1562:1563] CHIP:DMG: }
```

Observed behaviour: only the second subscription is active:

```
[1652172289.013168][1562:1563] CHIP:EM: Received message of type 0x5 with protocolId (0, 1) and MessageCounter:5891225 on exchange 51711r
[1652172289.013255][1562:1563] CHIP:EM: Handling via exchange: 51711r, Delegate: 0x55a2d3ed0c68
[1652172289.013334][1562:1563] CHIP:DMG: ReportDataMessage =
[1652172289.013358][1562:1563] CHIP:DMG: {
[1652172289.013379][1562:1563] CHIP:DMG:        SubscriptionId = 0x52f08c644642b534,
[1652172289.013401][1562:1563] CHIP:DMG:        InteractionModelRevision = 1
[1652172289.013420][1562:1563] CHIP:DMG: }
...
[1652172378.796086][1562:1563] CHIP:DMG: ReportDataMessage =
[1652172378.796115][1562:1563] CHIP:DMG: {
[1652172378.796143][1562:1563] CHIP:DMG:        SubscriptionId = 0x52f08c644642b534,
[1652172378.796164][1562:1563] CHIP:DMG:        AttributeReportIBs =
[1652172378.796194][1562:1563] CHIP:DMG:        [
[1652172378.796214][1562:1563] CHIP:DMG:                AttributeReportIB =
[1652172378.796242][1562:1563] CHIP:DMG:                {
[1652172378.796262][1562:1563] CHIP:DMG:                        AttributeDataIB =
[1652172378.796285][1562:1563] CHIP:DMG:                        {
[1652172378.796308][1562:1563] CHIP:DMG:                                DataVersion = 0x88d11cd2,
[1652172378.796331][1562:1563] CHIP:DMG:                                AttributePathIB =
[1652172378.796354][1562:1563] CHIP:DMG:                                {
[1652172378.796388][1562:1563] CHIP:DMG:                                        Endpoint = 0x1,
[1652172378.796412][1562:1563] CHIP:DMG:                                        Cluster = 0x6,
[1652172378.796437][1562:1563] CHIP:DMG:                                        Attribute = 0x0000_0000,
[1652172378.796459][1562:1563] CHIP:DMG:                                }
[1652172378.796485][1562:1563] CHIP:DMG:
[1652172378.796510][1562:1563] CHIP:DMG:                                        Data = true,
[1652172378.796535][1562:1563] CHIP:DMG:                        },
[1652172378.796567][1562:1563] CHIP:DMG:
[1652172378.796587][1562:1563] CHIP:DMG:                },
[1652172378.796615][1562:1563] CHIP:DMG:
[1652172378.796634][1562:1563] CHIP:DMG:        ],
[1652172378.796662][1562:1563] CHIP:DMG:
[1652172378.796682][1562:1563] CHIP:DMG:        InteractionModelRevision = 1
[1652172378.796701][1562:1563] CHIP:DMG: }
[1652172378.796813][1562:1563] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0006 Attribute 0x0000_0000 DataVersion: 2295405778
[1652172378.796844][1562:1563] CHIP:TOO:   OnOff: TRUE
```

sstruchtrup, 2022-05-10:
If the expected behaviour is correct, this could either be a bug in the Matter SDK implementation or in chip-tool, not sending the KeepSubscriptions flag.

doublemis1, 2022-05-19:
This issue also Impacts all tests where a subscription to events needs to be done. Such as IDM-7.1 IDM -6.2 IDM-6.1

bzbarsky-apple, 2022-05-21:
I think this is specifically a chip-tool issue.  It seems to have a single `mSubscribeClient`, so when a new subscription is issued it kills off the old one.  The server is in fact sending the right data reports, but there is no client to receive them.

@vivien-apple If the intent is to support multiple parallel subscriptions, then we should probably have a vector of ReadClients in `InteractionModelReports` and just keep adding them.  And if we fix OnDone to say which read client is done, maybe we could remove them from there.

mrjerryjohns, 2022-05-24:
Yes, this is working fine in the REPL (just validated this), so presumably, a chip-tool issue.

closed 2022-05-24 by andy31415