project-chip/connectedhomeip #18248: [TE9][TC-GC-2.1] some test steps need returned status UNSUPPORTED_WRITE(0x88) but (UNSUPPORTED_ATTRIBUTE)(0x86)

wilbur-se, 2022-05-10:
#### Problem
<what's wrong or missing, please include any applicable:
 * expected behavior
 * actual behavior
 * steps to reproduce
 * system configuration
 * screenshots, images, logs, etc.
 >
test step #31, #34, #37, #40, #43, #46, #52, #54, #57 send command "./chip-tool basic write-by-id" but use wrong attribute ID, cause returned status is not expected UNSUPPORTED_WRITE(0x88)

 #### Proposed Solution
<suggested fix, suggested enhancement>

just remove prefix "0x" as decimal attribute ID for this steps, or use right hex attribute ID

bzbarsky-apple, 2022-05-21:
Sounds like an issue in the test plan (which one?), not in the SDK.

Please file in https://github.com/CHIP-Specifications/chip-test-plans/ ?

@cjandhyala

closed 2022-05-21 by bzbarsky-apple

wilbur-se, 2022-05-24:
> In particular, @wilbur-se I don't see any write-by-id in any of the TC-GC steps, so which test is the above referencing?

Sorry  to write wrong title. It is TC-DM-2.1 not TC-GC2.1