project-chip/connectedhomeip #18253: [TC-IDM-1.2] Chip-tool needs to implement the  Optional argument commissioner-fabricid .

Rajashreekalmane, 2022-05-10:
**#### Problem**
As per [TC-IDM-1.2] Test-Plan test-step 5 says that Setup the TH such that there is no accessing fabric
 TH sends the Invoke Request Message to the DUT with a valid and fabric-scoped CommandDataIB for this step 
DUT should responds with  UNSUPPORTED_ACCESS Status Code(0x7e).
Currently Optional argument --commissioner-fabricid does not exist in chip-tool.

**PFA screenshot:-**
[image]

**Test-Plan reference:-**
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/interactiondatamodel.adoc#tc-idm-1-2-invoke-response-action-from-dut-to-th

**Spec reference:-**
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/data_model/Interaction-Model.adoc#commandstatusib

vivien-apple, 2022-05-18:
As discussed on slack can't you just do:

Launch chip-tool into interactive mode: `chip-tool interactive start`. Then once in interactive mode:

```
$ pairing qrcode-paseonly 0x12344321 MT:-24J029Q00KA0648G00
$ onoff toggle 0x12344321 1 # check that this command works
$ operationalcredentials update-fabric-label "Hello" 0x12344321 0 # check that this command fails
``

@mrjerryjohns or do you have some other means of doing this particular test step ?

Rajashreekalmane, 2022-05-19:
@vivien-apple Thank you.

closed 2022-05-19 by Rajashreekalmane