project-chip/connectedhomeip #18272: TE9 [TC-DD-3.14/17] `chip-tool-darwin` allows pairing despite invalid Onboarding Payload data

liamgonyea, 2022-05-10:
### Problem
 
 ####  [TC-DD-3.14]
`chip-tool-darwin` allows a TH=`chip-all-clusters-app` accessory to pair using an Onboarding Payload that does not match the TH's Onboarding Payload.

**Step 2.b)** `chip-tool-darwin` allows pairing to `chip-all-clusters-app` when QR code contains invalid version string=2 (spec says it should be 000)
`command: ./out/chip-tool/chip-tool pairing qrcode 1 MT:034J029Q00KA0648G00`

**Step 3.b)** `chip-tool-darwin`  allows pairing to `chip-all-clusters-app` when QR code contains user-intent flow but device uses custom commissioning flow
`command: ./out/chip-tool/chip-tool pairing qrcode 1 MT:-24J06VO00KA0648G00`

**Step 3.d)** `chip-tool-darwin` allows pairing to `chip-all-clusters-app` when QR code contains custom commissioning flow but device uses standard commissioning flow
`command: ./out/chip-tool/chip-tool pairing qrcode 1 MT:-24J029Q00KA0648G00`

**Step 3.f)** `chip-tool-darwin`  allows pairing to `chip-all-clusters-app` when QR code contains Reserved commissioning flow but device uses custom commissioning flow
`command: ./out/chip-tool/chip-tool pairing qrcode 1 MT:-24J0-OR00KA0648G00`

**Step 4.b)** `chip-tool-darwin`  allows pairing to `chip-all-clusters-app` when QR code contains BLE device discovery but device uses IP device discovery
`command: ./out/chip-tool/chip-tool pairing qrcode 1 MT:-24J0YXE00KA0648G00`

**Step 4.f)** `chip-tool-darwin` allows pairing to `chip-all-clusters-app`when QR code contains the Reserved discovery capability but device supports IP discovery
`command: ./out/chip-tool-darwin/chip-tool-darwin pairing qrcode 1 MT:-24J0EZA10KA0648G00`

 ####  [TC-DD-3.17]
DUT=`chip-tool-darwin` pairs to TH=`chip-all-clusters-app` even though the Onboarding Payload used for pairing does not match the Onboarding Payload for the TH. The payloads differ in one component per pairing attempt for the following 2 components of the payload:
1.  A different (invalid) Vendor ID 
2. A different (invalid) Product ID  

**Step 6.b)** Invalid Vendor IDs (0xFFF1, 0xFFF2, 0xFFF3, 0xFFF4)
Error: `chip-tool-darwin` pairs to `chip-all-clusters-app` with each of these invalid Vendor IDs. _This behavior is expected of non-darwin `chip-tool` for development purposes but `chip-tool-darwin` should not allow for the pairing_

 (0xFFF1)
command:  `./out/chip-tool-darwin/chip-tool-darwin pairing manualcode 1 749701123365521327694`

 (0xFFF2)
 command:   `./out/chip-tool-darwin/chip-tool-darwin pairing manualcode 1 749701123365522327692`

 (0xFFF3)
 command:   `./out/chip-tool-darwin/chip-tool-darwin pairing manualcode 1 749701123365523327697`

 (0xFFF4)
 command:   `./out/chip-tool-darwin/chip-tool-darwin pairing manualcode 1 749701123365524327693`

**Step 7.b)** Invalid Product ID (0x0000)
command:   `./out/chip-tool-darwin/chip-tool-darwin pairing manualcode 1 749701123365521000006`

 ### Test Cases
[3.3.14. [TC-DD-3.14] Commissioning Flow - QR Code - Negative Scenario [DUT - Commissioner]
](https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/devicediscovery.adoc#test-procedure-30)

[3.3.17. [TC-DD-3.17] Commissioning Flow - 21-digit Manual Pairing Code - Negative Scenario [DUT - Commissioner]
](https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/devicediscovery.adoc#3317-tc-dd-317-commissioning-flow---21-digit-manual-pairing-code---negative-scenario-dut---commissioner)

 ### Full Logs
[TC-DD-3.14 Error Logs.txt](https://github.com/project-chip/connectedhomeip/files/8664359/TC-DD-3.14.Error.Logs.txt)
[TC-DD-3.17 Error Logs .txt](https://github.com/project-chip/connectedhomeip/files/8664356/TC-DD-3.17.Error.Logs.txt)

krypton36, 2022-05-13:
@liamgonyea I am not seeing these issue on chip-tool-darwin, but I do see them on chip-tool. Can you please confirm it is chip-tool-darwin that you are seeing these issue?

liamgonyea, 2022-05-19:
@krypton36 I can confirm that they are issues on chip-tool-darwin.

### TC-DD-3.14 Testing Update
The same commands above are failing on commit: `d79a0e3784b4e829b5df7e50afc81fe3cb067405` 
[TC-DD-3.14 Logs.txt](https://github.com/project-chip/connectedhomeip/files/8721916/TC-DD-3.14.Logs.txt)

liamgonyea, 2022-05-19:
### TC-DD-3.17 Testing Update
The same commands above are failing on commit: `d79a0e3784b4e829b5df7e50afc81fe3cb067405` 
No additional failures. 

[TC-DD-3.17 Logs.txt](https://github.com/project-chip/connectedhomeip/files/8731034/TC-DD-3.17.Logs.txt)

bzbarsky-apple, 2022-05-25:
Should re-evaluate once #17205 is fixed.

bzbarsky-apple, 2022-05-26:
With adjustments to tip (using "code" instead of "qrcode" or "manualcode"):

> command: ./out/chip-tool/chip-tool pairing qrcode 1 MT:034J029Q00KA0648G00

Throws an error after #18606

> command: ./out/chip-tool/chip-tool pairing qrcode 1 MT:-24J06VO00KA0648G00

Not sure; don't have a device with custom commissioning flow.

> command: ./out/chip-tool/chip-tool pairing qrcode 1 MT:-24J029Q00KA0648G00

Still allows commissioning.

> command: ./out/chip-tool/chip-tool pairing qrcode 1 MT:-24J0-OR00KA0648G00

Throws an error after #18606

> command: ./out/chip-tool/chip-tool pairing qrcode 1 MT:-24J0YXE00KA0648G00

The test plan is wrong (@cjandhyala).  Per spec, just because the QR code says BLE only does not mean we shouldn't try IP, because this could be a basic commissioning window on an already-commissioned device.  So always doing IP discovery is correct.

> command: ./out/chip-tool-darwin/chip-tool-darwin pairing qrcode 1 MT:-24J0EZA10KA0648G00

Throws an error after #18606

> command: ./out/chip-tool-darwin/chip-tool-darwin pairing manualcode 1 749701123365521327694
> command: ./out/chip-tool-darwin/chip-tool-darwin pairing manualcode 1 749701123365522327692
> command: ./out/chip-tool-darwin/chip-tool-darwin pairing manualcode 1 749701123365523327697
> command: ./out/chip-tool-darwin/chip-tool-darwin pairing manualcode 1 749701123365524327693

Disallowing the pairing here (on the framework level) would mean that chip-tool-darwin and the iOS test app would not be usable with the example apps in the tree.  Can you please point me to the spec requirement that these vendor ids must not be commissionable by a commissioner?

> command: ./out/chip-tool-darwin/chip-tool-darwin pairing manualcode 1 749701123365521000006

Throws an error after #18606

sowmyassp, 2022-05-26:
@bzbarsky-apple 
./chip-tool pairing code 1 MT:-24J029Q00KA0648G00 is failing

andy31415, 2022-05-26:
Removed cert blocker, leaving #17205 as the equivalent cert blocker. I believe the fix for that will likely fix both, however for cert blocking we are blocking in `chip-tool` not on `chip-tool-darwin`

woody-apple, 2022-06-14:
Tagging 'request cert blocker' as we may need to retest, and verify we don't have a cert blocker here.

woody-apple, 2022-06-14:
Remaining items here to talk about are user notifications during app component cert.

andy31415, 2022-06-15:
To me it looks like #18686 fixed chip tool for most cases and the rest of failures seem like spec issues (based on comment from @bzbarsky-apple above). I believe this is not a cert blocker.

woody-apple, 2022-06-15:
Cert Blocker Review: Assigning to myself to break out.

liamgonyea, 2022-07-05:
Closing ticket, above issues are addressed in a [chip-test-plans PR:](https://github.com/CHIP-Specifications/chip-test-plans/pull/1593#issue-1288082131) or invalid test plan steps

closed 2022-07-05 by liamgonyea