project-chip/connectedhomeip #18307: [SQA]-THREAD In TC-DM-2.1 few optional non global attribute throws an error 0x86((UNSUPPORTED_ATTRIBUTE)) instead 0x88(UNSUPPORTED_WRITE(0x88)

Survensa, 2022-05-11:
**Problem**
While verifying TC-DM-2.1 few optional non global attribute to write chip-tool throws an error 0x86(UNSUPPORTED_ATTRIBUTE)  and  0x87(CONSTRAINT_ERROR)  instead 0x88(UNSUPPORTED_WRITE) those attributes are mentioned below.

1.     CapabilityMinima attribute
2.     Reachable attribute
3.     UniqueID attribute
4.     SerialNumber_test attribute
5.     ProductLabel attribute
6.     SoftwareVersionString attribute
7.     versionstring attribute

**expected behavior:-**  UNSUPPORTED_WRITE(0x88) 

**actual behavior:-**     CONSTRAINT_ERROR(0x87) or UNSUPPORTED_ATTRIBUTE (0x86)

**Steps to reproduce:** 

```
sudo ./chip-tool basic write-by-id 0x14 partlabel 76 0
sudo ./chip-tool basic write-by-id 0x15 newserialnumber 76 0
sudo ./chip-tool basic write-by-id 0x17 0 76 0
sudo ./chip-tool basic write-by-id 0x18 newid 76 0
sudo ./chip-tool basic write-by-id 0x19 4 76 0
sudo ./chip-tool basic write-by-id 0x10 versionstring 76 0
```

**Platform:** Thread

**Reference:** : [Test-plan](https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/devicemanagement.adoc#tc-dm-2-1-basic-information-cluster-server-attributes)

**Log:** 

> [1652261426.226783][15059:15064] CHIP:EM: Rxd Ack; Removing MessageCounter:14333353 from Retrans Table on exchange 13933i
[1652261426.226843][15059:15064] CHIP:EM: Removed CHIP MessageCounter:14333353 from RetransTable on exchange 13933i
[1652261426.226918][15059:15064] CHIP:DMG: WriteClient moving to [ResponseRe]
[1652261426.227018][15059:15064] CHIP:DMG: WriteResponseMessage =
[1652261426.227079][15059:15064] CHIP:DMG: {
[1652261426.227132][15059:15064] CHIP:DMG: 	AttributeStatusIBs =
[1652261426.227206][15059:15064] CHIP:DMG: 	[
[1652261426.227266][15059:15064] CHIP:DMG: 		AttributeStatusIB =
[1652261426.227347][15059:15064] CHIP:DMG: 		{
[1652261426.227412][15059:15064] CHIP:DMG: 			AttributePathIB =
[1652261426.227489][15059:15064] CHIP:DMG: 			{
[1652261426.227565][15059:15064] CHIP:DMG: 				Endpoint = 0x0,
[1652261426.227649][15059:15064] CHIP:DMG: 				Cluster = 0x28,
[1652261426.227732][15059:15064] CHIP:DMG: 				Attribute = 0x0000_0010,
[1652261426.227911][15059:15064] CHIP:DMG: 			}
[1652261426.228008][15059:15064] CHIP:DMG: 				
[1652261426.228081][15059:15064] CHIP:DMG: 			StatusIB =
[1652261426.228158][15059:15064] CHIP:DMG: 			{
[1652261426.228232][15059:15064] CHIP:DMG: 				status = 0x87 (CONSTRAINT_ERROR),
[1652261426.228310][15059:15064] CHIP:DMG: 			},
[1652261426.228384][15059:15064] CHIP:DMG: 			
[1652261426.228451][15059:15064] CHIP:DMG: 		},
[1652261426.228523][15059:15064] CHIP:DMG: 		
[1652261426.228581][15059:15064] CHIP:DMG: 	],
[1652261426.228652][15059:15064] CHIP:DMG: 	
[1652261426.228709][15059:15064] CHIP:DMG: 	InteractionModelRevision = 1
[1652261426.228766][15059:15064] CHIP:DMG: }
[1652261426.228906][15059:15064] CHIP:TOO: Response Failure: IM Error 0x00000587: General error: 0x87 (CONSTRAINT_ERROR)
[1652261426.228992][15059:15064] CHIP:DMG: WriteClient moving to [AwaitingDe]
[1652261426.229103][15059:15064] CHIP:EM: Sending Standalone Ack for MessageCounter:2678974 on exchange 13933i

cjandhyala, 2022-05-24:
@Survensa Pls use correct command codes

Survensa, 2022-05-25:
Its working for the command `./chip-tool basic write-by-id 15 newserialnumber 49 0`

closed 2022-05-25 by Survensa