project-chip/connectedhomeip #18316: Subscribe event by id brings device to a halt on nrfconnect platform

fessehaeve, 2022-05-11:
#### Problem
Using `./chip-tool powersource subscribe-event-by-id 1 120 3600 2 1 `with the lock-app on nrfconnect platform results in 

```
12:32:56.240   1038.530      191  0 uart:~$ I: 66224 [EM]Received message of type 0x30 with protocolId (0, 0) and MessageCounter:1688597937 on exchange 43973r
12:33:46.276   1088.566    50036  0 D: 66234 [EM]Handling via exchange: 43973r, Delegate: 0x200036b0
12:33:46.282   1088.572        6  0 I: 66240 [IN]CASE Server received Sigma1 message. Starting handshake. EC 0x20004b28
12:33:46.294   1088.584       12  0 D: 66247 [SC]Waiting for Sigma1 msg
12:33:46.294   1088.584        0  0 I: 66251 [IN]CASE Server disabling CASE session setups
12:33:46.300   1088.590        6  0 I: 66255 [SC]Received Sigma1 msg
12:33:46.305   1088.595        5  0 D: 66259 [SC]Found MRP parameters in the message
12:33:46.305   1088.595        0  0 D: 66263 [SC]Peer assigned session key ID 57822
12:33:46.669   1088.959      364  0 I: 66622 [SC]CASE matched destination ID: fabricIndex 1, NodeID 0x0000000000000002
12:33:46.896   1089.186      227  0 D: 66851 [SC]Including MRP parameters
12:33:46.902   1089.192        6  0 I: 66855 [IN]Prepared unauthenticated message 0x20004a44 to 0x0000000000000000 (0)  of type 0x31 and protocolId (0, 0) on exchange 43973r with MessageCounter:841321732.
12:33:46.917   1089.207       15  0 I: 66870 [IN]Sending unauthenticated msg 0x20004a44 with MessageCounter:841321732 to 0x0000000000000000 at monotonic time: 0000000000010536 msec
12:33:46.928   1089.218       11  0 I: 66885 [SC]Sent Sigma2 msg
12:33:47.052   1089.342      124  0 I: 67005 [EM]Received message of type 0x32 with protocolId (0, 0) and MessageCounter:1688597938 on exchange 43973r
12:33:47.057   1089.347        5  0 D: 67015 [EM]Found matching exchange: 43973r, Delegate: 0x200036c4
12:33:47.068   1089.358       11  0 D: 67021 [EM]Rxd Ack; Removing MessageCounter:841321732 from Retrans Table on exchange 43973r
12:33:47.073   1089.363        5  0 D: 67029 [EM]Removed CHIP MessageCounter:841321732 from RetransTable on exchange 43973r
12:33:47.079   1089.369        6  0 I: 67037 [SC]Received Sigma3 msg
12:33:47.091   1089.381       12  0 E: 67044 [SC]The device does not support GetClock_RealTimeMS() API. This will eventually result in CASE session setup failures.
12:33:47.594   1089.884      503  0 D: 67547 [SC]Sending status report. Protocol code 0, exchange 43973
12:33:47.600   1089.890        6  0 I: 67554 [IN]Prepared unauthenticated message 0x20004a44 to 0x0000000000000000 (0)  of type 0x40 and protocolId (0, 0) on exchange 43973r with MessageCounter:841321733.
12:33:47.615   1089.905       15  0 I: 67569 [IN]Sending unauthenticated msg 0x20004a44 with MessageCounter:841321733 to 0x0000000000000000 at monotonic time: 00000000000107F0 msec
12:33:47.630   1089.920       15  0 D: 67583 [IN]New secure session created for device <000000000001B669, 1>, LSID:20591 PSID:57822!
12:33:47.639   1089.929        9  0 I: 67592 [IN]CASE Session established to peer: <000000000001B669, 1>
12:33:47.645   1089.935        6  0 I: 67599 [IN]CASE Server enabling CASE session setups
12:33:47.651   1089.941        6  0 D: 67604 [IN]Received a duplicate message with MessageCounter:1688597938 on exchange 43973r
12:33:47.659   1089.949        8  0 I: 67612 [EM]Received message of type 0x32 with protocolId (0, 0) and MessageCounter:1688597938 on exchange 43973r
12:33:47.671   1089.961       12  0 D: 67624 [EM]Found matching exchange: 43973r, Delegate: (nil)
12:33:47.678   1089.968        7  0 D: 67631 [EM]Forcing tx of solitary ack for duplicate MessageCounter:1688597938 on exchange 43973r
12:33:47.684   1089.974        6  0 D: 67640 [EM]Sending Standalone Ack for MessageCounter:1688597938 on exchange 43973r
12:33:47.695   1089.985       11  0 I: 67648 [IN]Prepared unauthenticated message 0x2002b2a0 to 0x0000000000000000 (0)  of type 0x10 and protocolId (0, 0) on exchange 43973r with MessageCounter:841321734.
12:33:47.710   1090.000       15  0 I: 67663 [IN]Sending unauthenticated msg 0x2002b2a0 with MessageCounter:841321734 to 0x0000000000000000 at monotonic time: 000000000001084F msec
12:33:47.722   1090.012       12  0 I: 67678 [EM]Received message of type 0x3 with protocolId (0, 1) and MessageCounter:14897363 on exchange 43974r
12:33:47.735   1090.025       13  0 D: 67688 [EM]Handling via exchange: 43974r, Delegate: 0x20007974
12:33:47.739   1090.029        4  0 D: 67694 [IM]Received Subscribe request
12:33:47.744   1090.034        5  0 E: ***** BUS FAULT *****
12:33:47.744   1090.034        0  0 E:   Precise data bus error
12:33:47.744   1090.034        0  0 E:   BFAR Address: 0x65206e6f
12:33:47.750   1090.040        6  0 E: r0/a1:  0x00000000  r1/a2:  0x00000021  r2/a3:  0x65206e6f
12:33:47.756   1090.046        6  0 E: r3/a4:  0x61686378 r12/ip:  0x00000000 r14/lr:  0x00051283
12:33:47.761   1090.051        5  0 E:  xpsr:  0x21000000
12:33:47.761   1090.051        0  0 E: s[ 0]:  0x00000000  s[ 1]:  0x00000000  s[ 2]:  0x00000000  s[ 3]:  0x00000000
12:33:47.767   1090.057        6  0 E: s[ 4]:  0x00000000  s[ 5]:  0x00000000  s[ 6]:  0x00000000  s[ 7]:  0x00000000
12:33:47.778   1090.068       11  0 E: s[ 8]:  0x00000000  s[ 9]:  0x00000002  s[10]:  0x00000000  s[11]:  0x00000000
12:33:47.783   1090.073        5  0 E: s[12]:  0x00000000  s[13]:  0x00000000  s[14]:  0x00000000  s[15]:  0x00000000
12:33:47.789   1090.079        6  0 E: fpscr:  0x000a052d
12:33:47.795   1090.085        6  0 E: Faulting instruction address (r15/pc): 0x0005129c
12:33:47.800   1090.090        5  0 E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
12:33:47.800   1090.090        0  0 E: Current thread: 0x2000c400 (CHIP)
12:33:47.804   1090.094        4  0 E: Halting system

```
The cause seems to be 

```
➜  git:(f34761eee) ✗ arm-none-eabi-addr2line -i -C -f -e examples/lock-app/nrfconnect/build/zephyr/zephyr.elf 0x0005129c
chip::TLV::TLVReader::ReadElement()
src/lib/core/CHIPTLVReader.cpp:666
```

Need to nail down the cause.

bzbarsky-apple, 2022-05-11:
@yunhanw-google

bluebin14, 2022-05-16:
Any subscribe interaction crashes, e.g. >>> generaldiagnostics subscribe-event radio-fault-change 1 100 12344321 1

Damian-Nordic, 2022-05-18:
I think it might have been fixed on master: https://github.com/project-chip/connectedhomeip/pull/17841/files. @kkasperczyk-no is going to test that.

fessehaeve, 2022-05-18:
I had done a test on ToT before a few days and couldn't reproduce the issue. It is a TE9 issue which is  solved after the cutoff.

kkasperczyk-no, 2022-05-18:
Yes the https://github.com/project-chip/connectedhomeip/commit/fb68954f39d46fea37e7c0bcc9e9da7f8ea7f71a commit solved the issue on master.

Damian-Nordic, 2022-05-18:
@fessehaeve shall we close the issue then?

closed 2022-05-18 by fessehaeve