project-chip/connectedhomeip #18317: [TE9][TC-LC-2.3] UserLabel cannot write a list into the attribute label-list

jmeg-sfy, 2022-05-11:
#### Problem
<what's wrong or missing, please include any applicable:
 * /chip-tool userlabel write label-list '[{"label":"bedroom","value":"bedroom1"},{"label":"orientation","value":"east"},{"label":"floor","value":"3"},{"label":"direction","value":"down"}]
 * Give back the status = 0x01 (FAILURE),
 >

 #### Proposed Solution
fix the XML and cluster UserLabel

bzbarsky-apple, 2022-05-21:
@jmeg-sfy What platform is the server in this case?  user-label is implemented on some platforms but not others.

andy31415, 2022-05-25:
Assigning to myself. I believe as part of changes to #18687 we are pulling in device labelling support as well. I will have to test this and eventually this has to be added to multiple platforms.

andy31415, 2022-05-25:
Validated that his currently works in nrf-light using this script:

```sh
#!/usr/bin/env bash

set -ex

function finish {
  echo "STATUS: $?"
}

trap finish EXIT

export NODEID=$RANDOM

export THREAD_CRED=hex:0e..... # added my thread credentials here
export CHIP_TOOL="./out/linux-x64-chip-tool-ipv6only/chip-tool"

log_file="run_log_$(date +'%Y-%m-%d_%H-%M-%S').txt"
rm /tmp/chip_*

echo "NODE ID USED: $NODEID" | tee $log_file

echo "=============================================================" >>$log_file
echo $CHIP_TOOL pairing ble-thread $NODEID $THREAD_CRED 20202021 3840 | tee -a $log_file
$CHIP_TOOL pairing ble-thread $NODEID $THREAD_CRED 20202021 3840 >>$log_file 2>&1

echo "=============================================================" >>$log_file
echo $CHIP_TOOL userlabel write label-list '[{"label":"bedroom","value":"bedroom2"},{"label":"orientation","value":"east"},{"label":"floor","value":"3"},{"label":"direction","value":"down"}]' $NODEID 0 | tee -a $log_file
$CHIP_TOOL userlabel write label-list '[{"label":"bedroom","value":"bedroom2"},{"label":"orientation","value":"east"},{"label":"floor","value":"3"},{"label":"direction","value":"down"}]' $NODEID 0 2>&1 | tee -a $log_file | grep -i status

echo "=============================================================" >>$log_file
echo $CHIP_TOOL userlabel read label-list $NODEID 0 | tee -a $log_file
$CHIP_TOOL userlabel read label-list $NODEID 0 2>&1 | tee -a $log_file | grep -A 18 'label list'
```

andy31415, 2022-05-25:
I can also confirm this fails for efr32-brd4161a-light which seems in line with my expectations: using the DeviceInfoProvider enables this, not having it makes it not pass.

I will add this for EFR however for future items I will need platform owners to update their examples.
I was also unable to test with nrf on lock, light or all clusters since they fail to BLE advertise for me.

jmeg-sfy, 2022-05-25:
> @jmeg-sfy What platform is the server in this case? user-label is implemented on some platforms but not others.

It was on EFR32 in my case

jmeg-sfy, 2022-05-25:
Did a copy of those from src/platform/Linux

	src/platform/EFR32/DeviceInfoProviderImpl.cpp
	src/platform/EFR32/DeviceInfoProviderImpl.h

andy31415, 2022-05-25:
I updated EFR32 light to use the newly moved examples/provides/DeviceInfoProviderImpl.h/cpp however this crashes for me :( Need to debug why.

andy31415, 2022-05-25:
> I updated EFR32 light to use the newly moved examples/provides/DeviceInfoProviderImpl.h/cpp however this crashes for me :( Need to debug why.

Found that storage has to be initialized (makse sense ... we store the labels). What is unclear is why this was working for me on nrf and linux which do not initialize storage. So I added the storage init to all examples now.

PR #18817 (2022-05-26): Device info provider efr32, some fixes for nrf and linux
#### Problem
EFR32 does not have a device info provider, so setting user labels and querying locales on demo apps do not work.

#### Change overview
- Add deviceinfoprovider for efr32 targets
- Ensure storage is initialized also on nrf and linux (efr32 was crashing, let me to finding that storage needs initialization)

Fixes #18317 

#### Testing
Ran EFR32 test for setting and reading user labels.
closed 2022-05-26 by andy31415