project-chip/connectedhomeip #18321: [TC-DGETH-2.2], [TC-DGETH-3.2] - Reset command fails

sowmyassp, 2022-05-11:
**Issue** : Reset command fails 

**Commit used** : 11587f5bf0bb9e621edd4c867c0bc942d804d1ce			

**Expected behavior** : status = 0x00 (SUCCESS),

**Actual behavior** : status = 0x01 (FAILURE)

**Steps to reproduce** :

- DUT side :

1. sudo rm -rf /tmp/chip_*
2. sudo ./chip-all-clusters-app 

- TH side :

1. ./chip-tool pairing ethernet 1 20202021  3840  fe80::e65f:1ff:fe0f:3baf  5540
2. ./chip-tool ethernetnetworkdiagnostics reset-counts 1 0

**Test platform** :
Chip-tool - RPI/MAC
DUT - RPI

**PFA Logs** : 
[reset_dut.txt](https://github.com/project-chip/connectedhomeip/files/8669697/reset_dut.txt)
[reset_cntrl.txt](https://github.com/project-chip/connectedhomeip/files/8669698/reset_cntrl.txt)

bzbarsky-apple, 2022-05-19:
This command's implementation tries to set the attributes in the in-memory attr store, but they don't live there (they're in the `DiagnosticDataProvider`, so that fails).  @yufengwangca

bzbarsky-apple, 2022-05-19:
That said, even with this fixed this will fail on platforms where there is no diagnostics data provider (i.e. not Linux).

closed 2022-05-19 by bzbarsky-apple