project-chip/connectedhomeip #18322: [TC-TSTAT-3.2]--AttributeList attribute missed to list the Mandatory TestEventTriggersEnabled attribute

Rajashreekalmane, 2022-05-11:
**Test-case ID** 
 [TC-DGGEN-1.1] Global attributes with server as DUT

**Problem:-**
AttributeList attribute missed to list the Mandatory TestEventTriggersEnabled attribute

*** **expected behavior****:- TH should read mandatory attributes (values 0, 1, 8) from DUT
*** actual behavior**:- TH missed to read Mandatory TestEventTriggersEnabled(8) attribute from DUT

***system configuration:-**
Chip-Tool: RPI
DUT: RPI
Platform : IP and Wifi
Commit-ID: 11587f5
Dut:all-clusters-app

**Chip-tool screenshot:-**
[image]

**Test-Plan reference:-**
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/general_diagnostics.adoc#tc-dggen-1-1-global-attributes-with-server-as-dut

**Spec Reference:-**
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/service_device_management/DiagnosticsGeneral.adoc

woody-apple, 2022-05-19:
Assigning to @cjandhyala to follow up.

andy31415, 2022-05-21:
Removed @cjandhyala - based on the comment from Vivien, this is a case of "attribute and command missing implementation". A developer can check and I imagine 2 paths:
  - if these are spec mandatory, we should implement
  - if these are spec optional, we will take the bug out of cert blocker as we do not implement optional at this time

bzbarsky-apple, 2022-05-23:
@andy31415 This attribute and command are mandatory per spec.

woody-apple, 2022-05-23:
Cert Blocker Review: Given this is a diagnostic cluster issue, assigning to @andy31415 for comment and assignment.

jaehs6sam, 2022-05-23:
This TestEventTriggers command was recently introduced to the spec and its implementation will be manufacturer specific.  Its key and test mode values have to be pre-provisioned by a manufacturer to client and sever devices.  And this command will be used only during certification testing and to the certification candidate DUT only.  Again, testing mode value implementation & interpretation is upto the manufacturer (may not use PICS) such that it is difficult for a 3rd party to validate testing modes unless it is informed by the manufacturer a prior.  I personally think that this command should be optional and be post 1.0 implementation at this juncture.

tcarmelveilleux, 2022-05-25:
Test event trigger will be used for OTA test plans

tehampson, 2022-05-27:
https://github.com/project-chip/connectedhomeip/pull/18852

PR #18807 (2022-05-26): Test Event Trigger implementation for General Diagnostics Cluster
#### Problem
* Partially fixes #18322

#### Change overview
* Interface for TestEventTriggerDelegate

#### Testing
Untested since it is an interface header file that is added
PR #18852 (2022-05-30): Use TestEventTriggerDelegate in General Diagnostics Cluster
#### Problem
* Fixes #18322 

#### Change overview
* Enable TestEventTriggersEnabled attribute on general diagnostics cluster
* Enable TestEventTrigger command on general diagnostics cluster

#### Testing
* Manually tested on linux host
  * Confirmed attribute is listed in chip-tool
  * Confirmed reading TestEventTriggersEnabled attribute returns expected false
  * Confirmed sending command with 0s EnableKey gives `CONSTRAINT_ERROR`
  * Confirmed sending command with incorrectly sized EnableKey gives `CONSTRAINT_ERROR`
closed 2022-05-30 by tehampson