project-chip/connectedhomeip #18327: Read request during OTA blocks device

marius-alex-tache, 2022-05-11:
#### Problem
I am trying to read attribute **update-state-progress** during OTA, but the device blocks some time after receiving the read request. From my analysis, it seems that there is a retransmission timer that is started with delay 0ms. This means the **ReliableMessageMgr::Timeout** is called without delay and is not expecting that, since the assert **VerifyOrDie(!entry->retainedBuf.IsNull())** fails. 
This lambda function from **src/messaging/ReliableMessageMgr.cpp**, line 339, sets the **nextWakeTime** to the min value of all nextRetransTime from the retransmission table entries:
```
mRetransTable.ForEachActiveObject([&](auto * entry) {
        if (entry->nextRetransTime < nextWakeTime)
        {
            nextWakeTime = entry->nextRetransTime;
            ChipLogDetail(ExchangeManager, "nextWakeTime set to nextRetransTime %" PRIu64 "ms", nextWakeTime.count());
        }
        return Loop::Continue;
    });
```
In my case, I saw that at least one of the nextRetransTime values is 0. Is a value of 0 for retransmission time an actual use case? 
Looking at the specification, I see in section **4.11.8. Parameters and Constants** the default values for MRP_RETRY_INTERVAL_IDLE/ACTIVE are set to 300ms.

This is the stack trace:
[image]

This is the exact line (121) where the assert fails:
[image]

I have added some logging messages to print some time values. Also, I've added some logging to see the values of nextRetransTime and retainedBuf when AddToRetransTable is called. This is the .diff file with my modifications:
[ReliableMessageMgrDiff.txt](https://github.com/project-chip/connectedhomeip/files/8670416/ReliableMessageMgrDiff.txt)

I will also attach the chip-tool and the device log files. Please note that the device log file has some tokenized strings that could not be detokenized:
[chip-tool.txt](https://github.com/project-chip/connectedhomeip/files/8670423/chip-tool.txt)
[device.txt](https://github.com/project-chip/connectedhomeip/files/8670424/device.txt)

 #### Proposed Solution
I have not found a solution yet, since I'm not sure about the retransmission time question formulated above.
If I enhance the lambda function **if** to check that **entry->nextRetransTime** is also different than 0, then the device will never block if it receives a read request:
```
if (entry->nextRetransTime != System::Clock::Timestamp::zero() && entry->nextRetransTime < nextWakeTime)
```
This was just a quick way for me to confirm that the nextRetransTime set to 0 was causing the actual issue.
Please let me know if you need more details.

kghost, 2022-06-07:
I doubt this problem is caused by multi-thread racing, here a found piece of logs in the device log:

```
[110999][D][SWU]BDX::SendMessage
[111001][D][EM]AddToRetransTable rentry->nextRetransTime 0ms
[111004][D][EM]AddToRetransTable rentry->retainedBuf is null 1
[111010][P][IN]Prepared secure message 0x0400CB9C to 0x0000000000000001 (1)  of type 0x10 and protocolId (0, 2) on exchange 38934i with MessageCounter:10255752.
[111018][P][IN]Sending encrypted msg 0x0400CB9C with MessageCounter:10255752 to 0x0000000000000001 (1) at monotonic time: 000000000001B3AA msec
[111028][D][DL]Thread packet SENT: UDP, len 86
[111038][D][DL]    src  FD11:33::975C:2FD7:60DD:A130, port 5540
[111049][D][DL]    dest FD11:33::EEDE:C569:AF09:1CA3, port 5540
    [111061][D][DL]Thread packet RCVD: UDP, len 646
    [111069][D][DL]    src  FD11:33::EEDE:C569:AF09:1CA3, port 48132
    [111080][D][DL]    dest FD11:33::975C:2FD7:60DD:A130, port 5540
    [111092][P][EM]Received message of type 0x32 with protocolId (0, 0) and MessageCounter:852183893 on exchange 17265r
[111099][D][EM]c530741ee28d02e401c8e48540
[111104][D][EM]dc8baf2000
[111108][D][EM]now is 111620ms
[111112][D][EM]nextWakeTime is 0ms
[111117][D][EM]nextWakeDelay set to 0ms
[111120][D][EM]2113dfdfe1979e9509e28d02e401
    [111126][D][EM]Removed CHIP MessageCounter:3064740367 from RetransTable on exchange 17265r
    [111133][D][EM]nextWakeTime set to nextAckTime 111850ms
    [111136][D][EM]dc8baf2000
    [111141][D][EM]now is 111650ms
    [111144][D][EM]nextWakeTime is 0ms
    [111147][D][EM]nextWakeDelay set to 0ms
    [111150][P][SC]Received Sigma3 msg
```

In line with timestamp 111092, the exchange manager is staring process a received packet before last send is finished, the last send is still processing its retrans timer until 111120

kghost, 2022-06-08:
There are 2 implementations for different platform: `GenericThreadStackManagerImpl_OpenThread_LwIP` or `UDPEndPointImplOT::handleUdpReceive` both of them calls receive packets in the wrong thread, which can cause racing

bukepo, 2022-06-08:
Discussed with @kghost , this may not be a Thread stack issue. When Thread stack is running, it acquires the OpenThread lock.

bzbarsky-apple, 2022-06-08:
So the general message flow here seems to be:

* Outgoing messages: Something calls into LwIP, which ends up calling its `output_ip6` callback, which calls `GenericThreadStackManagerImpl_OpenThread_LwIP<ImplClass>::SendPacket`.  The assumption is that at that point the LwIP lock is held and hence the call could be on any thread.  SendPacket takes the OpenThread lock and then does openthread things.
* Incoming messages: The OpenThread stack calls into `GenericThreadStackManagerImpl_OpenThread_LwIP<ImplClass>::ReceivePacket` with the OpenThread lock held, and I expect on the openthread stack.  This function has:
    ```
    // Deliver the packet to the input function associated with the LwIP netif.
    // NOTE: The input function posts the inbound packet to LwIP's TCPIP thread.
    // Thus there's no need to acquire the LwIP TCPIP core lock here.
    lwipErr = threadNetIf->input(pbuf, threadNetIf);
    ```

The `input` function is, afaict, `tcpip_input`, which tries to do different things in practice depending on whether `LWIP_TCPIP_CORE_LOCKING_INPUT` is defined.  On k32w, I _think_ that's not defined, so it tries to post the message to the "LwIP thread".

Now what's not clear to me is whether the "LwIP thread" is the same as the Matter thread....

Anyway, on the "LwIP thread" we run `ip_input` which either is or calls `ip6_input` (depending on build flags), which I assume calls `udp_input`.  And that lands us in `UDPEndPointImplLwIP::LwIPReceiveUDPMessage`, I think.  Which does a `ScheduleLambda` to do the actual message processing on the Matter thread.  So this part is all sane.

So the race between the "SENT" and "RCVD" messages is espected, but the "Received message of type 0x32" should be happening on the Matter thread, not the thread where the "RCVD" log happened, right?

Now that's assuming I traced this through correctly.  @marius-alex-tache can you please check whether incoming packets in your setup do in fact end up in `UDPEndPointImplLwIP::LwIPReceiveUDPMessage` after going through `GenericThreadStackManagerImpl_OpenThread_LwIP<ImplClass>::ReceivePacket` ?

What is extremely unclear to me is what thread the `BDX::SendMessage` is happening on.  I would _hope_ it's the Matter thread, but....

The other thing that might be useful here, @marius-alex-tache is to insert `assertChipStackLockedByCurrentThread();` in the places where we think things might be going awry.  Most immediately, at the place where `[111144][D][EM]nextWakeTime is 0ms` is being logged, since we think we are racing there.  I _think_ `chip_stack_lock_tracking` is enabled on k32w, and if so that might just get us a stack to whatever is reaching that code on the wrong thread.

marius-alex-tache, 2022-06-09:
I've added a `assertChipStackLockedByCurrentThread()` call inside `ReliableMessageMgr::StartTimer`, after the lambda function 
```
// When do we need to next wake up for ReliableMessageProtocol retransmit?
mRetransTable.ForEachActiveObject([&](auto * entry) {
    if (entry->nextRetransTime < nextWakeTime)
    {
        nextWakeTime = entry->nextRetransTime;
    }
    return Loop::Continue;
});
assertChipStackLockedByCurrentThread();
```
and the assert fails after receiving the first BDX message.
This is the device log:
```
[103633[U][ATM]Handling received BDX Message
[103636[U][ATM]Block
[103638[U][ATM]  Block Counter: 0
[103641[U][ATM]  Data Length: 1024
[103711[U][ATM]Sending BDX Message
[103714[U][ATM]BlockQuery
[103716[U][ATM]  Block Counter: 1
[103719[D][SWU]BDX::SendMessage
[103723[P][IN]Prepared secure message 0x0400E36C to 0x0000000000000001 (1)  of type 0x10 and protocolId (0, 2) on exchange 2517i with MessageCounter:3700478.
[103731[P][IN]Sending encrypted msg 0x0400E36C with MessageCounter:3700478 to 0x0000000000000001 (1) at monotonic time: 0000000000018ECA msec
[103740[E][DL]Chip stack locking error at '../../third_party/connectedhomeip/src/messaging[...]:%d'. Code is unsafe/racy
[103752[E][-]chipDie chipDie chipDie
[103958[D][IN]UDP Message Received packet nb : 25 SrcAddr : fd11:33:0:0:a9e7:1f56:d76e:985[5540] DestAddr : fd11:33:0:0:3[...][%d] Payload Length %d
[103970[P][EM]Received message of type 0x10 with protocolId (0, 0) and MessageCounter:8939332 on exchange 2517i
[103975[D][EM]c530741eaa27d201d8c38040
[103979[D][EM]2113dfdffcdbc303aa27d201
[103983[D][EM]Removed CHIP MessageCounter:3700478 from RetransTable on exchange 2517i
[104604[D][IN]UDP Message Received packet nb : 26 SrcAddr : fd11:33:0:0:a9e7:1f56:d76e:985[5540] DestAddr : fd11:33:0:0:3[...][%d] Payload Length %d
[104619[P][EM]Received message of type 0x11 with protocolId (0, 2) and MessageCounter:8939333 on exchange 2517i
[104624[D][EM]c530741eaa27d201d8c38040
[104628[D][EM]CHIP MessageCounter:3700478 not in RetransTable on exchange 2517i
[104632[U][ATM]Handling received BDX Message
[104635[U][ATM]Block
[104637[U][ATM]  Block Counter: 1
[104640[U][ATM]  Data Length: 1024
[104643[D][DMG]Endpoint 0, Cluster 0x0000_002A update version to a0f3e64f
[104647[P][ZCL]Unknown attribute ID: 0x0000_0003
[104828[D][EM]Sending Standalone Ack for MessageCounter:8939333 on exchange 2517i
[104837[P][IN]Prepared secure message 0x04029908 to 0x0000000000000001 (1)  of type 0x10 and protocolId (0, 0) on exchange 2517i with MessageCounter:3700479.
[104853[P][IN]Sending encrypted msg 0x04029908 with MessageCounter:3700479 to 0x0000000000000001 (1) at monotonic time: 0000000000019334 msec
[104927[D][IN]UDP Message Received packet nb : 27 SrcAddr : fd11:33:0:0:a9e7:1f56:d76e:985[5540] DestAddr : fd11:33:0:0:3[...][%d] Payload Length %d
[104942[D][IN]Received a duplicate message with MessageCounter:8939333 on exchange 2517i
[104946[P][EM]Received message of type 0x11 with protocolId (0, 2) and MessageCounter:8939333 on exchange 2517i
[104951[D][EM]c530741eaa27d201d8c38040
[104955[D][EM]Forcing tx of solitary ack for duplicate MessageCounter:8939333 on exchange 2517i
[104959[D][EM]Sending Standalone Ack for MessageCounter:8939333 on exchange 2517i
[104964[P][IN]Prepared secure message 0x040296F8 to 0x0000000000000001 (1)  of type 0x10 and protocolId (0, 0) on exchange 2517i with MessageCounter:3700480.
[104972[P][IN]Sending encrypted msg 0x040296F8 with MessageCounter:3700480 to 0x0000000000000001 (1) at monotonic time: 00000000000193AC msec

```

bzbarsky-apple, 2022-06-09:
> and the assert fails after receiving the first BDX message.

Any chance of getting a stack to the assert failure?

But it sure looks like the BDX SendMessage is happening on the wrong thread...

marius-alex-tache, 2022-06-10:
Yes, sure. I will test this again today and will edit this comment.

**TODO**

bzbarsky-apple, 2022-06-11:
That's very useful, thank you!

The line numbers don't exactly match up with my `examples/lighting-app/nxp/k32w/k32w0/main/AppTask.cpp` on revision 287e47a0ee636bf47aa84f4043d0f6ef141eef71, in that what I see is:
```
        while (eventReceived == pdTRUE)
        {  <-- Line 244 here.
            sAppTask.DispatchEvent(&event);
            eventReceived = xQueueReceive(sAppEventQueue, &event, 0);
        }
```
What does line 244 of `AppTask.cpp` look like in your case?

But I do see calls to `FetchNextData` in `src/platform/nxp/k32w/k32w0/OTAImageProcessorImpl.cpp`.  It calls `FetchNextData` from `OTAImageProcessorImpl::TriggerNewRequestForData`, which is called from `examples/lighting-app/nxp/k32w/k32w0/main/AppTask.cpp` line 720, in `AppTask::OTAResumeEventHandler`.

Looks like that gets run off an event dispatched to sAppTAsk, which synchronously runs the event.  And that's happening from that loop inside `AppTaskMain`, so definitely not on the Matte thread.

So something here needs to either take the Matter lock or queue things over to the Matter thread to actually run the bdx bits....  Either in OTAImageProcessorImpl or in the caller in AppTask.

And we should add `assertChipStackLockedByCurrentThread` to our message send/receive paths, it sounds like.  I'll try a PR for that and see how much blows up.

bzbarsky-apple, 2022-06-11:
Creatd https://github.com/project-chip/connectedhomeip/pull/19493 for adding those send/receive asserts.

marius-alex-tache, 2022-06-13:
I've just wrapped the `AppTask::TriggerNewRequestForData` call within lock/unlock chip stack calls:
```
PlatformMgr().LockChipStack();
gImageProcessor.TriggerNewRequestForData();
PlatformMgr().UnlockChipStack();
``` 
I confirm that the `assertChipStackLockedByCurrentThread` added inside `ReliableMessageMgr::StartTimer` passes and I no longer have issues with reading attributes during OTA block transfers. Thank you for your help!

bzbarsky-apple, 2022-06-13:
@marius-alex-tache Are you going to create a PR for that change?

marius-alex-tache, 2022-06-14:
@bzbarsky-apple Hmm, we are currently working on an internal fork. We will integrate the modification later. I think we can close this issue. Thank you!

closed 2022-06-14 by marius-alex-tache