project-chip/connectedhomeip #18332: Operational Discovery fails for the OTA Provider

doru91, 2022-05-11:
#### Problem
* operational discovery of the OTA Provider fails on the OTA Requestor side if RPI Device has active both Ethernet and Wi-Fi interfaces 
* tested using NXP K32W0 with TE9 topology (RPI4 OS image, chip-tool compiled with platform mDNS)

- RPI has both Wi-Fi and Ethernet interfaces active (see the captures NOT_OK_Announce/NOT_OK_Capture_Announce - first capture is for air traffic, the second one is for RPi traffic, 192.168.0.103 - Wi-Fi, 192.168.1.102 - Ethernet)
[image]

1. Packet 732 contains the matter._tcp query initiated by the OTA Provider
2. Packet 741 contains the reply but it lacks the IPv6 address of the OTA Provider

Also, the RPi traffic doesn't look good:
[image]

- RPi has only the Wi-Fi interface active (see the captures OK_Announce/OK_Capture_Announce - first capture is for air traffic, the second one is for RPi traffic)
[image]
1. Packet 281contains the matter._tcp query initiated by the OTA Provider
2. Packet 287 contains the reply and it also contains the IPv6 address of the OTA Provider
[MDNS_captures.zip](https://github.com/project-chip/connectedhomeip/files/8671113/MDNS_captures.zip)

andy31415, 2022-05-11:
I only see SRV queries over the wire. I expect minmdns to reply with just the SRV entry in this case and it would not attach A/AAAA or TXT entries to it. 

What client side library is used here? I would expect a query for SRV to also be followed by AAAA and TXT if using some 3rd party mdns library. If using minmdns, we should always query for 'ANY'

andy31415, 2022-05-11:
Is the issue that nobody replies to SRV queries even though they seem valid?

tcarmelveilleux, 2022-05-11:
> I only see SRV queries over the wire. I expect minmdns to reply with just the SRV entry in this case and it would not attach A/AAAA or TXT entries to it.

No, according to DNS-SD RFC, AAAA at least are needed:
```
https://datatracker.ietf.org/doc/html/rfc6763#section-12.2
```

Also note that https://datatracker.ietf.org/doc/html/rfc6763#section-12 says:

>    This section recommends which additional records SHOULD be generated
>    to improve network efficiency, for both Unicast and Multicast DNS-SD
>    responses.
> 
>    Note that while **servers SHOULD add these additional records** for
>    efficiency purposes, as with all DNS additional records, it is the
>    client's responsibility to determine whether or not to trust them.
> 
>    Generally speaking, stub resolvers that talk to a single recursive
>    name server for all their queries will trust all records they receive
>    from that recursive name server (whom else would they ask?).
>    Recursive name servers that talk to multiple authoritative name
>    servers should verify that any records they receive from a given
>    authoritative name server are "in bailiwick" for that server, and
>    ignore them if not.
> 
>    **Clients MUST be capable of functioning correctly with DNS servers
>    (and Multicast DNS Responders) that fail to generate these additional
>    records automatically, by issuing subsequent queries for any further
>    record(s) they require.**  The additional-record generation rules in
>    this section are RECOMMENDED for improving network efficiency, but
>    are not required for correctness.

woody-apple, 2022-07-20:
Spec Review: @doru91 any update on this? Is this still happening? We may need help digging in if so.

woody-apple, 2022-07-20:
Spec Review: Per @andy31415 there have been changes in min-mDNS can you please re-test this?

woody-apple, 2022-08-31:
Closing this given this has had enough time to re-test. @doru91 can you please re-open if this is still happening?

closed 2022-08-31 by woody-apple