project-chip/connectedhomeip #18339: Add data version for ReadAttribute in CHIP.framework if OnDone is added.

krypton36, 2022-05-11:
#### Problem
https://github.com/project-chip/connectedhomeip/pull/18174 PR add data version to write, but does not add data version to read, because when the data version matches the server data version no data is reported back. Since this API relies on a report callback or error callback, none of these callbacks are executed. In order to add to CHIP.framework and chip-tool-darwin, the ReadAttribute cluster API needs to expose the OnDone callback or data version needs to be removed from the API and any test plan that requires the controller to set the data version on read.

 #### Proposed Solution
If the callback OnDone is added this issue is to be used to add to CHIP.framework and chip-tool-darwin. If the data version is removed, this issue is to be closed.

woody-apple, 2022-05-24:
Not clear this is required for V1.0, moving out of V1. @krypton36 please update if this isn't correct.

krypton36, 2022-05-24:
Not needed for V1.0

closed 2022-12-03 by stale[bot]