project-chip/connectedhomeip #18355: [TE9][TC-DM-4.17] Last 3 steps doesn't seem to meet the expected results

nansenLV, 2022-05-12:
[TC-DM-4.17]step6  step7  step8 doesn't seem to meet the expected results.
expected behavior:
step6
[1645773078.930516][3905:3910] CHIP:DMG:         SuppressResponse = true, 
[1645773078.930579][3905:3910] CHIP:DMG:         InteractionModelRevision = 1
[1645773078.930635][3905:3910] CHIP:DMG: }
[1645773078.930943][3905:3910] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Attribute 0x0000_0005DataVersion: 92121563
[1645773078.931067][3905:3910] CHIP:TOO:   LastNetworkingStatus: 0
[1645773078.931217][3905:3910] CHIP:EM: Sending Standalone Ack for MessageCounter:16576011 on exchange 44187i
 
step7
[1645773167.178501][3913:3918] CHIP:DMG:         SuppressResponse = true, 
[1645773167.178561][3913:3918] CHIP:DMG:         InteractionModelRevision = 1
[1645773167.178618][3913:3918] CHIP:DMG: }
[1645773167.178919][3913:3918] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Attribute 0x0000_0006DataVersion: 92121563
[1645773167.182241][3913:3918] CHIP:TOO:   LastNetworkID: 1111161622222211
[1645773167.182422][3913:3918] CHIP:EM: Sending Standalone Ack for MessageCounter:2029571 on exchange 18566i

step8
[1645773318.752774][3938:3943] CHIP:DMG:         SuppressResponse = true, 
[1645773318.752861][3938:3943] CHIP:DMG:         InteractionModelRevision = 1
[1645773318.753043][3938:3943] CHIP:DMG: }
[1645773318.753543][3938:3943] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Attribute 0x0000_0007DataVersion: 92121563
[1645773318.753724][3938:3943] CHIP:TOO:   LastConnectErrorValue: 0
[1645773318.753985][3938:3943] CHIP:EM: Sending Standalone Ack for MessageCounter:15303417 on exchange 41937i

actual behavior

step6:
./chip-tool networkcommissioning read last-networking-status 6789 0
[1652243168.708113][105596:105602] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Attribute 0x0000_0005 DataVersion: 2859301505
[1652243168.708141][105596:105602] CHIP:TOO:   LastNetworkingStatus: null
[1652243168.708181][105596:105602] CHIP:EM: Sending Standalone Ack for MessageCounter:2912464 on exchange 59340i

step7；
./chip-tool networkcommissioning read last-network-id 6789 0
[1652243222.580788][105615:105621] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Attribute 0x0000_0006 DataVersion: 2859301505
[1652243222.581498][105615:105621] CHIP:TOO:   LastNetworkID: null
[1652243222.581644][105615:105621] CHIP:EM: Sending Standalone Ack for MessageCounter:9439845 on exchange 24941i

step8：
./chip-tool networkcommissioning read last-connect-error-value 6789 0
[1652243468.817194][105683:105689] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Attribute 0x0000_0007 DataVersion: 2859301505
[1652243468.817381][105683:105689] CHIP:TOO:   LastConnectErrorValue: null
[1652243468.817466][105683:105689] CHIP:EM: Sending Standalone Ack for MessageCounter:2700104 on exchange 24577i
[Uploading tc-dm-4.17.md…]()

erjiaqing, 2022-05-12:
Are you trying to read these attributes in a clean setup or in a commissioned state?

These attributes are nullable attributes, and are expected to be null when they are not appliable.

Spec 11.7.8.6:

> If no such attempt was made, or no network configurations exist in the Networks attribute, then this attribute SHALL be set to null.

Similar words in 11.7.8.7 and 11.7.8.8

bzbarsky-apple, 2022-05-21:
Also, TC-DM-4.17 is a test where the DUT is the _commissioner_, which is not the situation above....

closed 2022-05-21 by bzbarsky-apple