project-chip/connectedhomeip #18362: [TC-GC-2.4] While verifying SetRegulatoryConfig command TH responds as debugText as a integer instead 'string' value.

Rajashreekalmane, 2022-05-12:
**#### Problem**
As per Test-Plan, test-step 4 expect that " H1 sends SetRegulatoryConfig command to the DUT with NewRegulatoryConfig data value as greater than '3', Breadcrumb as 0 and CountryCode is of type 'string'" when DUT sends SetRegulatoryConfigResponse command to TH1 with field ErrorCode as 'ValueOutsideRange'(1) and **DebugText as a 'string' value**. But we are getting response as ValueOutsideRange(1) with **debugText  integer value as 12.**

 *** expected behavior**:- DebugText should be 'string' value
 *** actual behavior**:- debugText  integer 12 value.
 *** steps to reproduce**:-./chip-tool generalcommissioning set-regulatory-config 4 12 0 1 0

**steps to reproduce:-**./chip-tool generaldiagnostics read feature-map 1 0
**system configuration:-**
Chip-Tool: RPI
DUT: RPI
Platform : IP and Wifi
Commit-ID: https://github.com/project-chip/connectedhomeip/commit/11587f5bf0bb9e621edd4c867c0bc942d804d1ce
Dut:all-clusters-app

**PFA screenshot:-**
[image]

**Test-plan Reference:-**
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/General_Commissioning.adoc#tc-gc-2-4-verification-for-commissioningerror-on-response-message-dut-commissionee

**Spec screenshot:-**
[image]

Rajashreekalmane, 2022-05-12:
[TC-GC-2.4]--While verifying  CommissioningComplete command  DUT responds as debugText as empty instead 'string' value.

PFA screenshot:-
[image]

andy31415, 2022-05-13:
@Rajashreekalmane could you explain the test case a bit? I look at the send and redceive data, you send 12 and receive 12. Actual formatting of the logs may not contain quotes.

Could you send debug text that is clearly text like `xy` and see if that goes through?

closed 2022-05-17 by Rajashreekalmane