project-chip/connectedhomeip #18363: [SQA]-THREAD Lock-Door Command throws the Failure Response for the PINCODE Argument.

Survensa, 2022-05-12:
**Problem:** 
While Executing TC-DL-2.2, The lock-door command Throws FAILURE response for the  PinCode Argument.

**Expected behavior** - Command should get SUCCESS response .

**Actual behavior** - Command  gets Failure response .

**Steps to reproduce -**` ./chip-tool doorlock lock-door 71 1 --timedInteractionTimeoutMs 1000 --PinCode 7896
`

**Commit used :**   11587f5bf0bb9e621edd4c867c0bc942d804d1ce

**Platform :** THREAD- LOCK-APP

**Reference:**
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/door_lock_Cluster.adoc

**Logs:** 

> [1652341619.526828][29934:29939] CHIP:EM: Removed CHIP MessageCounter:5057605 from RetransTable on exchange 7996i
[1652341619.526885][29934:29939] CHIP:DMG: ICR moving to [ResponseRe]
[1652341619.526964][29934:29939] CHIP:DMG: InvokeResponseMessage =
[1652341619.527009][29934:29939] CHIP:DMG: {
[1652341619.527053][29934:29939] CHIP:DMG: 	suppressResponse = false, 
[1652341619.527100][29934:29939] CHIP:DMG: 	InvokeResponseIBs =
[1652341619.527157][29934:29939] CHIP:DMG: 	[
[1652341619.527201][29934:29939] CHIP:DMG: 		InvokeResponseIB =
[1652341619.527272][29934:29939] CHIP:DMG: 		{
[1652341619.527320][29934:29939] CHIP:DMG: 			CommandStatusIB =
[1652341619.527379][29934:29939] CHIP:DMG: 			{
[1652341619.527433][29934:29939] CHIP:DMG: 				CommandPathIB =
[1652341619.527500][29934:29939] CHIP:DMG: 				{
[1652341619.527562][29934:29939] CHIP:DMG: 					EndpointId = 0x1,
[1652341619.527632][29934:29939] CHIP:DMG: 					ClusterId = 0x101,
[1652341619.527737][29934:29939] CHIP:DMG: 					CommandId = 0x0,
[1652341619.527794][29934:29939] CHIP:DMG: 				},
[1652341619.527870][29934:29939] CHIP:DMG: 				
[1652341619.527939][29934:29939] CHIP:DMG: 				StatusIB =
[1652341619.528032][29934:29939] CHIP:DMG: 				{
[1652341619.528117][29934:29939] CHIP:DMG: 					status = 0x01 (FAILURE),
[1652341619.528201][29934:29939] CHIP:DMG: 				},
[1652341619.528293][29934:29939] CHIP:DMG: 				
[1652341619.528367][29934:29939] CHIP:DMG: 			},
[1652341619.528447][29934:29939] CHIP:DMG: 			
[1652341619.528514][29934:29939] CHIP:DMG: 		},
[1652341619.528592][29934:29939] CHIP:DMG: 		
[1652341619.528650][29934:29939] CHIP:DMG: 	],
[1652341619.528722][29934:29939] CHIP:DMG: 	
[1652341619.528781][29934:29939] CHIP:DMG: 	InteractionModelRevision = 1
[1652341619.528839][29934:29939] CHIP:DMG: },
[1652341619.528974][29934:29939] CHIP:DMG: Received Command Response Status for Endpoint=1 Cluster=0x0000_0101 Command=0x0000_0000 Status=0x1
[1652341619.529051][29934:29939] CHIP:TOO: Error: IM Error 0x00000501: General error: 0x01 (FAILURE)
[1652341619.529135][29934:29939] CHIP:DMG: ICR moving to [AwaitingDe]

Survensa, 2022-05-16:
DUT LOG :

ubuntu@ubuntu:~/apps$ ./chip-tool doorlock lock-door 1 1 --timedInteractionTimeoutMs 1000 --PinCode 7896

```
[1652703038.246901][3318:3318] CHIP:DIS: Broadcasting mDns reply for query from fd01::e65f:1ff:fe0f:3a64
[1652703038.249128][3318:3318] CHIP:DIS: Broadcasting mDns reply for query from fd01::e65f:1ff:fe0f:3a64
[1652703038.342709][3318:3318] CHIP:EM: Received message of type 0x30 with protocolId (0, 0) and MessageCounter:445819419 on exchange 47861r
[1652703038.342829][3318:3318] CHIP:EM: Handling via exchange: 47861r, Delegate: 0xaaaadcf57860
[1652703038.342921][3318:3318] CHIP:IN: CASE Server received Sigma1 message. Starting handshake. EC 0xaaaadcf58190
[1652703038.343014][3318:3318] CHIP:DIS: CASE handshake initiated, closing all BLE Connections
[1652703038.343176][3318:3318] CHIP:SC: Waiting for Sigma1 msg
[1652703038.343233][3318:3318] CHIP:IN: CASE Server disabling CASE session setups
[1652703038.343292][3318:3318] CHIP:SC: Received Sigma1 msg
[1652703038.343392][3318:3318] CHIP:SC: Found MRP parameters in the message
[1652703038.343458][3318:3318] CHIP:SC: Peer assigned session key ID 5463
[1652703038.344290][3318:3318] CHIP:SC: CASE matched destination ID: fabricIndex 1, NodeID 0x0000000000000001
[1652703038.346700][3318:3318] CHIP:SC: Including MRP parameters
[1652703038.346829][3318:3318] CHIP:EM: Piggybacking Ack for MessageCounter:445819419 on exchange: 47861r
[1652703038.346919][3318:3318] CHIP:IN: Prepared unauthenticated message 0xaaaadcf57e58 to 0x0000000000000000 (0)  of type 0x31 and protocolId (0, 0) on exchange 47861r with MessageCounter:2411602807.
[1652703038.347199][3318:3318] CHIP:IN: Sending unauthenticated msg 0xaaaadcf57e58 with MessageCounter:2411602807 to 0x0000000000000000 at monotonic time: 0000000000B882B0 msec
[1652703038.347539][3318:3318] CHIP:SC: Sent Sigma2 msg
[1652703038.354784][3318:3318] CHIP:EM: Received message of type 0x32 with protocolId (0, 0) and MessageCounter:445819420 on exchange 47861r
[1652703038.354838][3318:3318] CHIP:EM: Found matching exchange: 47861r, Delegate: 0xaaaadcf57880
[1652703038.354896][3318:3318] CHIP:EM: Rxd Ack; Removing MessageCounter:2411602807 from Retrans Table on exchange 47861r
[1652703038.354931][3318:3318] CHIP:EM: Removed CHIP MessageCounter:2411602807 from RetransTable on exchange 47861r
[1652703038.355006][3318:3318] CHIP:SC: Received Sigma3 msg
[1652703038.355178][3318:3318] CHIP:SC: The device does not support GetClock_RealTimeMS() API. This will eventually result in CASE session setup failures.
[1652703038.359149][3318:3318] CHIP:SC: Unable to save session resumption state: ../../examples/door-lock-app/linux/third_party/connectedhomeip/src/platform/Linux/KeyValueStoreManagerImpl.cpp:53: CHIP Error 0x000000A0: Value not found in the persisted storage
[1652703038.359204][3318:3318] CHIP:SC: Sending status report. Protocol code 0, exchange 47861
[1652703038.359255][3318:3318] CHIP:EM: Piggybacking Ack for MessageCounter:445819420 on exchange: 47861r
[1652703038.359311][3318:3318] CHIP:IN: Prepared unauthenticated message 0xaaaadcf57e58 to 0x0000000000000000 (0)  of type 0x40 and protocolId (0, 0) on exchange 47861r with MessageCounter:2411602808.
[1652703038.359365][3318:3318] CHIP:IN: Sending unauthenticated msg 0xaaaadcf57e58 with MessageCounter:2411602808 to 0x0000000000000000 at monotonic time: 0000000000B882BC msec
[1652703038.359585][3318:3318] CHIP:IN: CASE Session established. Setting up the secure channel.
[1652703038.359640][3318:3318] CHIP:IN: New secure session created for device 0x000000000001B669, LSID:24572 PSID:5463!
[1652703038.359737][3318:3318] CHIP:IN: CASE secure channel is available now.
[1652703038.359771][3318:3318] CHIP:IN: CASE Server enabling CASE session setups
[1652703038.360666][3318:3318] CHIP:EM: Received message of type 0xa with protocolId (0, 1) and MessageCounter:40791 on exchange 47862r
[1652703038.360723][3318:3318] CHIP:EM: Handling via exchange: 47862r, Delegate: 0xaaaadcf51d38
[1652703038.360800][3318:3318] CHIP:DMG: TimedRequestMessage =
[1652703038.360841][3318:3318] CHIP:DMG: {
[1652703038.360876][3318:3318] CHIP:DMG:        TimeoutMs = 0x3e8,
[1652703038.360914][3318:3318] CHIP:DMG:        InteractionModelRevision = 1
[1652703038.360949][3318:3318] CHIP:DMG: }
[1652703038.360988][3318:3318] CHIP:DMG: Got Timed Request with timeout 1000: handler 0xaaab1789baf0 exchange 47862r
[1652703038.361047][3318:3318] CHIP:EM: Piggybacking Ack for MessageCounter:40791 on exchange: 47862r
[1652703038.361121][3318:3318] CHIP:IN: Prepared secure message 0xaaaadcf57e78 to 0x000000000001B669 (1)  of type 0x1 and protocolId (0, 1) on exchange 47862r with MessageCounter:4008880.
[1652703038.361170][3318:3318] CHIP:IN: Sending encrypted msg 0xaaaadcf57e78 with MessageCounter:4008880 to 0x000000000001B669 (1) at monotonic time: 0000000000B882BE msec
[1652703038.361338][3318:3318] CHIP:DMG: Timed Request time limit 0x0000000000B886A6: handler 0xaaab1789baf0 exchange 47862r
[1652703038.361509][3318:3318] CHIP:EM: Received message of type 0x10 with protocolId (0, 0) and MessageCounter:445819421 on exchange 47861r
[1652703038.361551][3318:3318] CHIP:EM: Found matching exchange: 47861r, Delegate: (nil)
[1652703038.361602][3318:3318] CHIP:EM: Rxd Ack; Removing MessageCounter:2411602808 from Retrans Table on exchange 47861r
[1652703038.361638][3318:3318] CHIP:EM: Removed CHIP MessageCounter:2411602808 from RetransTable on exchange 47861r
[... cut ...]
[1652703038.365432][3318:3318] CHIP:DMG: All ReadHandler-s are clean, clear GlobalDirtySet
[1652703038.367472][3318:3318] CHIP:EM: Received message of type 0x10 with protocolId (0, 0) and MessageCounter:40793 on exchange 47862r
[1652703038.367517][3318:3318] CHIP:EM: Found matching exchange: 47862r, Delegate: (nil)
[1652703038.367566][3318:3318] CHIP:EM: Rxd Ack; Removing MessageCounter:4008881 from Retrans Table on exchange 47862r
[1652703038.367601][3318:3318] CHIP:EM: Removed CHIP MessageCounter:4008881 from RetransTable on exchange 47862r

```

bzbarsky-apple, 2022-05-19:
In `examples/lock-app/efr32/src/LockManager.cpp` we have:
```
bool LockManager::setLockState(chip::EndpointId endpointId, DlLockState lockState, const Optional<chip::ByteSpan> & pin,
                               DlOperationError & err)
{
...
    if (curState == lockState)
    {
        ChipLogDetail(Zcl, "Door Lock App: door is already locked, ignoring command to set lock state to \"%s\" [endpointId=%d]",
                      lockStateToString(lockState), mEndpointId);
        return false;
    }
```
That means `emberAfPluginDoorLockOnDoorLockCommand` returns false, so `DoorLockServer::HandleRemoteLockOperation` does:
```
    bool success = credentialsOk && operationOk;

    // Send command response
    emberAfSendImmediateDefaultResponse(success ? EMBER_ZCL_STATUS_SUCCESS : EMBER_ZCL_STATUS_FAILURE);
```
with `operationOk` set to false.

@mykrupp could you please take a look?

bzbarsky-apple, 2022-05-19:
That is, either locking while already locked should succeed, and then the example app code is wrong.  Or it should fail and then the test plan is wrong?

mykrupp, 2022-05-19:
Yes, I will take a look. Can I get confirmation on how this was built, and that it is being run on the efr32 platform?

bzbarsky-apple, 2022-05-20:
@Survensa can you confirm?

@mykrupp I figured this was EFR32 because this line from the above log:
```
[1652703038.364566][3318:3318] CHIP:ZCL: Door Lock App: door is already locked, ignoring command to set lock state to "Locked" [endpointId=1]
```
has a log message that only appears in `examples/lock-app/efr32/src/LockManager.cpp`

mykrupp, 2022-05-23:
@Survensa Since this is run on the nordic platform, there's not much I can do to solve this issue. 

@bzbarsky-apple I still agree with what you found on the efr32 platform. I will go through the lock app and carefully follow the spec for each non-happy path.

kkasperczyk-no, 2022-05-25:
@Survensa as you wrote it fails on nRF, because you invoked the `lock-door` command with pinCode argument that is optional. According to spec the pin code argument has [COTA & PIN] conformance that is not met by the nRFConnect platform. Isn't getting failure response something we should expect in such case? The same comment for the https://github.com/project-chip/connectedhomeip/issues/18364 issue that seems to describe the same problem.

closed 2022-05-26 by cjandhyala