project-chip/connectedhomeip #18364: [SQA]-THREAD  Unlock Door Command throws the FAILURE Response.

Survensa, 2022-05-12:
**Problem:**
While Executing TC-DL-2.3, The Unlock-door command Throws FAILURE response 

**Expected behavior** - Command should get SUCCESS response .

**Actual behavior** - Command gets Failure response .

**Steps to reproduce** -  ` ./chip-tool doorlock unlock-door 71 1 --timedInteractionTimeoutMs 1000 --PinCode xxxxxxxx`

**Commit** used : https://github.com/project-chip/connectedhomeip/commit/11587f5bf0bb9e621edd4c867c0bc942d804d1ce

**Platform** : THREAD- LOCK-APP

**Reference:**
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/door_lock_Cluster.adoc

**Logs:**

> [1652346790.956337][30148:30153] CHIP:DMG: ICR moving to [ResponseRe]
[1652346790.956416][30148:30153] CHIP:DMG: InvokeResponseMessage =
[1652346790.956461][30148:30153] CHIP:DMG: {
[1652346790.956528][30148:30153] CHIP:DMG: 	suppressResponse = false, 
[1652346790.956574][30148:30153] CHIP:DMG: 	InvokeResponseIBs =
[1652346790.956645][30148:30153] CHIP:DMG: 	[
[1652346790.956691][30148:30153] CHIP:DMG: 		InvokeResponseIB =
[1652346790.956770][30148:30153] CHIP:DMG: 		{
[1652346790.956818][30148:30153] CHIP:DMG: 			CommandStatusIB =
[1652346790.956893][30148:30153] CHIP:DMG: 			{
[1652346790.956948][30148:30153] CHIP:DMG: 				CommandPathIB =
[1652346790.957031][30148:30153] CHIP:DMG: 				{
[1652346790.957110][30148:30153] CHIP:DMG: 					EndpointId = 0x1,
[1652346790.957176][30148:30153] CHIP:DMG: 					ClusterId = 0x101,
[1652346790.957258][30148:30153] CHIP:DMG: 					CommandId = 0x1,
[1652346790.957319][30148:30153] CHIP:DMG: 				},
[1652346790.957442][30148:30153] CHIP:DMG: 				
[1652346790.957539][30148:30153] CHIP:DMG: 				StatusIB =
[1652346790.957627][30148:30153] CHIP:DMG: 				{
[1652346790.957737][30148:30153] CHIP:DMG: 					status = 0x01 (FAILURE),
[1652346790.957879][30148:30153] CHIP:DMG: 				},
[1652346790.957985][30148:30153] CHIP:DMG: 				
[1652346790.958067][30148:30153] CHIP:DMG: 			},
[1652346790.958169][30148:30153] CHIP:DMG: 			
[1652346790.958236][30148:30153] CHIP:DMG: 		},
[1652346790.958336][30148:30153] CHIP:DMG: 		
[1652346790.958394][30148:30153] CHIP:DMG: 	],
[1652346790.958489][30148:30153] CHIP:DMG: 	
[1652346790.958547][30148:30153] CHIP:DMG: 	InteractionModelRevision = 1
[1652346790.958604][30148:30153] CHIP:DMG: },
[1652346790.958742][30148:30153] CHIP:DMG: Received Command Response Status for Endpoint=1 Cluster=0x0000_0101 Command=0x0000_0001 Status=0x1
[1652346790.958818][30148:30153] CHIP:TOO: Error: IM Error 0x00000501: General error: 0x01 (FAILURE)
[1652346790.958902][30148:30153] CHIP:DMG: ICR moving to [AwaitingDe]
[1652346790.959012][30148:30153] CHIP:EM: Sending Standalone Ack for MessageCounter:14966235 on exchange 27401i

Survensa, 2022-05-16:
DUT Log:

```
[1652703120.702132][3318:3318] CHIP:EM: Received message of type 0x30 with protocolId (0, 0) and MessageCounter:2160415259 on exchange 22911r
[1652703120.702262][3318:3318] CHIP:EM: Handling via exchange: 22911r, Delegate: 0xaaaadcf57860
[1652703120.702358][3318:3318] CHIP:IN: CASE Server received Sigma1 message. Starting handshake. EC 0xaaaadcf58190
[1652703120.702416][3318:3318] CHIP:DIS: CASE handshake initiated, closing all BLE Connections
[1652703120.702579][3318:3318] CHIP:SC: Waiting for Sigma1 msg
[1652703120.702637][3318:3318] CHIP:IN: CASE Server disabling CASE session setups
[1652703120.702733][3318:3318] CHIP:SC: Received Sigma1 msg
[1652703120.702858][3318:3318] CHIP:SC: Found MRP parameters in the message
[1652703120.702927][3318:3318] CHIP:SC: Peer assigned session key ID 43722
[1652703120.703902][3318:3318] CHIP:SC: CASE matched destination ID: fabricIndex 1, NodeID 0x0000000000000001
[1652703120.706348][3318:3318] CHIP:SC: Including MRP parameters
[1652703120.706475][3318:3318] CHIP:EM: Piggybacking Ack for MessageCounter:2160415259 on exchange: 22911r
[1652703120.706568][3318:3318] CHIP:IN: Prepared unauthenticated message 0xaaaadcf57e58 to 0x0000000000000000 (0)  of type 0x31 and protocolId (0, 0) on exchange 22911r with MessageCounter:2411602809.
[1652703120.706656][3318:3318] CHIP:IN: Sending unauthenticated msg 0xaaaadcf57e58 with MessageCounter:2411602809 to 0x0000000000000000 at monotonic time: 0000000000B9C467 msec
[1652703120.707161][3318:3318] CHIP:SC: Sent Sigma2 msg
[1652703120.714637][3318:3318] CHIP:EM: Received message of type 0x32 with protocolId (0, 0) and MessageCounter:2160415260 on exchange 22911r
[1652703120.714695][3318:3318] CHIP:EM: Found matching exchange: 22911r, Delegate: 0xaaaadcf57880
[1652703120.714758][3318:3318] CHIP:EM: Rxd Ack; Removing MessageCounter:2411602809 from Retrans Table on exchange 22911r
[1652703120.714792][3318:3318] CHIP:EM: Removed CHIP MessageCounter:2411602809 from RetransTable on exchange 22911r
[1652703120.714839][3318:3318] CHIP:SC: Received Sigma3 msg
[1652703120.715068][3318:3318] CHIP:SC: The device does not support GetClock_RealTimeMS() API. This will eventually result in CASE session setup failures.
[1652703120.718749][3318:3318] CHIP:SC: Unable to save session resumption state: ../../examples/door-lock-app/linux/third_party/connectedhomeip/src/platform/Linux/KeyValueStoreManagerImpl.cpp:53: CHIP Error 0x000000A0: Value not found in the persisted storage
[1652703120.718818][3318:3318] CHIP:SC: Sending status report. Protocol code 0, exchange 22911
[1652703120.718866][3318:3318] CHIP:EM: Piggybacking Ack for MessageCounter:2160415260 on exchange: 22911r
[1652703120.718923][3318:3318] CHIP:IN: Prepared unauthenticated message 0xaaaadcf57e58 to 0x0000000000000000 (0)  of type 0x40 and protocolId (0, 0) on exchange 22911r with MessageCounter:2411602810.
[1652703120.719006][3318:3318] CHIP:IN: Sending unauthenticated msg 0xaaaadcf57e58 with MessageCounter:2411602810 to 0x0000000000000000 at monotonic time: 0000000000B9C474 msec
[1652703120.719293][3318:3318] CHIP:IN: CASE Session established. Setting up the secure channel.
[1652703120.719350][3318:3318] CHIP:IN: New secure session created for device 0x000000000001B669, LSID:24573 PSID:43722!
[1652703120.719452][3318:3318] CHIP:IN: CASE secure channel is available now.
[1652703120.719499][3318:3318] CHIP:IN: CASE Server enabling CASE session setups
[1652703120.720486][3318:3318] CHIP:EM: Received message of type 0xa with protocolId (0, 1) and MessageCounter:6744032 on exchange 22912r
[1652703120.720546][3318:3318] CHIP:EM: Handling via exchange: 22912r, Delegate: 0xaaaadcf51d38
[1652703120.720632][3318:3318] CHIP:DMG: TimedRequestMessage =
[1652703120.720670][3318:3318] CHIP:DMG: {
[1652703120.720717][3318:3318] CHIP:DMG:        TimeoutMs = 0x3e8,
[1652703120.720761][3318:3318] CHIP:DMG:        InteractionModelRevision = 1
[1652703120.720798][3318:3318] CHIP:DMG: }
[1652703120.720849][3318:3318] CHIP:DMG: Got Timed Request with timeout 1000: handler 0xaaab17875bd0 exchange 22912r
[1652703120.720909][3318:3318] CHIP:EM: Piggybacking Ack for MessageCounter:6744032 on exchange: 22912r
[1652703120.721002][3318:3318] CHIP:IN: Prepared secure message 0xaaaadcf57e78 to 0x000000000001B669 (1)  of type 0x1 and protocolId (0, 1) on exchange 22912r with MessageCounter:3790647.
[1652703120.721066][3318:3318] CHIP:IN: Sending encrypted msg 0xaaaadcf57e78 with MessageCounter:3790647 to 0x000000000001B669 (1) at monotonic time: 0000000000B9C476 msec
[1652703120.721295][3318:3318] CHIP:DMG: Timed Request time limit 0x0000000000B9C85E: handler 0xaaab17875bd0 exchange 22912r
[1652703120.721504][3318:3318] CHIP:EM: Received message of type 0x10 with protocolId (0, 0) and MessageCounter:2160415261 on exchange 22911r
[1652703120.721546][3318:3318] CHIP:EM: Found matching exchange: 22911r, Delegate: (nil)
[1652703120.721598][3318:3318] CHIP:EM: Rxd Ack; Removing MessageCounter:2411602810 from Retrans Table on exchange 22911r
[1652703120.721631][3318:3318] CHIP:EM: Removed CHIP MessageCounter:2411602810 from RetransTable on exchange 22911r
[1652703120.722234][3318:3318] CHIP:EM: Received message of type 0x8 with protocolId (0, 1) and MessageCounter:6744033 on exchange 22912r
[1652703120.722373][3318:3318] CHIP:EM: Found matching exchange: 22912r, Delegate: 0xaaab17875bd0
[... cut ...]
[1652703120.725187][3318:3318] CHIP:DMG: ICR moving to [AwaitingDe]
[1652703120.727326][3318:3318] CHIP:EM: Received message of type 0x10 with protocolId (0, 0) and MessageCounter:6744034 on exchange 22912r
[1652703120.727369][3318:3318] CHIP:EM: Found matching exchange: 22912r, Delegate: (nil)
[1652703120.727414][3318:3318] CHIP:EM: Rxd Ack; Removing MessageCounter:3790648 from Retrans Table on exchange 22912r
[1652703120.727437][3318:3318] CHIP:EM: Removed CHIP MessageCounter:3790648 from RetransTable on exchange 22912r
```

bzbarsky-apple, 2022-05-19:
This seems like the key part in the server log:
> [1652703120.724463][3318:3318] CHIP:ZCL: Door Lock App: specified PIN code was not found in the database, ignoring command to set lock state to "Unlocked" [endpointId=1]

@Survensa what exact commands were run on the client here?  Specifically, what were the steps to ensure that pre-condition 3 ("PINCode valid value is 'xxxxxxxx'") was met?

Survensa, 2022-05-25:
@bzbarsky-apple  I have raised issue for that https://github.com/project-chip/connectedhomeip/issues/18418

woody-apple, 2022-05-27:
SDK Spec Review: @cjandhyala to follow up and update this issue.

bzbarsky-apple, 2022-05-27:
It sounds, based on #18418, that this command was in fact being executed without any PINs set up on the server, in which case unlock with PIN is expected to fail.

andy31415, 2022-06-08:
Bumping this up because not touched for a long time.

I expect door lock certification tests to fail if they are setup without a pin. This failure should be a "cert failure".

sumaky, 2022-06-14:
We can close this issue. Created new issue for Test Plan Update for tracking
https://github.com/CHIP-Specifications/chip-test-plans/issues/1466

closed 2022-06-14 by Survensa